esp8266/Arduino #5324: NTP TZ DZT | DZT not working correct and localtime same as gmt time

TKaluza, 2018-11-08:
Could it be that:

- local and gmtime are the same? 
- Daylight saving time implentation is wrong?

I'm in Germany (so TZ = 1 and DST_MN = 60). The time should be something with 22:10.
The result is: 
```
localtime:isdst=0:yday=0:wday=4:year=70:mon=0:mday=1:hour=8:min=0:sec=2
gmtime   :isdst=0:yday=0:wday=4:year=70:mon=0:mday=1:hour=8:min=0:sec=2

clock:2/54712000ns millis:2054 micros:2054731 gtod:2/30344us time:28802 ctime:(UTC+120mn)Thu Jan  1 08:00:02 1970
------------------ settimeofday() was called ------------------

localtime:isdst=0:yday=309:wday=2:year=118:mon=10:mday=6:hour=23:min=3:sec=59
gmtime   :isdst=0:yday=309:wday=2:year=118:mon=10:mday=6:hour=23:min=3:sec=59

clock:3/66632000ns millis:3066 micros:3066648 gtod:1541538240/903398us time:1541545439 ctime:(UTC+120mn)Tue Nov  6 23:03:59 2018
```

When using TZ 1 and DST_MN 0 I'm getting the right time;
```
localtime:isdst=0:yday=0:wday=4:year=70:mon=0:mday=1:hour=8:min=0:sec=2
gmtime   :isdst=0:yday=0:wday=4:year=70:mon=0:mday=1:hour=8:min=0:sec=2

clock:7/742692000ns millis:7742 micros:7742713 gtod:2/30217us time:28802 ctime:(UTC+60mn)Thu Jan  1 08:00:02 1970
------------------ settimeofday() was called ------------------

localtime:isdst=0:yday=309:wday=2:year=118:mon=10:mday=6:hour=22:min=13:sec=14
gmtime   :isdst=0:yday=309:wday=2:year=118:mon=10:mday=6:hour=22:min=13:sec=14

clock:8/754610000ns millis:8754 micros:8754626 gtod:1541538795/20171us time:1541542394 ctime:(UTC+60mn)Tue Nov  6 22:13:14 2018

```

_Originally posted by @TKaluza in https://github.com/esp8266/Arduino/issues/5298#issuecomment-436413026_

5chufti, 2018-11-08:
as you were not willing to fill the template presented when opening the issue,
and did not even search for related issues the adequate answer is just
#5298  and #4637

TKaluza, 2018-11-09:
Thank you very much for your answer! This should be closed, I did search but as it seems not good enough, so sorry!

An I'm sorry, but I couldn't see the template because I just hit the **...** (rupper right corner) and _open new issue_.

So next time, I'll will of course use the template! sorry!

closed 2018-11-09 by TKaluza