esp8266/Arduino #5328: After upgrade to 2.4.2 WiFi.hostByName returns 0.0.0.0

vanniaz, 2018-11-09:
----------------------------- Delete above -----------------------------

### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:				ESP-WROOM-02
- Core Version:			2.4.2
- Development Env:		PlatformIO
- Operating System:		Windows

### Problem Description

After upgrading to core 2.4.2 (Platformio Espressif 8266 1.8.0) a call to WiFi.hostByName always returns 0.0.0.0. The same sketch compiled under 2.4.1 (Platformio Espressif 8266 1.7.0) returns the correct IP address.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <Arduino.h>
#include <ESP8266WiFi.h>

#define DEBUG_PORT	Serial1

char NTP_server[] = "time.google.com";
IPAddress timeServerIP;     // NTP server address
IPAddress ip(192, 168, 1, 210);
IPAddress gw(192, 168, 1, 1);
IPAddress sn(255, 255, 255, 0);

void setup() {

	// init diagnostic serial port
	DEBUG_PORT.begin ( 115200 );

  // start wireless network in STATION mode
  WiFi.mode(WIFI_STA);
  WiFi.config(ip, gw, sn);
  WiFi.begin("XXX", "$$$$$$$$");	// ssid and password for connection

  // resolve host address
  WiFi.hostByName(NTP_server, timeServerIP);
  DEBUG_PORT.print("NTP server address is ");
  DEBUG_PORT.println(timeServerIP);
}

void loop() {
  // put your main code here, to run repeatedly:
}
```

### Debug Messages

```
with 2.4.2:
NTP server address is 0.0.0.0

with 2.4.1:
NTP server address is 216.239.35.4
```

penfold42, 2018-11-10:
Maybe try telling it which dns server to use explicitly ?

vanniaz, 2018-11-10:
Yes, if I set the dns server explicitly it works. But this means that the end user of the product has to configure DNS addresses, and this could be difficult for some users.
With core 2.4.1 the user had only to set the static address to be assigned to the ESP8266 and the router address, the DNS was handled automatically.
I suppose that with 2.4.1 the DNS request was sent by default to the router address, and the router forwarded it to the DNS addresses set by the ISP.
Indeed, if I set the router address as DNS in WiFi.config, it works even with 2.4.2 (so this could be a workaround).
I do not know the details of the DNS protocol, so any suggestions will be appreciated.

penfold42, 2018-11-10:
Personal view - the DNS should not magically be set by the core to the gateway address.

If you want your product to do it, all cool but your code should be doing this.

Btw, assuming every network’s netmask is 255.255.255.0 is poor form and should be specified by the user too.

Can you not just use DHCP and make all these issues go away ?

vanniaz, 2018-11-10:
DHCP is not the best choice for embedded devices because if you want to reach them you are not sure what address they have been given. OK you can set your DHCP server to assign fixed IP addresses to specific MAC addresses, but again this is something you have to explain to your end users. So a fixed IP is the simplest solution.
I agree that the netmask should be configurable (and already is in my setup page), but you can preset it to 255.255.255.0 and let the users change it if needed.
You say that the DNS should be not set by the core to the gateway address, but it seems that this was happening in 2.4.1, or was it just an unintended side effect?

Pablo2048, 2018-11-10:
To reach DHCP addressed device there are MDNS (aka .local) and/or NETBIOS (aka Windows) addressing so all of these devices can be addressed easily IMHO.

d-a-v, 2018-11-10:
What about `while (WiFi.status() != WL_CONNECTED) { delay(500); Serial.print('.'); }`

vanniaz, 2018-11-10:
@d-a-v : yes the real code uses WiFi.waitForConnectResult(), but the behaviour is the same.
I inserted this line in the sample code:
 while (WiFi.status() != WL_CONNECTED) { delay(500); Serial.print(WiFi.status()); Serial.print(","); }
And the result with 2.4.2 is:
6,6,6,6,6,3,NTP server address is 0.0.0.0
While with 2.4.1 is:
6,6,6,6,6,3,NTP server address is 216.239.35.0
If the router address is passed to WiFi.config as DNS, the NTP server address is resolved also with core 2.4.2.
So the question is: was the gateway address intentionally used as default DNS in 2.4.1? Or was it just a side effect, and the DNS should always be explicitly set when working with static IP?

vanniaz, 2018-11-11:
I have come to the conclusion that a reasonable approach for default DNS configuration, when working with static IP, could be the following:
set dns1 equal to the router IP address
set dns2 to a public dns, e.g. Google DNS service
This should work in most situations, however it could be advisable to give the end user the possibility to enter a specific dns address (perhaps in an "advanced" configuration page, to avoid inadvertent modifications).
This overcomes the differences between core 2.4.1 and 2.4.2, so I am closing this issue. Thanks for your comments and suggestions.

closed 2018-11-11 by vanniaz

diraniyoussef, 2019-08-20:
I have some notes here that may be useful to others. 
1) Setting the primary DNS (e.g. to 8.8.8.8) takes effect even when the esp8266 is assigned a dynamic IP (from the router's DHCP). I've tested it but correct me if this might produce new problems.
2) `WiFi.hostByName( "github.com" , the_IP );` did return 0.0.0.0 not at power on to esp8266, but actually when the router (AP Access Point) restarted and I restarted the esp8266 following it. This was a problem (actually hard resetting using a button solved the problem, but this is not the solution I'm proposing). After all settles, setting again the primary DNS to the same old primary DNS (8.8.8.8) actually didn't solve anything ! 
So what was the solution ? You guessed it. Setting the primary DNS to a different one like 8.8.4.4
After that `WiFi.hostByName( "github.com" , the_IP );` functioned correctly.
Here I will place some code 
`#include "espconn.h" 
IPAddress DNS_IP( 8, 8, 8, 8 ); 
//then down in setup() 
espconn_dns_setserver(0, DNS_IP); //to set the primary DNS to 8.8.8.8 
WiFi.dnsIP(0).printTo(Serial); //to make sure`

ddp75, 2023-04-29:
To be simple, the problem come from WiFi.config (Static IP)

This not work :
WiFi.config(localIP, localGateway, subnet);
WiFi.hostByName(ntpServerName, timeServerIP);

This work 👍 
WiFi.config(localIP, localGateway, subnet, localGateway); //or WiFi.config(localIP, localGateway, subnet, yourDNS1, yourDNS2);
WiFi.hostByName(ntpServerName, timeServerIP);