esp8266/Arduino #5331: analogRead not working correctly in 2.4.2 - worked fine in 2.3.0

Sterling-101, 2018-11-11:
When I first started out with a Wemos D1 Mini board I was using core v2.3.0 and had no problems at all getting a stable reading from A0 using analogRead.
Recently though I updated to v2.4.2 and now it seems the readings are completely different.  In fact it seems that the higher voltages (nearer to 3.1 volts) are getting clipped and will generally always read 1024 whereas the same voltage on 2.3.0 would read 960.

Remember this is a Wemos D1 mini v2 so the max input voltage is 3.3v as per the spec of the board.

I did some searching but haven't found a reference to this on here at all only the getVcc() function [#4282.](https://github.com/esp8266/Arduino/issues/4282#issue-294106650)

Could this be related to this problem at all?   Would rather be using the more up to date core but if this function doesn't work then it won't be viable as the project I am working on requires the analogRead to be accurate for data logging.

5chufti, 2018-11-11:
esp8266 and 'analogRead ... accurate' don't go together at all ...

and for exaplen, on the NodeMcu's the voltage divider is made up from 210k and 100k, giving a max. voltage of 3.1V ... tolerance of resistors unknown.

devyte, 2018-11-12:
@Sterling-101 if you look at our [implementation of analogRead()](https://github.com/esp8266/Arduino/blob/055748ff6fab345511985d684c9e67fb84cddaa5/cores/esp8266/core_esp8266_wiring_analog.c#L29) you'll notice it's a simple wrapper around an SDK system call.
If you read our changelogs for releases after 2.3.0, you'll notice that the SDK has been updated several times.
If you read the SDK changelogs relevant to the updates, you'll notice mention of the adc reading.
Closing, as there is nothing to be done in our core.

closed 2018-11-12 by devyte