esp8266/Arduino #5332: package.json - file name does not follow format guidelines

lecommander, 2018-11-11:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
no stack dump
- [ ] I have filled out all fields below.

#### Platform

- Hardware:				not applicable
- Core Version:			"version": "2.5.0-dev"
- Development Env:		sloeber 2.4]
- Operating System:		Windows

### Settings in IDE

n/a

### Problem Description

I wanted to add this as a third party index url in sloeber (ecliple arduino integration).
The url I had used is: [https://raw.githubusercontent.com/esp8266/Arduino/master/package.json](https://raw.githubusercontent.com/esp8266/Arduino/master/package.json)

Unfortunately the name of package.json does not follow the naming guidelines and and cannot be included - this was also discussed in the [sloeber forum](https://github.com/Sloeber/arduino-eclipse-plugin/issues/1037).

This is the spec:
[https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.6.x-package_index.json-format-specification](https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.6.x-package_index.json-format-specification)
It should be named more like package_esp8266.json.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

not applicable to this issue

igrr, 2018-11-11:
I think the package.json in the root of the repository is a Platformio thing... The Arduino board manager package URL is given in the README.md, it is `http://arduino.esp8266.com/stable/package_esp8266com_index.json`.
(The package index file is not maintained in the repository. It is automatically generated on each release using the package file of the previous version, and the information about the new version.)

lecommander, 2018-11-11:
Thank you very much for the correct link

closed 2018-11-11 by lecommander