esp8266/Arduino #5333: Change SoftAP SSID causes errors in 2.4.2 (works in 2.3.0)

ikin77, 2018-11-11:
#### Platform

- Hardware:	                ESP12 / ESP01 / Nodemcu
- Core Version:			2.3 (work) vs 2.4.2 (fails)
- Development Env:		Arduino IDE
- Operating System:		Ubuntu

### Settings in IDE

- Module:  			Generic ESP8266 
- Flash Size: 		        4MB/1MB
- Flash Frequency:	        [40Mhz]
- CPU Frequency:		[80Mhz|160MHz]
- Upload Using:		[SERIAL]

### Problem Description

I'm having problems migrating from core 2.3.0 to core 2.4.2. At first, I thought it was a MQTT problem, so I opened issue #5271 because I was losing connection to the server and after that, I coudn't reconnect any more.

I've done more research on it. The problem happends when you use wifi in STA and SOFTAP mode, when you change the SoftAP SSID. 

I provide a sketch where you can check that. The sketch connects every "loop" to a HTTP server and shows the HTTP result code. If every so often the SoftAP SSID is changed, works with 2.3.0 core, but fails with 2.4.2.

Thanks in advance and best regards!!!!!! 

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <ESP8266HTTPClient.h>

byte counter = 0;

void setup() {

  Serial.begin(115200);

  WiFi.mode(WIFI_AP_STA);  
  WiFi.softAP("OPEN-SSID");
  WiFi.begin("my-router", "my-passwd");

  Serial.print("Connecting");
  while (WiFi.status() != WL_CONNECTED)
  {
    delay(500);
    Serial.print(".");
  }

  Serial.print("Connected, IP address: ");
  Serial.println(WiFi.localIP());  

}

void loop() {

  if (WiFi.status() == WL_CONNECTED) {

    /* EVERY SO OFTEN WE CHANGE SOFT AP SSID */
    counter++;
    if(counter%5==0) {
        WiFi.softAP("OPEN-SSID");
    }
    if(counter%5==4) {
        WiFi.softAP("LOCK-SSID","mypassword123");
    }
    /* END SOFT AP SSID CHANGE */

    HTTPClient http;

    Serial.print("[HTTP] begin...\n");
    //http.begin("http://api.ipify.org?format=json"); //HTTP
    http.begin("http://jsonplaceholder.typicode.com/todos/1"); 

    Serial.print("[HTTP] GET...\n");
    // start connection and send HTTP header
    int httpCode = http.GET();

    // httpCode will be negative on error
    if (httpCode > 0) {
      // HTTP header has been send and Server response header has been handled
      Serial.printf("[HTTP] GET... code: %d\n", httpCode);

      // file found at server
      if (httpCode == HTTP_CODE_OK) {
        String payload = http.getString();
        //Serial.println(payload);
      }
    } else {
      Serial.printf("[HTTP] GET... failed, error: %s\n", http.errorToString(httpCode).c_str());
    }

    http.end();
  }

  delay(10000);
}

```

### Debug Messages

```

VERSIÓN 2.4.2
==========================

Connecting...Connected, IP address: 192.168.120.76
[HTTP] begin...
[HTTP] GET...
[HTTP] GET... code: 200
[HTTP] begin...
[HTTP] GET...
[HTTP] GET... code: 200
[HTTP] begin...
[HTTP] GET...
[HTTP] GET... code: 200
[HTTP] begin...
[HTTP] GET...
[HTTP] GET... failed, error: connection refused
[HTTP] begin...
[HTTP] GET...
[HTTP] GET... failed, error: connection refused
[HTTP] begin...
[HTTP] GET...
[HTTP] GET... failed, error: connection refused
[HTTP] begin...
[HTTP] GET...
[HTTP] GET... failed, error: connection refused
[HTTP] begin...
[HTTP] GET...
[HTTP] GET... failed, error: connection refused
[HTTP] begin...
[HTTP] GET...
[HTTP] GET... failed, error: connection refused

// Never connects again.....

VERSIÓN 2.3.0
==========================

⸮Connecting......Connected, IP address: 192.168.120.76
[HTTP] begin...
[HTTP] GET...
[HTTP] GET... code: 200
[HTTP] begin...
[HTTP] GET...
[HTTP] GET... code: 200
[HTTP] begin...
[HTTP] GET...
[HTTP] GET... code: 200
[HTTP] begin...
[HTTP] GET...
[HTTP] GET... code: 200
[HTTP] begin...
[HTTP] GET...
[HTTP] GET... code: 200
[HTTP] begin...
[HTTP] GET...
[HTTP] GET... code: 200

// Never fails when connecting.....

```

ikin77, 2018-11-27:
Hi, Any help on this issue? 

I made more research: as client station, local IP/gateway/DNS remains unchanged when changing the SoftAP settings. That's the expected behaviour, but the HTTP/MQTT connections starts failing the same once the SoftAP settings change. Worked with 2.3.0 core and fails with 2.4.2 core.

Thanks again and best regards.

Benhur91, 2018-12-06:
Hi, I'm encountering the same problem. 
When compiling the esp8266 softAP example file it runs perfectly in 2.3.0. I may connect to the AP using my phone or a PC. But in 2.4.2, the same code does not create any AP. Nothing happens.
Thanks in advance for your help.
Best regards.

svefro, 2018-12-17:
Looks like i have encountered this also.

without going too mouch in depth of my code. runs this after boot if configured for ap.

`
WiFi.persistent(false);
WiFi.setSleepMode(WiFiSleepType::WIFI_NONE_SLEEP);
WiFi.mode(WIFI_STA);
WiFi.softAPConfig(*wifi_ap_ip, *wifi_ap_gateway, *wifi_ap_subnet); // pointers to IPAddresses
WiFi.disconnect();
WiFi.mode(WIFI_AP);
if (WiFi.softAP("test", "ps"){
// yey!!
}else{
//failed
// Always fails on 2.4.2
}`

d-a-v, 2018-12-17:
By enabling debug options, you would see
```
        DEBUG_WIFI("[AP] fail passphrase to long or short!\n");
```

svefro, 2018-12-18:
Thanks, That solved my problem. documentation says 8 chars min.
It used work on 2.3.0. with only 2 chars.

d-a-v, 2018-12-18:
Thanks for the report,
This extra check might have been added because of SDK changes since core-2.3.0.

d-a-v, 2018-12-18:
@ikin77 This is not related to your issue, reopening

d-a-v, 2018-12-18:
@Benhur91
SoftAP example (Examples>ESP8266WiFi/WiFiAccessPoint) is working here (just tested with a linux laptop).
Can you elaborate ?

d-a-v, 2018-12-18:
@ikin77 Bug reproduced

d-a-v, 2018-12-19:
It turns out the lwIP default interface is set to AP interface after call to `softAP()`,
that's why packets don't get out through STA interface.

ikin77, 2018-12-19:
Hi, @d-a-v, I appreciate your help. If there's something I can do to help you to fix the bug, please don't hesitate to let me know. Please let us know also if a parch is providen :o)

Thanks again and best regards.

d-a-v, 2018-12-19:
@ikin77 Would you be able to try PR #5526 ?

Assuming you are using latest git or -beta2, and config `lwIP-v2 / lower memory`, you can try it by replacing your `tools/sdk/lib/liblwip2-536-feat.a` by this one
[`tools/sdk/lib/liblwip2-536-feat.a`](https://github.com/esp8266/Arduino/raw/8f3db8860316fc9aba032cf4b6783581f932cd22/tools/sdk/lib/liblwip2-536-feat.a)

ikin77, 2018-12-20:
@d-a-v  Hi, Dave. I first tried with original 2.5.0-beta2 and the bug still was there. Then, I replaced the liblwip2-536-feat.a file as you suggested....et voilà, the sample sketch I provided started to work. You fixed it!!!!! :o)

My actual sketck is more complex than the one provided as sample. Actually, I was having issues with MQTT connections, not with HTTP connections. Let me have a couple of days to test it, I'll let you know if it also works as does with core 2.3.0.

Thanks a lot!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!! :o))))))

PR #5526 (2018-12-21): let lwIP route packets, do not let SDK teaching how to do so
should fix #5333
closed 2018-12-21 by devyte