esp8266/Arduino #5335: esp8266HTTPClient GET/POST request returns -1 status code

8483, 2018-11-11:
The following code gets a server response of -1.

**What am I doing wrong?** I've done a bunch of research and this is the suggested code.

```cpp
#include <ESP8266WiFi.h>
#include <ESP8266HTTPClient.h>

void setup() {
    WiFi.begin("SSID", "password");
}

void loop() { 
    HTTPClient http;

    http.begin("https://jsonplaceholder.typicode.com/posts/1");
    http.addHeader("Content-Type", "application/x-www-form-urlencoded");

    int httpCode = http.GET();
    String payload = http.getString();

    Serial.println(httpCode);           // -1
    Serial.println(payload);            // nothing

    http.end();
}
```

Jeroen88, 2018-11-11:
You should wait for your WiFi to connect. You do a secure (https) request, but do not provide any credentials (fingerprint, certificate). You use the deprecated api.
A working example can be found here: https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266HTTPClient/examples/BasicHttpsClient/BasicHttpsClient.ino

8483, 2018-11-11:
Interesting. Thank you for the answer.

I tried an API **without** SSL and **it worked!** So I guess I need to provide the credentials for HTTPS ones. I tried this one `http://dummy.restapiexample.com/api/v1/employee/34` and the response was printed in the serial output.

Can you please explain what the fingerprint part is? What does this do?  

```cpp
// Fingerprint for demo URL, expires on June 2, 2019, needs to be updated well before this date
const uint8_t fingerprint[20] = {0x5A, 0xCF, 0xFE, 0xF0, 0xF1, 0xA6, 0xF4, 0x5F, 0xD2, 0x11, 0x11, 0xC6, 0x1D, 0x2F, 0x0E, 0xBC, 0x39, 0x8D, 0x50, 0xE0};
```

Also, I am trying to access an API on localhost, but I again get the -1 status. **What am I missing here?**

```cpp
void loop() { 
    HTTPClient http;

    http.begin("http://localhost:3003");
    http.addHeader("Content-Type", "application/x-www-form-urlencoded");

    int httpCode = http.GET();
    String payload = http.getString();

    Serial.println(httpCode);           // -1
    Serial.println(payload);            // nothing

    http.end();
}
```
I am waiting for the Wi-fi to connect, just didn't post the code.  Here it is.

```cpp
  WiFi.begin("ZeGermans", "Veselinovski100"); 
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.println("Waiting for connection...");
  }
  Serial.println("Wi-fi connected!");
  Serial.println(WiFi.localIP()); 
```