esp8266/Arduino #5337: ESP8266 Crash during interrupt serve

bhavikmer, 2018-11-12:
### Basic Infos

- [ ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [ ] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:			[ESP-12|
- Core Version:	
- Development Env:		[Arduino IDE]
- Operating System:	[Ubuntu]

### Settings in IDE

- Module:  			[Nodemcu 1.0]
- Flash Mode:			SPIFFS
- Flash Size: 		        4 MB
- lwip Variant:                  [v2 Lower Memory]
- Reset Method:		[nodemcu]
- Flash Frequency:	
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:              115200 

### Goal
- Whenever there is "Low" signal in pin D1, interrupt will be generated, it will reset ESP8266 and publish some data over network and enter to light sleep mode.
- Interrupt frequency is  higher than 10 ms.

### Problem Description
- Whenever Interrupt occurred on pin, esp8266 got crashed. Debug message is mentioned below.
- I have seen some issue regarding this, in that they have suggest to use ICACHE_RAM_ATTR, that i have also tried, but outcome is same.

#include <ESP8266WiFi.h>
#include <PubSubClient.h>
#include <ArduinoJson.h>

void ICACHE_RAM_ATTR handleInterrupt()
{  
  int int_cntr;

   int_cntr = 0;
   
    for (int a = 0; a < 50; a++) {
      
      if (digitalRead(ESP_TOUCH_MODE_PIN) != 0) {
          int_cntr++;   
          if (int_cntr >= 10) {    
           break;

          }
          
      }
      delay(1);
      
   }
   if (int_cntr < 10) {      
      ESP.restart();

   } 
   timer1_write(TMR1_DLY_CNTS);
    
}
void setup() {
   pinMode(D1, INPUT_PULLUP);
  
  attachInterrupt(digitalPinToInterrupt(D1), handleInterrupt, FALLING);

  setup_wifi();
  client.setServer(mqtt_server, mqtt_port);
  client.setCallback(callback);
  Push_Data_To_Gateway();    // Publish Data (MQTT) and enter in to light sleep mode

}

void loop() {
      // main loop is blank
}

```

### Debug Messages
Exception (9):
epc1=0x40104bb5 epc2=0x00000000 epc3=0x00000000 excvaddr=0xffffffff depc=0x00000000

ctx: sys 
sp: 3fffeb90 end: 3fffffb0 offset: 01a0

>>>stack>>>
3fffed30:  3fffed40 3fff03dc 00000000 3ffef7e1  
3fffed40:  4021cc5e 00000030 0000001e ffffffff

bertmelis, 2018-11-12:
First: Please decode the stack. See https://arduino-esp8266.readthedocs.io/en/latest/faq/a02-my-esp-crashes.html

Second: I didn't closely look at your interrupt handler but it should be short and fast. you can always set a flag and do the work (+ clear the flag) in the `loop()` function.

devyte, 2018-11-12:
You can't delay(1) or yield() or anything like that in an ISR. Also, please keep in mind that an ISR has very strict timing restrictions.
Closing due to user error.

closed 2018-11-12 by devyte

bhavikmer, 2018-11-13:
Ya its working as I have made ISR shorter.
Thanks