esp8266/Arduino #5340: I2C clock stretching not implemented at all occurrences

maarten-pennings, 2018-11-14:
When using as I2C slave the CCS811 gas sensor from ams, there are communication errors.
These are due to the CCS811 doing clock stretching.

The file "core_esp8266_si2c.c - Software I2C library for esp8266" implements clock stretching.
But at two places clock stretching is missing.
Functions `twi_writeTo()` and `twi_readFrom()` pull the clock line low using SCL_HIGH(), without checking whether SCL really goes high.

In both functions, one line should be added:
```
  while(SDA_READ() == 0 && (i++) < 10){
    SCL_LOW();
    twi_delay(twi_dcount);
    SCL_HIGH();

    unsigned int t=0; while (SCL_READ() == 0 && (t++) < twi_clockStretchLimit); // Clock stretching

    twi_delay(twi_dcount);
  }
```

devyte, 2018-11-15:
@maarten-pennings I don't have a way to test this. Of note, there are currently no I2C examples or tests. Is there any chance you can implement something that doesn't depend on specialized hardware? 
Example: two ESPs where one is master and the other is slave, and one sends a hello world to the other. Something like that could then be used as a starting point to test functionality like the issue you describe here.

devyte, 2018-11-19:
@suculent @mrwgx3 @bjoham You've contributed to our I2C. Could one of you please help out with assessing the changes proposed here, and with my request for a Wire master/slave example with 2 ESPs?

suculent, 2018-11-19:
I have working implementation of master-slave communication, it’s just a bit complex as it involves simple protocol with CRC check and request-retransfer logic. 

It would all need to be stripped down first. 

Odesláno z iPhonu

devyte, 2018-11-20:
@suculent that would make a great example.
Out of curiosity, why the crc and retry? Did you experience communication errors?

suculent, 2018-11-21:
Here is working version of the I2C master/slave example with two ESPs. Please comment, so I can clean it, style it and get ready for PR (in case it will work to someone else than me :o).

https://github.com/suculent/esp8266-I2C-slave

devyte, 2018-11-21:
@suculent it looks very good! There's a lot of code duplicity between the master and slave sketches, but it's an example, so I'm ok with that.
My other comments are specific to code style and conventions. I think we can discuss that directly in the PR.
The next step after this would be to figure out how to mock clock stretching in the slave to test the code change proposed by @maarten-pennings here.

suculent, 2018-11-21:
I have no idea, honestly :o)

Odesláno z iPhonu

suculent, 2018-11-21:
I was wondering whether the code duplicity could be solved by shared header but including ../ seems expectably forbidden. 

Odesláno z iPhonu

suculent, 2018-11-21:
Clock-stretch fix as proposed by @maarten-pennings: https://github.com/esp8266/Arduino/pull/5362
My humble master-slave I2C test converted to example: https://github.com/esp8266/Arduino/pull/5360

suculent, 2018-11-21:
The clock-stretching seems to be done from slave side by holding the clock LOW. What the fix does is that it reads the clock and waits until it is low in its two cases as highlighted in copy-paste below.

[Source](https://www.i2c-bus.org/clock-stretching/)
> In an I2C communication, the master device determines the clock speed. The I2C bus provides an explicit clock signal which relieves master and slave from synchronizing exactly to a predefined baud rate.
However, there are situations where an I2C slave is not able to co-operate with the clock speed given by the master and needs to slow down a little. This is done by a mechanism referred to as clock stretching.
An I2C slave is allowed to hold down the clock if it needs to reduce the bus speed. **The master, on the other hand, is required to read back the clock signal after releasing it to the high state and wait until the line has actually gone high.**

maarten-pennings, 2018-11-22:
This is the official specification of i2c:

https://www.nxp.com/docs/en/user-guide/UM10204.pdf

"Generation of clock signals on the I2C-bus is always the responsibility of master devices; each master generates its own clock signals when transferring data on the bus. Bus clock signals from a master can only be altered when they are stretched by a slow slave device holding down the clock" ... and several other quotes

suculent, 2018-11-22:
Maarten, your input has been already merged to the master branch and will be present in next release. We're following up with the I2C peer-to-peer communication example, which still needs some cleanup after being thoroughly reviewed by @devyte, which must have been hell of a job.

devyte, 2018-11-23:
Closing since #5362 is already merged. 
The master-slave example PR covers #5288 , and will be handled in that context.

closed 2018-11-23 by devyte

devyte, 2018-12-20:
Yes, comments are allowed after closing!
I can't answer your comment, because I'm not really familiar with the protocol specifics. However, someone else might. If you find a specific problem, feel free to open a new issue for discussion. 
In the meantime, please continue investigating!
Also, there are other I2C libs out there, you could compare behavior.

maarten-pennings, 2018-12-21:
I have written an i2c clock stretch injector

  https://github.com/maarten-pennings/I2C-tool

and a suggestion how to solve both https://github.com/esp8266/Arduino/issues/5340 and https://github.com/esp8266/Arduino/issues/5528:

  https://github.com/maarten-pennings/I2C-tool/blob/master/I2Ctest8266/README.md#how-to-fix