esp8266/Arduino #5341: espota.py update routed from different subnet fails

Harvie, 2018-11-14:
Hello, i have espota.py working in local network, but when i try to run espota from neighbor network (behind router) it fails with following message. ESP device is otherwise OK, it pings and i can reach the WEB UI running on it.

```
18:07:22 [INFO]: Starting on 0.0.0.0:51843
18:07:22 [INFO]: Upload size: 625536
Sending invitation to 10.123.123.123
18:07:22 [INFO]: Waiting for device...
18:07:32 [ERROR]: No response from device
```

devyte, 2018-11-14:
Given that it works in the local network, and not from the other network behind a router, the issue is not in the core, but with your network setup, probably your router.
This is not the right place to seek such help, this is an issue tracker, meant to track issues in the core. I suggest seeking help at a community forum, such as esp8266.com or stackoverflow.
Closing.

closed 2018-11-14 by devyte

Harvie, 2018-11-14:
I think that problem is that ArduinoOTA code, that handles OTA in the sketch is not using default gateway to direct the traffic to other subnets.

Harvie, 2018-11-14:
Everything works when i try netcat to do test of both UDP and TCP connections between these two subnets on the same port as OTA used.