esp8266/Arduino #5346: write uint8_t array truncated

devel0, 2018-11-17:
### Platform

- Hardware:				NodeMCU 1.0 (ESP-12E Module)
- Core Version:			2f4380777ef3f1a32a7d539371d910820d0c6f6a
- Development Env:		VSCode + arduino extension
- Operating System:		Ubuntu

### Settings in IDE

- Module:  			[Generic ESP8266 Module|Wemos D1 mini r2|Nodemcu|other]
- Flash Size: 		4MB
- CPU Frequency:		80Mhz
- Upload Using:		SERIAL
- Upload Speed:   115200

### Problem Description

using client.write with array bigger than 2.9k fails

### Sketch

just uncomment [this](https://github.com/devel0/iot-arduino-ioexpander-esp8266-controller-twi/blob/8b63365c6df6174e5cfd3dbe7440e552d92a8e3d/esp8266-controller-twi/WiFiUtil.cpp#L182)

### workaround

I wrote a helper to write through a [buffered method](https://github.com/devel0/iot-arduino-ioexpander-esp8266-controller-twi/blob/8b63365c6df6174e5cfd3dbe7440e552d92a8e3d/esp8266-controller-twi/WiFiUtil.cpp#L127)

d-a-v, 2018-11-17:
Your "buffered" version would better be called a "chunked" version.
Could you enable debug options, and also check write() return value ?
`client.write()` should transfer everything unless a timeout happens.
It there is no timeout, it may be a bug in the core.
The loop could be:
```
for (auto remain = size; remain -= write(data + size - remain, remain); )
    Serial.println("timeout?");
```

devel0, 2018-11-17:
Could you suggest me which macro define to enable debug mode ?
I tried DEBUG_ESP_CORE but in serial output I see nothing about it.

```c
Serial.println(client.write((uint8_t *)image, image_len));
```

Function `write` for a 24728 bytes input array returns 2920.
Browser not reports a timeout as follows
[image]

dragondaud, 2018-11-17:
In the Arduino IDE you enable a debug port and then which types of debug messages you want in the Board config. You can look in boards.txt to see what the actual defines are for each option.

devel0, 2018-11-18:
Thank you dragondaud, I was able to enable debug, I didn't know that feature and I had to set in boards.txt for my nodemcu 1.0 following by hand because I think in vscode that I use it not yet implemented the menu feature for these parameters; sincerely I tried Arduino IDE too but Debug menu doesn't appears under tools.

```
nodemcuv2.build.debug_port=-DDEBUG_ESP_PORT=Serial
nodemcuv2.build.debug_level=-DDEBUG_ESP_CORE
```

debug execution result follows ( I cutted debug for non related files ) : 

```
readed ssid [labwlan] from eeprom
Trying connecting SSID:[labwlan]
Connecting WiFi ( press CTRL+C to stop )
..You're connected to the network
SSID: labwlan
BSSID: B0:4E:26:C4:F:56
signal strength (RSSI): -67
IP Address: 10.10.3.11
MAC address: DC:4F:22:1C:2E:E2
Type ? for commands
WS:ac
:ref 1
WS:av
:ref 2
:ur 2
:rn 415
header [GET /image.png HTTP/1.1
]
:wr
WS:ac
:rn 312
:sent 15
:ww
:wr
:sent 2
:ww
:wr
:sent 23
:ww
:wr
:sent 2
:ww
:wr
:sent 2
:ww
:wr
:sent 1460
:sent 1460
:ww
:ur 1
:close
WS:dis
:del
:ref 1
WS:av
:ref 2
:ur 2
```

devyte, 2019-11-20:
@earlephilhower @d-a-v 
Chunked algorithm?

earlephilhower, 2020-01-01:
This is clearly a difference between the Arduino's (wrong IMO) definition of `write()` vs. our own.

Per Arduino.cc https://www.arduino.cc/en/Reference/WiFiClientWrite :

>write()
Description
Write data to the server the client is connected to.
...
Returns
byte: the number of characters written. it is not necessary to read this value.
````

So Arduino `write` a) only supports up to 255 bytes (can only return a `byte` written length!) and b) is 100% blocking and will not return until all bytes are written.  They also say "no need to read the value" but then you have no idea if an error occurred, which is really bad programming method.

Our own `write` follows POSIX `man 2 write` semantics and writes >=1 byte and returns the number written, which may be less than the number requested.  A 0 indicates an error.  The user needs to use a loop like @d-a-v to check (and potentially yield(), do other stuff, etc.) when sending data, and to check for errors.

Closing as a MCVE source issue.

closed 2020-01-01 by earlephilhower