esp8266/Arduino #5348: [HTTP-Client][writeToStreamDataBlock] too less ram! need 1460

mjoelmendes, 2018-11-18:
Hello. I'm using a esp8266 12E with these libraries:

#include <Arduino.h>
#include <ESP8266HTTPClient.h>
#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <ESP8266mDNS.h>
#include <ESPAsyncTCP.h>
#include <ESPAsyncWebServer.h>
#include <ArduinoJson.h>
#include "Crypto.h"
#include "base64.h" //
#include <EEPROM.h>
#include <ESP8266httpUpdate.h>
#include <Hash.h>
#include <FS.h>
#include <user_interface.h>
#include <WiFiUdp.h>
#include <mDNSResolver.h>
#include <DNSServer.h>

I'm connecting to a local server.
Spent some time random, appears to me this error:

RX: 'Content-Length: 0' [HTTP-Client] [handleHeaderResponse] RX: 'HTTP-Client' [handleHeaderResponse] RX: 'Accept-Ranges: none' [HTTP-Client] [handleHeaderResponse] RX: 'Connection: close' [HTTP-Client] [handleHeaderResponse] RX: '[HTTP-Client] [handleHeaderResponse] code: 200 [HTTP-Client] [writeToStreamDataBlock] too less ram! need 1460 [HTTP-Client] [returnError] error (-8): too less ram [HTTP-Client] [returnError] tcp stop [HTTP-Client] [end] tcp is closed

[HTTP-Client] connected to 192.168.1.75:80[HTTP-Client] sending request header --- [HTTP-Client] [begin] host: 192.168.1.75 port: 80 uri: / md /? Id = 1 --GET / md /? Id = 1 & cmd = data HTTP / 1.1
Host: 192.168.1.75
User-Agent: ESP8266HTTPClient
Connection: keep-alive
Accept-Encoding: identity; q = 1, chunked; q = 0.1, *; q = 0

[HTTP-Client] [returnError] error (-4): not connected [HTTP-Client] [returnError] error (-2): send header failed [HTTP-Client] [returnError] -Client] [end] tcp is closed
-2

Could someone help me get around this?

dragondaud, 2018-11-18:
Ignored template. This is an issue board not a help forum. Try esp8266.com for general help, close issue.

mjoelmendes, 2018-11-19:
HELLO @dragondaud 

it is not a help but a problem, because it seems to me that ESP8266HTTPClient has a bug, because after some time of "gets" or "post" to a server, (whatever it is), it returns:
error -2 and then goes to -1, the solution and restart the esp8266 to be able to work again, so I say it is a bug.
It starts well after some time stays like this.

devyte, 2018-11-19:
@mjoelmendes what @dragondaud means is that there is an issue template that you're requested to fill out, which provides us with the context of your issue. You have ignored it, and failed to even provide a sketch to reproduce.
Off the top of my head, your problem has to do with mem fragmentation,  and/or mem leak, and/or mem corruption.
BTW, there are known mem leaks with certain functionality of the current MDNS implementation. It has been rewritten, and the new code will be evaluated in the next few days for integration as an alternative to  the current one. I'm not saying this is necessarily your problem, but it is something to be aware of.
I suggest monitoring available heap, fragmentation index, and max available block size, see EspClass and readthedocs for details This is available in latest git and hasn't been released yet. If all of that still shows ok, then reduce your project to an MCVE sketch that reproduces your problem, and open a new issue here with that along with the other info requested in the issue template. If you need help reducing, please ask at a community forum.
Closing due to lack of info.

closed 2018-11-19 by devyte