esp8266/Arduino #5357: ESP8266 Blue Not flashing

HarukiHirosan, 2018-11-20:
### Basic Infos

- [x ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x ] I have tested that the issue is present in current master branch (aka latest git).
- [x ] I have searched the issue tracker for a similar issue.
- [x ] If there is a stack dump, I have decoded it.
- [ x] I have filled out all fields below.

#### Platform

- Hardware:				Blue Esp8266
- Development Env:		Arduino IDE
- Operating System:		Windows

### Settings in IDE

- Module:  		Generic ESP8266 Module
- Flash Mode:		qio
- Flash Size: 		512K(64K spiffs)
- lwip Variant:  v2 Lower Memory
- Reset Method:	ck
- Flash Frequency:	40Mhz
- CPU Frequency:	80Mhz
- Upload Using:	SERIAL
- Upload Speed:   115200

### Problem Description

This never happend to me before, I was working on black ESPs instead of the current blue ones and the software flashed without any problems on the same circuit. I know how to flash the ESP. I have done it countless times before. When I flash it, the error below pops up. The ESP does communicate with the PC as I can see the previous codes serial outputs from the ESP. 

Scketch is BareMinimum

### Debug Messages

```
Build options changed, rebuilding all
Archiving built core (caching) in: C:\Users\demir\AppData\Local\Temp\arduino_cache_821590\core\core_esp8266_esp8266_generic_CpuFrequency_80,VTable_flash,ResetMethod_ck,CrystalFreq_26,FlashFreq_40,FlashMode_dio,FlashSize_512K64,led_2,LwIPVariant_v2mss536,Debug_Disabled,DebugLevel_None____,FlashErase_none,UploadSpeed_115200_38bace12de6c336b3623be9fea17f5c9.a
Sketch uses 246080 bytes (56%) of program storage space. Maximum is 434160 bytes.
Global variables use 27572 bytes (33%) of dynamic memory, leaving 54348 bytes for local variables. Maximum is 81920 bytes.
Uploading 250224 bytes from C:\Users\demir\AppData\Local\Temp\arduino_build_909967/BareMinimum.ino.bin to flash at 0x00000000
warning: espcomm_send_command: didn't receive command response
warning: espcomm_send_command(FLASH_DOWNLOAD_BEGIN) failed
error: espcomm_upload_mem failed
error: espcomm_upload_mem failed
```

Jeroen88, 2018-11-21:
Did you try the other reset methods?

HarukiHirosan, 2018-11-21:
Other methods? Can you give me an example please?

devyte, 2018-11-21:
This could be a board parameter that is different for your new boards (like the reset method mentioned above, or the upload speed, etc), or it could be something in the wiring that needs to be different for putting that ESP into flash mode, or any number of other reasons. 
Whatever the cause of your issue, it's not a problem with the core hosted in this repo.
Closing.

closed 2018-11-21 by devyte