esp8266/Arduino #5359: Cannot connect to WEP network with WiFi.enableInsecureWEP()

porkyneal, 2018-11-21:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:				Wroom02
- Core Version:			74ca42f
- Development Env:		Platformio
- Operating System:		Windows

### Settings in IDE

- Module:  			Generic ESP8266 Module
- Flash Size: 		2MB
- lwip Variant:  v2 Lower Memory
- Reset Method:		nodemcu
- CPU Frequency:		160MHz

### Problem Description

An existing working setup has stopped being able to connect to a WEP network. If i change AP settings to WPA2 everything works as expected. This has been since upgrading the git core, unfortunately I cannot pinpoint the exact previous working git version.

I have tried before WiFi begin:
WiFi.enableInsecureWEP(true); 
WiFi.enableInsecureWEP(); 

WiFi error is 1  - Unreachable WiFi Network

All other device connect and work perfectly fine with the access point and the same credentials.

Here is a wifiscan output from the ESP, I have obscured the SSIDs, SSIDWPA2 is WPA2 encrytption and SSID WEP - WEP.

"SSID" : SSIDWEP,
"RSSI" : -30,
"encryption" : 5,
"BSSID" : 46:D9:E7:F3:35:B8,
"channel" : 6,
"isHidden" : false

"SSID" : SSIDWPA2,
"RSSI" : -30,
"encryption" : 4,
"BSSID" : 44:D9:E7:F3:35:B8,
"channel" : 6,
"isHidden" : false

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

I will create an MVCE if required, but my setup routine looks like this:

`void wifiSetup(){
    WiFi.hostname(config.deviceID);
    WiFi.enableInsecureWEP(); 
    WiFi.mode(WIFI_AP_STA);

    if(config.wifiSettings.fixedSettings){
        Serial.println(F("Applying fixed wifi settings"));
        IPAddress staticIP, gateway, subnet, dns1, dns2;
        if(staticIP.fromString(config.wifiSettings.ip)){
            if(gateway.fromString(config.wifiSettings.gateway)){
                if(subnet.fromString(config.wifiSettings.mask)){
                    if(dns1.fromString(config.wifiSettings.dns1)){
                        if(dns1.fromString(config.wifiSettings.dns2)){
                            WiFi.config(staticIP, gateway, subnet, dns1, dns2);
                        } else {
                            Serial.println(F("Failed DNS2 Address conversion"));
                            WiFi.config(staticIP, gateway, subnet, dns1);
                        }
                    }else{
                        Serial.println(F("Failed DNS1 Address conversion"));
                        WiFi.config(staticIP, gateway, subnet);
                    }        
                } else Serial.println(F("Failed Net Mask conversion"));  
            } else Serial.println(F("Failed Gateway Address conversion"));
        } else Serial.println(F("Failed IP Address conversion"));
    }
    WiFi.softAP(config.apSSID, config.apPassword);    
    if(strlen(config.wifiSettings.ssid) != 0){
        Serial.println(F("Begin WiFi"));
        Serial.print(F("Connecting to "));
        Serial.println(config.wifiSettings.ssid);
        WiFi.begin(config.wifiSettings.ssid, config.wifiSettings.password);
    }
}`
### Debug Messages

```
mode : sta(84:f3:eb:46:dd:c3) + softAP(86:f3:eb:46:dd:c3)
add if0
Begin WiFi
Connecting to SSIDWEP
scandone
no SSIDWEP found, reconnect after 1s
reconnect
scandone
no SSIDWEP found, reconnect after 1s
reconnect
scandone
no SSIDWEP found, reconnect after 1s
reconnect
scandone
no SSIDWEP found, reconnect after 1s
reconnect
scandone
no SSIDWEP found, reconnect after 1s
reconnect
scandone
no SSIDWEP found, reconnect after 1s
reconnect
WiFi Failure - 1
Unreachable WiFi Network
scandone
del if0
mode : softAP(86:f3:eb:46:dd:c3)
```

devyte, 2018-11-21:
@porkyneal said:
> I will create an MVCE if required

It's required, please edit your post and complete the sketch.

In the meantime:
- What happens at 80MHz CPU freq?
- What happens if you build and flash from the Arduino IDE?
- What happens if you set mode to only STA instead of AP_STA?
- Is the same problem present in 2.4.2 release?

Out of curiosity, why do you want WEP? It's trivially breakable, and the recommendation is to not use it in any app that has even a tiny shred of seriousness. There was a reason why it was left exposed in our core, but I can't even remember it anymore.

porkyneal, 2018-11-21:
@devyte While WEP is a legitimate configurable option for an access point (even Ubiquiti leave it as an option) I believe I should support it for the ease of my customers.

I will get back to you on the points you have raised in due course.

porkyneal, 2018-11-22:
My quick tests and MVCE:

Changing CPU Freq, using the Arduino IDE and setting WIFI_STA only had no effect. 

With regards to the core versions:

I had a git master installed in Arduino IDE from 23/10/18 in which the WiFi.enableInsecureWEP(); acted as expected, when not in the code no WEP connection, when added connection is fine. <-- WORKING
(Is there somewhere I can find the git version inside the esp8266 folder?)

In the latest git the same code does not connect, with or without WiFi.enableInsecureWEP();. <-- NOT WORKING

In 2.4.2 WiFi.enableInsecureWEP(); was not implemented and WEP connection works fine. <-- WORKING

Here is my MVCE that I used for the above, based on the WiFiClientBasic example:
```
#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>

const char* ssid     = "WEPSSID";
const char* password = "WEP_PASS"; // the wep pass is 5 chars

ESP8266WiFiMulti WiFiMulti;

void setup() {
  Serial.begin(115200);
  delay(10);
  WiFi.enableInsecureWEP(); 
  // We start by connecting to a WiFi network
  WiFi.mode(WIFI_STA);
  WiFiMulti.addAP(ssid, password);

  Serial.println();
  Serial.println();
  Serial.print("Wait for WiFi... ");

  while (WiFiMulti.run() != WL_CONNECTED) {
    Serial.print(".");
    delay(500);
  }

  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());

  delay(500);
}

void loop() {
  delay(5000);
}

```

porkyneal, 2018-11-22:
UPDATE: When I revert #5317 in the latest git, WiFi.enableInsecureWEP(); works again.

dav1901, 2018-11-22:
I believe this #5364 should fix this.

porkyneal, 2018-11-22:
Thanks @dav1901, this does seem to solve my problem. 👍

PR #5364 (2018-11-22): Check for WEP when setting authmode
Fix #5359.
closed 2018-11-22 by porkyneal