esp8266/Arduino #5361: pgm_read_word(addr) fails if (addr & 0x3 == 3)

amendoim, 2018-11-21:
### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [  ] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below.

#### Platform

- Hardware:					ESP-12E
- Core Version:			2.4.0
- Development Env:		Arduino IDE, Platformio
- Operating System:		Windows

### Settings in IDE

- Module:  			Nodemcu
- Flash Mode:			qio
- Flash Size: 		4MB
- lwip Variant:   
- Reset Method:		nodemcu
- Flash Frequency:	40Mhz
- CPU Frequency:		80Mhz
- Upload Using:		SERIAL
- Upload Speed:   115200

### Problem Description

Using pgm_read_word on an address where (addr & 0x03 == 3) will only return 8 bits, not the requested 16 bits. 
In the example below, one of the 16-bit word reads returns 1 and not 257 (which is 1 << 8 | 1) as it should.
This happens because the inner function pgm_read_with_offset() reads aligned 32 bits and then shifts as needed. This fails on the 3rd byte which only holds the remaining 8 bits.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <Arduino.h>
PROGMEM const uint8_t table[] = {
  0x01, 0x01, 
  0x01, 0x01,
  0x01, 0x01,
  0x01, 0x01,
};

void setup() {
  Serial.begin(115200);
}

void loop() {
  Serial.println( pgm_read_word( &table[0]) );
  Serial.println( pgm_read_word( &table[1]) );
  Serial.println( pgm_read_word( &table[2]) );
  Serial.println( pgm_read_word( &table[3]) );
  Serial.println( pgm_read_word( &table[4]) );
  delay(7*1000);
}

```

### Debug Messages

```
257
257
257
1
257
```

earlephilhower, 2018-11-23:
This is working as-expected.  pgm_read_* requires the same alignment rules as the rest of the processor when accessing memory.  pgm_read_word is only valid on even addresses, pgm_read_dword on every 4 bytes, etc.

You will need to assemble values byte-by-byte if you want to use packed, misaligned elements.

closed 2018-11-23 by earlephilhower

amendoim, 2018-11-23:
Thanks for looking into this. However, isn't the point of ESP8266 Arduino to be compatible with Arduino sketches? 

I've run the above code on an AVR Arduino UNO, and on an STM32 Maple Mini and got the correct 257 result on all addresses. This seems to be the expected behavior of pgm_read_word() on other "Arduino" platforms.

I think having this Arduino incompatible behavior is a bad thing that will keep creating problems for people, and a fix would be relatively trivial and compatible with the current behavior of only working on even addresses...

amendoim, 2018-11-23:
Oh - indeed! Unfortunately I had to learn this the hard way on the ESP8266 with mysterious crashes and exceptions of misaligned accesses. 
However it's my understanding that at least the current gcc deals with most accesses - for example the same array in memory (tableR below) works fine:

```cpp
#include <Arduino.h>
PROGMEM const uint8_t table[] = {
  0x01, 0x01, 
  0x01, 0x01,
  0x01, 0x01,
  0x01, 0x01,
};

const uint8_t tableR[] = {
  0x01, 0x01, 
  0x01, 0x01,
  0x01, 0x01,
  0x01, 0x01,
};

void setup() {
  Serial.begin(115200);
}

void loop() {
  Serial.println( "--------FLASH-------" );
  Serial.println( pgm_read_word( &table[0]) );
  Serial.println( pgm_read_word( &table[1]) );
  Serial.println( pgm_read_word( &table[2]) );
  Serial.println( pgm_read_word( &table[3]) );
  Serial.println( pgm_read_word( &table[4]) );

  Serial.println( "--------RAM---------" );
  Serial.println( *(u16*)( &tableR[0]) );
  Serial.println( *(u16*)( &tableR[1]) );
  Serial.println( *(u16*)( &tableR[2]) );
  Serial.println( *(u16*)( &tableR[3]) );
  Serial.println( *(u16*)( &tableR[4]) );

  delay(7*1000);
}
```

---------------------------------------

But nevertheless, what's important in my point above is that since Arduino for ESP8266 is a compatibility layer, I believe this should be corrected to have a behavior identical to all other "Arduino" platforms. Otherwise, this will only cause misery for unaware people porting into ESP8266 (which I believe is the main point of this Arduino ESP8266 core).

Functions in pgmspace.h, like  pgm_read_word() should return a correct word no matter the address. I confirmed this is the behavior in other Arduino pgmspace.h implementations like STM32 and AVR 8-bit.

I'm not familiar with the fix suggestion process: if this is not the right place to raise this issue, could you please point me in the right direction?

devyte, 2018-11-23:
@amendoim This is the right place to discuss this, the implementation of pgm_read_*() is ours.

There are two sort of opposing points of view regarding this topic.
On the one hand, there is Arduino core compatibility, which pulls towards having compatible behavior.
On the other hand, there is an architecture change here, and the request boils down to trying to support 8bit behavior on a 32bit platform.

The latter is not done as normal practice. Even when migrating from 32bit to 64bit architecture there is impact and in most cases code changes are required. That means that it's not really possible to seamlessly support 32bit behavior on 64bit architecture for all app cases. The same holds true when migrating from 8 or 16bit architectures to 32bit architectures.

My personal point of view is that Arudino compatibility should be maintained across different uCs of similar architecture, that support similar behavior, but not across architectures that are vastly different. In this last case, the user must be made aware that certain code pieces must be adapted in order to allow porting.

In addition, per an internal discussion, implementation of 8bit behavior for misaligned reads would impact performance of all aligned reads throughout the code by a significant amount. This wasn't measured in practice, but comes from reading the current implementation.

Current consensus is that we won't spend effort on this request. 
Having said that, if you were to implement the support, and prove that performance impact for current aligned reads is not significant, we'd be open to discussing your proposal. If you need guidance for this, we can help.
If you choose to pursue this, please report back here your performance comparison results (i.e.: current pgm_read* code vs. your code) , and request that this issue be reopened. The next step after that would be for you to make a PR with the proposal, which we could then test out, and eventually merge.

Makuna, 2018-11-23:
Another viewpoint, from someone who regularly holds Arduino classes.
EEPROM is a compatibility layer at this point.  It should be as compatible as possible.  Extended with methods that more efficient but don't mess with the common methods.  This is how I fixed the original one here.

If one believes that if code requires changes when changing platforms, then throw out compatibility and the library completely and suggest the replacement; in this case, maybe using SPIFFS instead as a better approach.  

I DO NOT agree with the that they should just work different using the same method call; as all this does is put road blocks to the consumers of Arduino (who are often very naïve programmers) which may drive they away from the platform of they hit enough of these road blocks.  I have seen this from several of my students in the past.  With another side effect of creating more support request noise here.

amendoim, 2018-11-26:
As far as I can see, a solution would need to be implemented in asm (along the lines of the pgm_read_with_offset() function) but unfortunately I'm not familiar enough with this CPU, nor I have the time to learn it now. 
But I can suggest the following workaround which works fine: replace pgm_read_word_inlined() with the function below and move it to the bottom of pgmspace.h. 
People looking for the old performance can globally define ESP8266_ORIGINAL_PGMSPACE to use old code (return values for valid addresses are the same in both cases):

```cpp
/* Although this says "word", it's actually 16 bit, i.e. half word on Xtensa */
static inline uint16_t pgm_read_word_inlined(const void* addr) {

#if defined(ESP8266_ORIGINAL_PGMSPACE) // won't work if addr & 3 == 3
  register uint32_t res;
  pgm_read_with_offset(addr, res);
  return (uint16_t) res;    /* This masks the lower half-word from the returned word */

#else // Arduino platform compatible behavior
  return (uint16_t)(pgm_read_byte(addr) | 
                   (pgm_read_byte((u8*)addr + 1) << 8));  
#endif

}
```

Regarding policy, I have to agree with Makuna: why have an ESP8266 Arduino compatibility layer if some code is admittedly not compatible although it could be made compatible? Of course ESP8266 Arduino is very useful just by itself but why not also have it as compatible as possible with other "Arduino" platforms, regardless of differences?

Since Arduino is a layer to ease portability between platforms, I think incompatible behavior such as this only creates unneeded problems.