esp8266/Arduino #5365: wificlint issue with clint.write :: disconnected after write request.

prateek027, 2018-11-22:
Hi All,
i am new in arduino, i have to send 500kb data to server using wificlint.write. Data comes through spi.
here is the handling.

spi.onData{
 memcpy(&data[index],&spiData,32); 
index += 32;

if(index > 12288){
index2 = index;
 memcpy(&data1[0],&data[0],index);
index = 0; 
}
}

void loop {
if(wificlint.connected){
  size_t respLen   =  wificlint.write((const char*)&data1[0],sizeof(data));
  serial.print(respLen,DEC);
}
}

---
in which index2 = 12288 but i am getting respLen = 4096 & after that wificlint.connected is failed. 
can you  please help me to resolved this issue.

Pablo2048, 2018-11-22:
You have been told twice that this is not an support forum. Why are you still ignoring this and the Issue template?

d-a-v, 2018-11-22:
@Pablo2048 is right: You *must* fill the issue template when you open a new issue.
About `client.write(data, len)` try to not read more spi data than `client.availableForWrite()` bytes.

closed 2018-11-22 by d-a-v