esp8266/Arduino #5374: ESP8266(NODEMCU+AWS SHADOW){ets Jan  8 2013,rst cause:2, boot mode:(3,6)}

vivektalwar13071999, 2018-11-24:
`#include <ESP8266WiFi.h>
#include <AmazonIOTClient.h>
#include <ESP8266AWSImplementations.h>
#include <AWSFoundationalTypes.h>
#include <DHT.h>

Esp8266HttpClient httpClient;
Esp8266DateTimeProvider dateTimeProvider;

AmazonIOTClient iotClient;
ActionError actionError;

// DHT Sensor setup
#define DHTPIN 0
#define DHTTYPE DHT11
DHT dht(DHTPIN, DHTTYPE);

char *ssid="";
char *password="";

void setup() {
  Serial.begin(115200);
  delay(10);
  ESP.wdtDisable();
 ESP.wdtEnable(WDTO_8S);

  // LED on nodemcu
  pinMode(BUILTIN_LED, OUTPUT);

  // DHT Sensor
  dht.begin();

  // Connect to WAP
  Serial.print("Connecting to ");
  Serial.println(ssid);
  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());

  iotClient.setAWSRegion("");
  iotClient.setAWSEndpoint("");
  iotClient.setAWSDomain("");
  iotClient.setAWSPath("");
  iotClient.setAWSKeyID("");
  iotClient.setAWSSecretKey("");
  iotClient.setHttpClient(&httpClient);
  iotClient.setDateTimeProvider(&dateTimeProvider);
}

void loop(){
  ESP.wdtFeed();
  float t = dht.readTemperature(true);
  char temp[10];
  dtostrf(t,1,2,temp);
 
  float h = dht.readHumidity();
  char hum[10];
  dtostrf(h,1,2,hum);
  delay(3000);

  MinimalString shadow = ("{\"state\":{\"reported\":{\"temperature\":" + String(temp) + ",\"humidity\":" + String(hum) + "}}}").c_str();

  char* result = iotClient.update_shadow(shadow, actionError);
  Serial.print(result);

  delay(6000);
}`
eRROR ON SERIAL MONITOR
👍ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
vbb28d4a3
~ld

vivektalwar13071999, 2018-11-24:
PLS HELP............

devyte, 2018-11-27:
It's likely that AmazonIoTClient is using the standard WiFiClientSecure, which is deprecated due to too much brokenness and it no longer being maintained. It is mminently about to be dropped to legacy, so instead, BearSSL::WiFiClientSecure should be used.
This is not a support forum, but an issue tracker. Please either open an issue against AmazonIoTClient, or seek peer support at a community forum like esp8266.com or stackoverflow.
Closing.

closed 2018-11-27 by devyte