esp8266/Arduino #5375: subsequent GET failure - connection missed

Pedroalbuquerque, 2018-11-25:
### Basic Infos

- [ ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12]
- Core Version:			[latest git hash or date]
- Development Env:		[Platformio]
- Operating System:		[MacOS]

### Settings in IDE

- Module:  			[Nodemcu]
- Flash Mode:			[qio|dio|other]
- Flash Size: 		[4MB]
- lwip Variant:   [v1.4|v2 Lower Memory|Higher Bandwidth]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] (serial upload only)

### Problem Description

if 2 GET request are send in a loop and on is pointing to a existing url, both fail

###  Sketch

```cpp

[ **** SERVER side code ******] working fine when called from a browser

  server.on("/", handleRoot);

  server.on("/inline", [](){
    server.send(200, "text/plain", "this works as well");
  });

  server.onNotFound(handleNotFound);

  server.begin();
```

```cpp
[ ***** CLIENTE side code *****]

void loop(){
if(WiFi.status() != WL_CONNECTED){
  WiFi.begin(ssid, password);
  delay(500);
  Serial.println("no WiFi connection");
}else{
  //HTTPClient http;
        Serial.print("[HTTP] begin...\n");
        // configure traged server and url
        http.begin(host, 80, "/inline"); //HTTp
        int httpCode1 = http.GET();
        rptaSrv(httpCode1);
        delay(2000);
        http.begin(host, 80, "/led"); //HTTP
        int httpCode2 = http.GET();
        rptaSrv(httpCode2);
        delay(1000);
    }
}

```
note that /led does not exist in the server

### Debug Messages

```
[HTTP-Client][begin] host: 192.168.1.79 port: 80 uri: /led
[HTTP-Client] connected to 192.168.1.79:80
[HTTP-Client] sending request header
-----
GET /led HTTP/1.1
Host: 192.168.1.79
User-Agent: ESP8266HTTPClient
Connection: close
Accept-Encoding: identity;q=1,chunked;q=0.1,*;q=0

-----
[HTTP-Client][handleHeaderResponse] RX: 'HTTP/1.1 404 Not Found'
[HTTP-Client][handleHeaderResponse] RX: 'Content-Type: text/plain'
[HTTP-Client][handleHeaderResponse] RX: 'Content-Length: 51'
[HTTP-Client][handleHeaderResponse] RX: 'Connection: close'

[HTTP-Client][handleHeaderResponse] RX: ''
[HTTP-Client][handleHeaderResponse] code: 404
[HTTP-Client][handleHeaderResponse] size: 51
[HTTP] GET... failed, no connection or no HTTP server
[HTTP] begin...
[HTTP-Client][begin] host: 192.168.1.79 port: 80 uri: /inline
[HTTP-Client] connect. already connected, try reuse!
[HTTP-Client] sending request header
-----
GET /inline HTTP/1.1
Host: 192.168.1.79
User-Agent: ESP8266HTTPClient
Connection: close
Accept-Encoding: identity;q=1,chunked;q=0.1,*;q=0

-----
[HTTP-Client][returnError] error(-5): connection lost
[HTTP] GET... failed, no connection or no HTTP server
[HTTP-Client][begin] host: 192.168.1.79 port: 80 uri: /led
[HTTP-Client] connected to 192.168.1.79:80
[HTTP-Client] sending request header
-----
GET /led HTTP/1.1
Host: 192.168.1.79
User-Agent: ESP8266HTTPClient
Connection: close
Accept-Encoding: identity;q=1,chunked;q=0.1,*;q=0

-----
[HTTP-Client][handleHeaderResponse] RX: 'HTTP/1.1 404 Not Found'
[HTTP-Client][handleHeaderResponse] RX: 'Content-Type: text/plain'

[HTTP-Client][handleHeaderResponse] RX: 'Content-Length: 51'
[HTTP-Client][handleHeaderResponse] RX: 'Connection: close'
[HTTP-Client][handleHeaderResponse] RX: ''
[HTTP-Client][handleHeaderResponse] code: 404
[HTTP-Client][handleHeaderResponse] size: 51
[HTTP] GET... failed, no connection or no HTTP server
[HTTP] begin...
[HTTP-Client][begin] host: 192.168.1.79 port: 80 uri: /inline
[HTTP-Client] connect. already connected, try reuse!
[HTTP-Client] sending request header
-----
GET /inline HTTP/1.1
Host: 192.168.1.79
User-Agent: ESP8266HTTPClient
Connection: close
Accept-Encoding: identity;q=1,chunked;q=0.1,*;q=0

-----

```
note the "[HTTP-Client][returnError] error(-5): connection lost "

If I change the HTTPClient::connect() to

```cpp

bool HTTPClient::connect(void)
{

    if(connected()) {
        end();
    }
 ...

```
everything starts to work fine.

```
[HTTP] begin...
[HTTP-Client][begin] host: 192.168.1.79 port: 80 uri: /inline
[HTTP-Client] connected to 192.168.1.79:80
[HTTP-Client] sending request header
-----
GET /inline HTTP/1.1
Host: 192.168.1.79
User-Agent: ESP8266HTTPClient
Connection: close
Accept-Encoding: identity;q=1,chunked;q=0.1,*;q=0

-----
[HTTP-Client][handleHeaderResponse] RX: 'HTTP/1.1 200 OK'
[HTTP-Client][handleHeaderResponse] RX: 'Content-Type: text/plain'
[HTTP-Client][handleHeaderResponse] RX: 'Content-Length: 18'

[HTTP-Client][handleHeaderResponse] RX: 'Connection: close'
[HTTP-Client][handleHeaderResponse] RX: ''
[HTTP-Client][handleHeaderResponse] code: 200
[HTTP-Client][handleHeaderResponse] size: 18
[HTTP-Client][writeToStreamDataBlock] connection closed or file end (written: 18).
[HTTP-Client][end] tcp stop
this works as well
[HTTP-Client][begin] host: 192.168.1.79 port: 80 uri: /led
[HTTP-Client] connected to 192.168.1.79:80
[HTTP-Client] sending request header
-----
GET /led HTTP/1.1
Host: 192.168.1.79
User-Agent: ESP8266HTTPClient
Connection: close
Accept-Encoding: identity;q=1,chunked;q=0.1,*;q=0

-----
[HTTP-Client][handleHeaderResponse] RX: 'HTTP/1.1 404 Not Found'
[HTTP-Client][handleHeaderResponse] RX: 'Content-Type: text/plain'
[HTTP-Client][handleHeaderResponse] RX: 'Content-Length: 51'
[HTTP-Client][handleHeaderResponse] RX: 'Connection: close'
[HTTP-Client][handleHeaderResponse] RX: ''
[HTTP-Client][handleHeaderResponse] code: 404
[HTTP-Client][handleHeaderResponse] size: 51

[HTTP] GET... failed, no connection or no HTTP server
[HTTP] begin...
[HTTP-Client][begin] host: 192.168.1.79 port: 80 uri: /inline
[HTTP-Client][end] still data in buffer (51), clean up.
[HTTP-Client][end] tcp stop
[HTTP-Client] connected to 192.168.1.79:80
[HTTP-Client] sending request header
-----
GET /inline HTTP/1.1
Host: 192.168.1.79
User-Agent: ESP8266HTTPClient
Connection: close
Accept-Encoding: identity;q=1,chunked;q=0.1,*;q=0

-----
[HTTP-Client][handleHeaderResponse] RX: 'HTTP/1.1 200 OK'
[HTTP-Client][handleHeaderResponse] RX: 'Content-Type: text/plain'
[HTTP-Client][handleHeaderResponse] RX: 'Content-Length: 18'
[HTTP-Client][handleHeaderResponse] RX: 'Connection: close'
[HTTP-Client][handleHeaderResponse] RX: ''
[HTTP-Client][handleHeaderResponse] code: 200
[HTTP-Client][handleHeaderResponse] size: 18
[HTTP-Client][writeToStreamDataBlock] connection closed or file end (written: 18).
[HTTP-Client][end] tcp stop
this works as well

```

devyte, 2018-11-27:
Note: sounds like something wrong with reuse.

earlephilhower, 2019-08-21:
Dupe of #6149 .  Let's track there.

closed 2019-08-21 by earlephilhower