esp8266/Arduino #5381: ESP hangs if wifi connection losses during an open tcp connection

crndn2018, 2018-11-28:
### Basic Infos

- [ ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:				[ESP-12]
- Core Version:			[Release 2.4.2]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Generic ESP8266 Module]
- Flash Mode:			[qio]
- Flash Size: 		        [4MB]
- lwip Variant:                  [v2 Lower Memory]
- Reset Method:		[ck]
- Flash Frequency:	        [40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:              [115200]

### Problem Description

ESP hangs if the wifi connection losses during an open tcp connection. No output. No watchdog reset. 
Please try the example sketch below. Fill in SSID and password. Wait for WIFI connection. Open a terminal to Serial1. Open a putty session on port 23. Switch off your WIFI access point  and have a look on the built-in live-beat .
This problem does not occur if there is no open tcp connection. 

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <ESP8266WiFi.h>
#include <ESP8266mDNS.h>
#include <WiFiUdp.h>

#define MAX_TELNET_CLIENTS 2

const char* ssid = "SSID";
const char* password = "PASSWORD";

WiFiServer TelnetServer(23);
WiFiClient TelnetClient[MAX_TELNET_CLIENTS];
int WifiState = 0;
int WifiStatePrevious = -128;
unsigned int TimerCounterLiveBeat = 0;
unsigned int TimerCounterSend[MAX_TELNET_CLIENTS];
int i = 0;

void setup()
{
  for (i = 0;i < MAX_TELNET_CLIENTS; i++) TimerCounterSend[i] = 0;
  
  Serial1.begin(115200);
  Serial1.println("Telnet Server Example");

  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);

  unsigned long TimerCounterStartTime = millis();
  Serial1.print("Waiting for WIFI connection ");
  while (WiFi.status() != WL_CONNECTED && (millis() - TimerCounterStartTime < 10000))
  {
    delay(200);
    Serial1.print(".");
  }
  Serial1.println();
  
  if (WiFi.status() != WL_CONNECTED)
  {
    Serial1.println("WIFI connection Failed! Rebooting...");
    delay(3000);
    ESP.restart();
  }

  Serial1.print("IP address: ");
  Serial1.println(WiFi.localIP());

  Serial1.println("Starting Telnet Server");
  TelnetServer.begin();
  TelnetServer.setNoDelay(true);
}

void loop() 
{
  Telnet();  
}
      
void Telnet()
{
  delay(1);
  
  // Livebeat
  if (millis() - TimerCounterLiveBeat > 1000)
  {
    TimerCounterLiveBeat = millis();
    Serial1.println("Livebeat.");  
  }
  
  // Watch WIFI state
  WifiState = WiFi.status();
  if (WifiState == WL_CONNECTED && WifiStatePrevious != WL_CONNECTED) Serial1.println("WIFI connected.");
  if (WifiState != WL_CONNECTED && WifiStatePrevious == WL_CONNECTED) Serial1.println("WIFI disconnected.");
  WifiStatePrevious = WifiState;

  // Handle connections
  if (WifiState == WL_CONNECTED)
  {
    // Cleanup disconnected session
    for(i = 0; i < MAX_TELNET_CLIENTS; i++)
    {
      if (TelnetClient[i] && !TelnetClient[i].connected())
      {
        Serial1.print("Client disconnected ... terminate session "); Serial1.println(i+1); 
        TelnetClient[i].stop();
      }
    }
  
    // Check new client connections
    if (TelnetServer.hasClient())
    {
      bool ConnectionEstablished = false; 
      
      for(i = 0; i < MAX_TELNET_CLIENTS; i++)
      {
        // find free socket
        if (!TelnetClient[i])
        {
          TelnetClient[i] = TelnetServer.available(); 
          TelnetClient[i].flush();
          
          Serial1.print("New Telnet client connected to session "); Serial1.println(i+1);
          ConnectionEstablished = true; 
          
          break;
        }
        else
          Serial1.println("Session is in use.");
      }
  
      if (ConnectionEstablished == false)
      {
        Serial1.println("No free sessions ... drop connection");
        TelnetServer.available().stop();
      }
    }
  
    for(i = 0; i < MAX_TELNET_CLIENTS; i++)
    {
      if (TelnetClient[i] && TelnetClient[i].connected())
      { 
        if (millis() - TimerCounterSend[i] > 500)
        {
          TelnetClient[i].println("This is a test string.");
          TimerCounterSend[i] = millis();   
        } 
      }
    }
  }
}

### Debug Messages

crndn2018, 2018-11-28:
The problem is caused by Core 2.4.2. With 2.4.1 it works !

d-a-v, 2018-11-28:
You sketch gives with current git version (FW has been updated since 2.4.2):
```
19:02:49.437 -> SDK:3.0.0-dev(c0f7b44)/Core:2.4.2-169-gffaa93f/lwIP:STABLE-2_1_2_RELEASE/glue:1.0/BearSSL:0e71e46
19:02:49.437 -> Telnet Server Example
...
19:02:52.656 -> IP address: 10.0.1.225
19:02:52.656 -> Starting Telnet Server
19:02:52.656 -> Livebeat.
19:02:52.656 -> WIFI connected.
19:02:53.649 -> Livebeat.
19:02:54.643 -> Livebeat.
19:02:55.636 -> Livebeat.
19:02:56.663 -> Livebeat.
19:02:57.657 -> Livebeat.
19:02:58.517 -> New Telnet client connected to session 1
19:02:58.650 -> Livebeat.
19:02:59.577 -> pm open,type:2 0
19:02:59.644 -> Livebeat.
19:03:00.671 -> Livebeat.
19:03:01.664 -> Livebeat.
19:03:02.658 -> Livebeat.
19:03:03.651 -> Livebeat.
19:03:04.644 -> Livebeat.
19:03:05.671 -> Livebeat.
<----- AP MYSSID is shut down
19:03:12.162 -> bcn_timout,ap_probe_send_start
19:03:14.647 -> ap_probe_send over, rest wifi status to disassoc
19:03:14.647 -> state: 5 -> 0 (1)
19:03:14.647 -> rm 0
19:03:14.647 -> pm close 7
19:03:14.978 -> Livebeat.
19:03:14.978 -> WIFI disconnected.
19:03:15.971 -> Livebeat.
19:03:16.965 -> Livebeat.
19:03:17.693 -> scandone
19:03:17.694 -> no MYSSID found, reconnect after 1s
19:03:17.793 -> reconnect
19:03:17.959 -> Livebeat.
19:03:18.952 -> Livebeat.
19:03:19.979 -> Livebeat.
19:03:20.674 -> scandone
19:03:20.674 -> no MYSSID found, reconnect after 1s
...
----> restart AP MYSSID
...
19:04:28.975 -> no MYSID found, reconnect after 1s
19:04:29.041 -> Livebeat.
19:04:29.074 -> reconnect
19:04:30.035 -> Livebeat.
19:04:31.029 -> Livebeat.
19:04:31.957 -> scandone
19:04:31.957 -> state: 0 -> 2 (b0)
19:04:31.990 -> state: 2 -> 3 (0)
19:04:31.990 -> state: 3 -> 5 (10)
19:04:31.990 -> add 0
19:04:31.990 -> aid 1
19:04:31.990 -> cnt 
19:04:32.023 -> 
19:04:32.023 -> connected with MYSSID, channel 1
19:04:32.023 -> dhcp client start...
19:04:32.023 -> Livebeat.
19:04:32.089 -> ip:10.0.1.225,mask:255.255.255.0,gw:10.0.1.254
19:04:32.089 -> WIFI connected.
19:04:33.050 -> Livebeat.
19:04:34.044 -> Livebeat.
19:04:35.038 -> Livebeat.
19:04:36.032 -> Livebeat.
19:04:37.059 -> Livebeat.
19:04:38.052 -> Livebeat.
19:04:39.045 -> Livebeat.
19:04:40.039 -> Livebeat.
19:04:41.065 -> Livebeat.
```
(previous telnet session is of course broken, new ones work)

devyte, 2018-11-28:
Closing due to already fixed.

closed 2018-11-28 by devyte

kherNI, 2018-12-10:
@devte I look to be experiencing the same issue but I'm using Arduino IDE's Board Manager to manage version control.  Could you advise which library files were updated to correct this?  Thanks.

devyte, 2018-12-11:
A whole lot of them.
You can try latest git (install instructions are in the repo readme and in readthedocs, don't forget to uninstall from IDE before installing git), or wait a day or two until we finish release 2.5.0.