esp8266/Arduino #5383: ESP8266  web server stops responding after sending sensor data

BlackBlood007, 2018-11-28:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", is related to 3rd party libs, or is related to hardware, please discuss at a community forum like esp8266.com.

INSTRUCTIONS
============
If you do not follow these instructions, your issue may be dismissed.

1. Follow the checklist under Basic Infos and fill in the [ ] spaces with an X.
2. Fill in all the fields under Platform and Settings in IDE marked with [ ] (pick the correct option for you in each case, delete the others).
3. If you haven't already done so, test your issue against current master branch (aka latest git), because it may have been already fixed.
4. Describe your problem.
5. If you have a STACK DUMP decode it:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/stack_dump.html

6. Include a Minimal Complete Reproducible Example sketch that shows your issue. Do not include your entire project, or a huge piece of code.
7. Include debug messages:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/debugging.html 

8. Use markup (buttons above) and the Preview tab to check what the issue will look like.
9. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

### Basic Infos

- [ ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [ ] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12|ESP-01|ESP-07|ESP8285 device|other]
- Core Version:			[latest git hash or date]
- Development Env:		[Arduino IDE|Platformio|Make|other]
- Operating System:		[Windows|Ubuntu|MacOS]

### Settings in IDE

- Module:  			[Generic ESP8266 Module|Wemos D1 mini r2|Nodemcu|other]
- Flash Mode:			[qio|dio|other]
- Flash Size: 		[4MB/1MB]
- lwip Variant:   [v1.4|v2 Lower Memory|Higher Bandwidth]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz|160MHz]
- Upload Using:		[OTA|SERIAL]
- Upload Speed:   [115200|other] (serial upload only)

### Problem Description

Detailed problem description goes here.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp#include <ESP8266WiFi.h> 
#include <WiFiClient.h> 
#include <ESP8266WebServer.h>
#include <ESP8266mDNS.h>
#include<dht.h>

//multiplexing
//int x=D3;
int s0 = D0;
int s1 = D1;
int s2 = D2;
int s3 = D6;
int sig=D5;

//Temperature 
dht DHT;
int d=2;

//LIGHT
//analog pin
int ldr=A0;
//D3
int led=0;

//PIR SENSOR
//D8
int pir=15;
//D7
int pirled=13;
String sensordata="";
ESP8266WebServer server(80);

String page="";

void handleClient()
{
String message="";
    message+="number of args  ";
message+=server.args();
message+=" "+server.argName(0)+" "+server.arg(0);
  // server.send(200, "text/plain",message);  
  
   if(server.arg(0).equals("on"))
   {
    
digitalWrite(s0,HIGH);
digitalWrite(s1,HIGH);
digitalWrite(s2,HIGH);
digitalWrite(s3, HIGH);
digitalWrite(sig,LOW);
//server.send(200, "text/plain","on"); 
   }
   else if(server.arg(0).equals("off"))
   {
    
digitalWrite(s0,LOW);
digitalWrite(s1,LOW);
digitalWrite(s2,LOW);
digitalWrite(s3,LOW);
digitalWrite(sig,LOW);
//server.send(200, "text/plain","off"); 
   }
}
void setup() 
{
  Serial.begin(9600);
  //WiFi.begin("sanghi@19","9165269100@19699");
WiFi.begin("Satyam","hell1234");
  delay(50);
  while(WiFi.status() != WL_CONNECTED)
  {
    delay(500);
    Serial.println(".");
  }
  
   Serial.println("");
  Serial.print("Connected to ");
  Serial.println("sanghi@19");
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());

  //LIGHT
  
  pinMode(led,OUTPUT);
  pinMode(ldr,INPUT);

  //PIR
  pinMode(pir,INPUT);
  pinMode(pirled,OUTPUT);

  //multiplexing
  pinMode(s0, OUTPUT);
  pinMode(s1, OUTPUT);
  pinMode(s2, OUTPUT);
  pinMode(s3, OUTPUT);
  pinMode(sig,OUTPUT);
  digitalWrite(s0, LOW);
  digitalWrite(s1, LOW);
  digitalWrite(s2, LOW);
  digitalWrite(s3, LOW);
  digitalWrite(sig,LOW);

  if (MDNS.begin("esp8266")) {
    Serial.println("MDNS responder started");
  }
 server.on("/", handleClient);
   server.on("/sensordata", [](){
   
  server.send(200, "text/plain",sensordata);
  });
   
  server.begin();
   Serial.println("HTTP server started");
  delay(3000);
pinMode(s0, OUTPUT);
pinMode(s1, OUTPUT);
pinMode(s2, OUTPUT);
pinMode(s3, OUTPUT);
pinMode(sig,OUTPUT);
digitalWrite(s0, LOW);
digitalWrite(s1, LOW);
digitalWrite(s2, LOW);
digitalWrite(s3, LOW);
digitalWrite(sig,LOW);
}

void loop() {

 server.handleClient();
  int chk = DHT.read11(d);
 // Serial.print("Temperature = ");
 // Serial.println(DHT.temperature);
 // Serial.print("Humidity = ");
 // Serial.println(DHT.humidity);

 // sensordata+="<br>   temperature : "+String(DHT.temperature)+"  humidity :  "+String(DHT.humidity);
 sensordata+=",temperature : "+String(DHT.temperature)+"  humidity :  "+String(DHT.humidity);
 // delay(500);
///// LIGHT  ////////////////////////////////// 

 // Serial.print("Light Intensity :");
 // Serial.println(analogRead(ldr));
int reading=analogRead(ldr);
sensordata+="    Light Intensity :  "+String(reading);
if(reading<35)
{
  digitalWrite(led,HIGH);
}
else
{
  digitalWrite(led,LOW);
}
 // delay(500);

//////// PIR SENSOR /////////////////////////////////

String pirstatus;
if(digitalRead(pir)==HIGH)
{
  pirstatus="  detected motion";
 // Serial.println("detected motion");
  digitalWrite(pirled,HIGH);
//delay(50);
}
else{
  pirstatus="  motion not detected";
 // Serial.println("not detected");
  //delay(50);
digitalWrite(pirled,LOW);
}
sensordata+="    PIR value : "+pirstatus;
 delay(500);

/////////////////////////////////////////////////
delay(500);
  
}

d-a-v, 2018-11-28:
Your MCVE is not working, sensordata is filling memory, causing an uncaught exception.
Even with this fixed, it still does not work even once. Please test it, update it.
I am currently making requests in a loop running the example ESP8266WebServer>Helloserver and this one works flawlessly.
Please restart your app using this example as a basis.

devyte, 2018-11-28:
@BlackBlood007 said (with commented lines removed):
```
void loop() {

 server.handleClient();
 int chk = DHT.read11(d);
 sensordata+=",temperature : "+String(DHT.temperature)+"  humidity :  "+String(DHT.humidity);
...
```
sensordata is a String that will keep growing and growing, because it's a global and is never being reset.
Closing due to user error.

closed 2018-11-28 by devyte

d-a-v, 2018-11-28:
Also, the unconditional `delay(500)` in the main loop where `server.handleClient()` is called is killing the webserver.
There is absolutely no need of any unconditional `delay()` in the main loop.

BlackBlood007, 2018-11-29:
> @BlackBlood007 said (with commented lines removed):
>
> ```
> void loop() {
>
>  server.handleClient();
>  int chk = DHT.read11(d);
>  sensordata+=",temperature : "+String(DHT.temperature)+"  humidity : [...]

> @BlackBlood007 said (with commented lines removed):
>
> ```
> void loop() {
>
>  server.handleClient();
>  int chk = DHT.read11(d);
>  sensordata+=",temperature : "+String(DHT.temperature)+"  humidity : [...]
@d-a-v  Thank you so much sir for going through my code and making it bug free . After making the changes you told , this works very smoothly .