esp8266/Arduino #5385: Just launching  wifi.status(); freezes Wemos D1 mini

TomAvecLeVdsl, 2018-11-28:
### Basic Infos

- [ ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [ ] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:					[other]
- Core Version:			[Arduino IDE Plugin]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows & MacOS]

### Settings in IDE

- Module:  			[Wemos D1 mini r2]
- Flash Mode:			[Unknown]
- Flash Size: 		[4MB]
- lwip Variant:   [Unknown]
- Reset Method:		[ck]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[USB]
- Upload Speed:   [115200] (serial upload only)

### Problem Description

I can not get my ESP8266-01 module working
Just launching 
wifi.status(); 
Freezes it.
I added a cap on the power supply.

Some ideas ? 

Thanks for replying

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp
#include <SPI.h>
#include <WiFi.h>

void setup() {
  // initialize serial and wait for the port to open:
  pinMode(LED_BUILTIN, OUTPUT);
  Serial.begin(9600);
  while(!Serial) ;
}

void loop() {
  digitalWrite(LED_BUILTIN, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(1000);                       // wait for a second
  digitalWrite(LED_BUILTIN, LOW);
  delay(500);  // turn the LED off by making the voltage LOW
  Serial.println("avant status");
  Serial.print(WiFi.status());
  Serial.println("apres status");
  digitalWrite(LED_BUILTIN, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(1000);                       // wait for a second
  digitalWrite(LED_BUILTIN, LOW); 
  delay(500);
  digitalWrite(LED_BUILTIN, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(1000);                       // wait for a second
  digitalWrite(LED_BUILTIN, LOW); 
  delay(2000);
}

```

### Debug Messages

```
ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v6000001c
~ld
```

d-a-v, 2018-11-28:
You don't have `avant status` in your logs, so you can't be sure that's the cause.
Try with enabling debug mode, using the `Tools>Debug Port` and `Tools>Debug Level`,
and also `Tools>Erase Flash>All Flash Content` (this one just once, not every time).

TomAvecLeVdsl, 2018-11-28:
now i have 
`
⸮avant status

 ets Jan  8 2013,rst cause:4, boot mode:(1,4)

wdt reset
`
Nothing after that

d-a-v, 2018-11-28:
> i dont have Tools>Debug Port on French ardiuno GUI or Tools>Debug Level or also Tools>Erase Flash>All Flash Content

Did you select an esp8266 board in your french version of the IDE ?

TomAvecLeVdsl, 2018-11-29:
Yep

d-a-v, 2018-11-29:
So if you have an esp8266-based board and if your IDE menus are in french, then you don't have a `Tools>` menu, but an `Outils>` menu which is the french translation of Tools.

Now, you would have to test examples, that you will find under the `File>Examples` menu  (which is "`Fichier>Exemples`" in french) then you need to get examples from your selected board, not examples for all boards. For example "`Fichier>Exemples>ESP8266>Blink`" and "`Fichier>Exemples>ESP8266WiFi>WiFiClient`".

Your sketch cannot work on esp8266:
`#include <WiFi.h>` **must** be changed to `#include <ESP8266WiFi.h>`

TomAvecLeVdsl, 2018-11-29:
thanks , i will test that :) 👍

devyte, 2018-11-29:
Closing due to user error.