esp8266/Arduino #5390: Strange compiling errors since change to git version.

crndn2018, 2018-11-29:
### Basic Infos

- [ ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:			        [ESP-12]
- Core Version:			[e7d3cf6]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Generic ESP8266 Module]
- Flash Mode:			[qio]
- Flash Size: 		        [4MB]
- lwip Variant:                  [v2 Lower Memory]
- Reset Method:		[ck]
- Flash Frequency:	        [40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:              [115200] (serial upload only)

### Problem Description

After i have changed from Core 2.4.2 to latest git version (2.5.0-dev, Commit e7d3cf6) my project does not compile. I get many errors related to private libraries (e.g. W5500 library for ESP) but also related to ESP internal libraries.
Unfortunately i cannot provide any source code because its a professional iot project for my company.
Please have a look a the following error messages. The most error messages concern classes containing virtual methods.  All classes with virtual methods now dealed as abstract classes and cannot be instanced.
A problem i was not faced with Core 2.4.2. Additionally i get conflicting declarations.

### Debug Messages

Arduino: 1.8.5 (Windows 7), Board: "Generic ESP8266 Module, 80 MHz, Flash, ck, 26 MHz, 40MHz, QIO, 4M (1M SPIFFS), 2, v2 Lower Memory, Disabled, None, Only Sketch, 115200"

In file included from C:\Users\CD\Documents\Arduino\SmartEnergyMeter\2018-11-29\SmartEnergyMeter\SmartEnergyMeter.ino:232:0:

C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266/tools/sdk/include/espconn.h:32:15: error: conflicting declaration 'typedef sint8 err_t'

 typedef sint8 err_t;

               ^

In file included from C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266/tools/sdk/lwip2/include/lwip/netif.h:44:0,

                 from C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\cores\esp8266/AddrList.h:79,

                 from C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\cores\esp8266/IPAddress.h:196,

                 from C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266WiFi\src/ESP8266WiFi.h:31,

                 from C:\Users\CD\Documents\Arduino\SmartEnergyMeter\2018-11-29\SmartEnergyMeter\SmartEnergyMeter.ino:221:

C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266/tools/sdk/lwip2/include/lwip/err.h:94:20: error: 'err_t' has a previous declaration as 'typedef s32_t err_t'

 typedef LWIP_ERR_T err_t;

                    ^

In file included from C:\Users\CD\Documents\Arduino\libraries\EthernetESP\src/EthernetESP.h:12:0,

                 from C:\Users\CD\Documents\Arduino\SmartEnergyMeter\2018-11-29\SmartEnergyMeter\SmartEnergyMeter.ino:240:

C:\Users\CD\Documents\Arduino\libraries\EthernetESP\src/EthernetServer.h:18:18: error: invalid abstract return type for member function 'EthernetClient EthernetServer::available()'

   EthernetClient available();
[... 190 lines cut ...]

C:\Users\CD\Documents\Arduino\libraries\EthernetESP\src/EthernetClient.h:8:7: note:   since type 'EthernetClient' has pure virtual functions

 class EthernetClient : public Client {

       ^

SmartEnergyMeter:5550: error: cannot declare variable 'ecClient' to be of abstract type 'EthernetClient'

       EthernetClient ecClient = esEthernetStreamingServer.available(i);

                      ^

In file included from C:\Users\CD\Documents\Arduino\libraries\EthernetESP\src/EthernetESP.h:11:0,

                 from C:\Users\CD\Documents\Arduino\SmartEnergyMeter\2018-11-29\SmartEnergyMeter\SmartEnergyMeter.ino:240:

C:\Users\CD\Documents\Arduino\libraries\EthernetESP\src/EthernetClient.h:8:7: note:   since type 'EthernetClient' has pure virtual functions

 class EthernetClient : public Client {

       ^

exit status 1
invalid abstract type 'EthernetClient' for 'ecEthernetClient'

d-a-v, 2018-11-29:
This is related to latest version of lwIP-2.1.2.
I can't reproduce with Ethernet examples.
Are you using espconn ?
Can you try ethernet examples on your side and report ?

d-a-v, 2018-11-29:
Try with this first
```
#include <../../lwip2/include/lwip-err-t.h>
```
This should work with both lwIP-v1.4 and lwIP-v2.

This file is autogenerated and extract `err_t` from espressif lwip-v1.4 sources.

d-a-v, 2018-11-29:
What about modifying `tools/sdk/include/espconn.h:32:15`, by replacing
```
typedef sint8 err_t;
```
by
```
//typedef sint8 err_t;
#include <../../lwip2/include/lwip-err-t.h>
```

crndn2018, 2018-11-29:
Conflicting declaration vanished. 

Problems with virtual methods remaining. Something changed in the toolchain ?

d-a-v, 2018-11-29:
> Conflicting declaration vanished.

Thanks, I will commit this fix

> Problems with virtual methods remaining. Something changed in the toolchain ?

Not with the toolchain, but mainly with `IPAddess`.
You were saying that you have a modified version of W5100 to allow W5500. 
Here are the changes to the core:  https://github.com/esp8266/Arduino/pull/5136/files
You also have another [way](https://github.com/d-a-v/W5500lwIP) to use your W5500 but it is still beta.

crndn2018, 2018-11-29:
The errors related to sections we did not change. E.g. Class "EthernetClient" contains virtual methods since beginning. And what's about the following. These are ESP internal libraries. The alternative W5500 library i have already checked. That's not enough for us.
I am a little bit confused. If i change to 2.4.2 all works fine. I have repeated the installtion process for git version on windows three times. No change. By the way, this not really documented. Fished it from a discussion board.  Perhaps you find an answer to the open questions in the next days. 
But first, thanks for your support ! 

C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\cores\esp8266/Udp.h:81:27: note: 	virtual IPAddress UDP::remoteIP() const

         virtual IPAddress remoteIP() const =0;

                           ^

C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\cores\esp8266/Udp.h:83:26: note: 	virtual uint16_t UDP::remotePort() const

         virtual uint16_t remotePort() const =0;

devyte, 2018-11-29:
@crndn2018 said:
> The alternative W5500 library i have already checked. That's not enough for us.

Can you please provide details as to why? This is important. We're moving towards a unified tcp/ip stack that includes the Ethernet class, and could potentially include other interfaces in the future. There are plans for after this is finalized.

crndn2018, 2018-11-29:
Looking for interfaces like WifiClient/WifiServer.

d-a-v, 2018-11-29:
> E.g. Class "EthernetClient" contains virtual methods since beginning

Signatures recently changed, I provided the link with all changes.
```
virtual IPAddress remoteIP() const =0;
virtual uint16_t remotePort() const =0;
```
Both are working in current git version.

> Looking for interfaces like WifiClient/WifiServer.

That is the point of this new library.

devyte, 2018-11-29:
In fact, in @d-a-v 's [alternate Ethernet](https://github.com/d-a-v/W5500lwIP) implementation EthernetClient would essentially be replaced, or aliased to, WiFiClient. Same for WiFiServer.
The plus side: you get ssl, http client, web server, and all the other goodies available for WiFi for free, or very close to it, on Ethernet.

crndn2018, 2018-11-29:
OK, sorry, i misunderstood this. Perhaps you add an basic example, how to use it. For donkeys, like me ;)

d-a-v, 2018-11-29:
It's WIP, check https://github.com/d-a-v/W5500lwIP/issues/3

crndn2018, 2018-11-30:
My version of the Ethernet-Library overloads the methods whose signatures have changed. Corrected. Running.
My last bug:

#Sketch
extern "C" 
{
  #include "user_interface.h" 
  #include "espconn.h" 
}

void setup() {
  espconn_tcp_set_max_con(8);
}

void loop() {
  // put your main code here, to run repeatedly:

}

#Debug messages
sketch\sketch_nov30a.ino.cpp.o:(.text.setup+0x0): undefined reference to `espconn_tcp_set_max_con'

sketch\sketch_nov30a.ino.cpp.o: In function `setup':

C:\Users\CD\Documents\Arduino\sketch_nov30a/sketch_nov30a.ino:8: undefined reference to `espconn_tcp_set_max_con'

d-a-v, 2018-11-30:
`espconn` is a specific espressif api shipped with lwIP-v1.4
(it's a modified version of lwIP 's `netconn` API).
Try selecting lwIP-v1.4 in Tools menu instead (lwIP-v2 is default since core-2.4).

I ported it to lwIP-v2 but you show it is incomplete. It is anyway far to be tested nor proven to be working. Please open a new issue for this specific request but I'm afraid that will not be a priority.
Maybe you could port your code to use our (arduino/esp8266) WiFi API ?

crndn2018, 2018-11-30:
Tried lwIP v1.4. Not working.
This worked lwIP v2 in Core 2.4.2. I will open a new issue.
Let's close this issue.

closed 2018-11-30 by crndn2018