esp8266/Arduino #5394: ESP8266 Arduino EEPROM can only use the first 4KiB flash data?

dportabella, 2018-11-30:
The standard arduino `EEPROM.write(address, value)` is a bit different on ESP8266. You need to specify the size of the "sector" and commit. [ESP8266 EEPROM][1]

    EEPROM.begin(size);
    EEPROM.write(addr, val);
    EEPROM.commit();

`addr` must be between `0` and `size`. [ESP8266 EEPROM.cpp L100-L101][2].

`EEPROM.begin` allocates that amount of memory into the RAM (even if you only need to write one byte!) and `size` must be between `0` and `4096`. [ESP8266 EEPROM.cpp L54-L65][3], [spi_flash_geometry.h][4].

If I understand correctly, ESP8266 does not have an internal flash memory. The module ESP-12E has an external 4 MiB flash (512 KiB). https://en.wikipedia.org/wiki/ESP8266

So, the ESP8266 EEPROM library only allows to use only the first 4KiB flash data. Why is this? How can I use the rest of the 512 KiB of the ESP-12 module? Or how can I use the standard Arduino EEPROM library?

  [1]: https://arduino-esp8266.readthedocs.io/en/latest/libraries.html#eeprom]
  [2]: https://github.com/esp8266/Arduino/blob/master/libraries/EEPROM/EEPROM.cpp#L100-L101
  [3]: https://github.com/esp8266/Arduino/blob/master/libraries/EEPROM/EEPROM.cpp#L54-L65
  [4]: https://github.com/esp8266/Arduino/blob/master/tools/sdk/include/spi_flash_geometry.h#L8-L13

d-a-v, 2018-11-30:
> EEPROM library uses one sector of flash located just after the SPIFFS.

https://arduino-esp8266.readthedocs.io/en/2.4.2/libraries.html#eeprom
https://arduino-esp8266.readthedocs.io/en/2.4.2/libraries.html#improved-eeprom-library-for-esp-esp-eeprom

> How can I use the rest of the 512 KiB of the ESP-12 module?

There is SPIFFS

https://arduino-esp8266.readthedocs.io/en/2.4.2/filesystem.html

In brief,

https://arduino-esp8266.readthedocs.io/en/2.4.2
arduino-esp8266.readthedocs.io
readthedocs

devyte, 2018-11-30:
The ESP's SoC doesn't have internal flash: correct.
4 Megabytes of flash on the ESP-12 board: correct
size can be any number up to 4096: correct (must be smaller than the size of one flash sector).

In theory you could have e.g. an array or std::vector of EEPROM objects, one per flash sector in use, with the current code. 
In practice, that would be a Very Bad Idea due to flash burn of specifics sectors when committing, because EEPROM doesn't do wear leveling.

If you need that much flash, use SPIFFS and files.

Closing due to not an issue. Please ask further questions at a community forum, this is an issue tracker and not meant for Q&A.

closed 2018-11-30 by devyte

dportabella, 2018-11-30:
Hi, thanks.
Then this is a feature request: allow to use the simple and convenient standard arduino EEPROM library.

What do you think about this feature request?

d-a-v, 2018-11-30:
> So, the ESP8266 EEPROM library only allows to use only the first 4KiB flash data. Why is this?

Not really the first 4KiB, it's location is reserved in specific address hidden to user.
4KiB because that's a fair size regarding EEPROM on AVR.
`commit()` to avoid wear leveling.

What you could do is your own EEPROM class, inheriting from EEPROMClass, automatically calling `begin()` in its constructor, and `commit()` in `write()` and `update()`. You could also do that using the other EEPROM class instead (link above), to reduce writes to flash.
This esp8266/arduino's EEPROM has been here for a long time. Changing it now would be breaking the current API.

dportabella, 2018-12-03:
> What you could do is your own EEPROM class, inheriting from EEPROMClass, 
> automatically calling begin() in its constructor, and commit() in write() and update().

I would still be limited to 4Kb maximum.

> You could also do that using the other EEPROM class instead (link above), to reduce writes to flash.

Which one? I do not see the link.

> This esp8266/arduino's EEPROM has been here for a long time. Changing it now would be breaking the current API.

I understand. What I propose is to make a new EEPROM class (such as EEPROM2) which ignores the file system, and use all the space currently used by the filesystem + the 4Kb. Without the begin/commit feature, which requires to have a copy of the content into RAM. Would that be difficult? hint how to achieve this?