esp8266/Arduino #5397: NTP Client fails in WIFI_AP_STA mode

RevPhil, 2018-11-30:
How do I successfully use a NTP Client when connected to a router and acting as an Access Point?

Here's the scenario, I connect to my router using WiFi.begin(ssid,password) (Gateway 192.168.0.1) and once connected I can issue a request through port 123 and get a response from a NTP Server. If I start up an AP using WiFi.softAP(ssid,password) (Gateway 192.168.4.1) the NTP Client no longer works and times out every time.

I can accept incoming HTTP requests via the router and the AP, if I connect to it, and respond to them perfectly. I can accept incoming UDP packets on the AP network and respond to them every time. I cannot Ping out via the router either.

It seems to me that there is a problem with the internal routing of the NTP Client requests and Ping attempts to the router network and they seem to go nowhere.

Am I trying to do something impossible? I don't think it's unreasonable to expect basic routing of packets based on the IP address but, I happy to be educated on this.

d-a-v, 2020-11-09:
Can you please close this issue because it is not anymore relevant for you,
or check if you still have the issue. Current version of this core should work well with your use case.
Probably lwIP's default interface was wrongly set back then, that prevented from properly routing NTP requests.

RevPhil, 2020-11-10:
Due to the lack of meaningful responses to this issue I have now closed it.

d-a-v, 2020-11-10:
Here's the example I conducted. I took NTP-TZ-DST example from head of git master branch, that I modified like this:

```
diff --git a/libraries/esp8266/examples/NTP-TZ-DST/NTP-TZ-DST.ino b/libraries/esp8266/examples/NTP-TZ-DST/NTP-TZ-DST.ino
index 376086fb..0cafa90b 100644
--- a/libraries/esp8266/examples/NTP-TZ-DST/NTP-TZ-DST.ino
+++ b/libraries/esp8266/examples/NTP-TZ-DST/NTP-TZ-DST.ino
@@ -233,7 +233,8 @@ void time_is_set(bool from_sntp /* <= this parameter is optional */) {
 
 void setup() {
   WiFi.persistent(false);
-  WiFi.mode(WIFI_OFF);
+  WiFi.mode(WIFI_AP_STA);
+  WiFi.softAP("espap","thereisnospoon");
 
   Serial.begin(115200);
   Serial.println("\nStarting in 2secs...\n");
@@ -271,7 +272,7 @@ void setup() {
   yield();
 
   // start network
-  WiFi.mode(WIFI_STA);
+  //WiFi.mode(WIFI_STA);
   WiFi.begin(STASSID, STAPSK);
 
   // don't wait for network, observe time changing
```
Then I run the example.
ESP start as being an AP and seconds later also connect as station.
All went fine and it showed London time so routing is currently OK when both AP and STA are up.

I'd like to apologize for my yesterday message, it should have been
> Can you please close this issue ~~because~~ ***if*** it is not anymore relevant for you,

because I explain there have been changes since you opened your issue, answering to this:

> > It seems to me that there is a problem with the internal routing of the NTP Client requests and Ping attempts to the router network and they seem to go nowhere.

If you still have an issue with release 2.7.4 or git master, please post your MCVE.

RevPhil, 2020-11-10:
d-a-v Thanks for your comments and example, however, your example appears to work but the AP doesn't actually start up as it is not visible on the list of available APs.

If I modify the code and put the AP setup 'after' the 'WiFi.begin(<ssid>,<password>)' statement, it works as expected and Lo and Behold. the NTP client now connects to the server.

Clearly, the core code has changed since I tried and failed to get this to work.

Thanks for your input!

closed 2020-11-15 by RevPhil