esp8266/Arduino #5398: Unable to set number of tcp connections via espconn.h.

crndn2018, 2018-11-30:
### Basic Infos

- [ ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [ ] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:				[ESP-12]
- Core Version:			[92373a9]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Generic ESP8266 Module]
- Flash Mode:			[qio]
- Flash Size: 		        [4MB]
- lwip Variant:                  [v2 Lower Memory]
- Reset Method:		[ck]
- Flash Frequency:	        [40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:              [115200|] 

### Problem Description

Sketch is self-explaining.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

extern "C"
{
#include "user_interface.h"
#include "espconn.h"
}

void setup() {
espconn_tcp_set_max_con(8);
}

void loop() {
// put your main code here, to run repeatedly:

}

### Debug Messages

sketch\sketch_nov30a.ino.cpp.o:(.text.setup+0x0): undefined reference to `espconn_tcp_set_max_con'

sketch\sketch_nov30a.ino.cpp.o: In function `setup':

C:\Users\CD\Documents\Arduino\sketch_nov30a/sketch_nov30a.ino:8: undefined reference to `espconn_tcp_set_max_con'

devyte, 2018-11-30:
I strongly advise against using the espconn interface. It has historically been very buggy, particularly when interfaces go up/down, or with weak wifi signal where the connection is intermittent. Either migrate your code to WiFiClient/Server, or use netconn directly. If it were up to me, I'd disallow espconn altogether.

crndn2018, 2018-12-04:
I use espconn_tcp_set_max_con() only in setup. No further relations to espconn. 
Default is 5. Need 8.
What is the alternative way ? Example code please.
By the way, worked in 2.4.2 !

crndn2018, 2018-12-04:
Was mentioned in some discussuion boards. If it has no effect, i will remove it. Thanks. Closed.

closed 2018-12-04 by crndn2018