esp8266/Arduino #5399: esp8266 will write anything to flash when WiFi.persistent(false)?

henry42, 2018-11-30:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:					ESP-12
- Core Version:			2.4.2
- Development Env:		Arduino IDE
- Operating System:		Windows

### Settings in IDE

- Module:  			Generic ESP8266 Module
- Flash Mode:			qio
- Flash Size: 		4MB
- lwip Variant:   v2 Lower Memory
- Reset Method:		ck
- Flash Frequency:	[40Mhz]
- CPU Frequency:		80Mhz
- Upload Using:		SERIAL
- Upload Speed:   115200

### Problem Description

My program is like this:  it starts up , connect the wifi, check data and go to sleep for 1 minute,  it has run for months normally. but now it may run but after about 1 day or less it will stop from running without any information and you can not reboot it unless you flash it again.

I think it may because the flash is corrupted, so I add `WiFi.persistent(false)`, but it does not help.

My questions is:

1. Does it write the wifi setting to flash even it is the same as the one stored( `WiFi.persistent(true)` )?
2. If the flash is corrupted, even I do not write to it, it will also fail somehow?
3. I write my code below, I think there is nowhere writing to the flash, right?
4. anything I can do to save it?

Thank you very much!

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <Arduino.h>

void setup() {
Serial.begin(115200, SERIAL_8N1, SERIAL_TX_ONLY);
WiFi.persistent(false); //I added this line when it is already broken
WiFi.begin ( ssid, password );
//do some wifi checks

  if (SPIFFS.begin()) {
//Read from the spifss
  }
}

void loop() {
//do something
//deep sleep for 1 minutes
}

```

devyte, 2018-11-30:
If you didn't change anything and it stopped working, then the flash is likely burned.
Consider this: 1 write/min * 60min/h * 24h/day * 365days/year = 525600 writes in one year to the same sector. I've seen ESP flash die after 1000 writes, some are really bad quality.
IMHO get a new board, they're cheap enough.
Closing due to not an issue.

closed 2018-11-30 by devyte

ortegafernando, 2019-05-29:
Hi, sorry for re-open it. 

@devyte the answer to this question: "Does it write the wifi setting to flash even it is the same as the one stored( WiFi.persistent(true) )?" is TRUE, isn't it ?

I think that we need only Wifi.persistent(true) when we want to erase the saved credentials.

What is the problem to use always Wifi.persistent(false) ??? ESP8266 will only write credentials if they are different, that is what 99,99% of users will like, won't they?

Thanks for your help.