esp8266/Arduino #5403: Finalize and add SD device test

devyte, 2018-11-30:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] I have filled out all fields below.

#### Platform

- Hardware:					All
- Core Version:			latest git
- Development Env:		All
- Operating System:		All

### Problem Description

From #4444 : Sketch below to test SD on device. 
Possibly add to devices test, similar to @d-a-v 's interactive WiFi test sketch, maybe add more actions.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

Steps:

Send m to mount
Send u and y to unmount
Send m to mount again, if SD.end() is commented, you are in a endless "SD initialization failed!" loop.

```cpp

File myFile;
bool mounted = false;

void setup() {
    Serial.begin(115200);
}

void loop() {
    if (mounted && Serial.peek() == 'u') {
        Serial.read();
        Serial.println("Unmount SD and press 'y' to continue");
        while (Serial.peek() != 'y') {
            delay(1000);
        }
        mounted = false;
        Serial.println("SD unmounted.");
        Serial.read();
    } else if (!mounted && Serial.peek() == 'm') {
        Serial.read();

        //Uncomment this to finish while (!SD.begin(D2)) endless loop
        //SD.end();

        while (!SD.begin(D2)) {
            Serial.println("SD initialization failed!");
            delay(1000);
        }
        mounted = true;
        Serial.println("Initialization succeded!");
    }

    if (mounted) {
        myFile = SD.open("test.txt");
        Serial.println("Printing test.txt contents:");
        while (myFile.available()) {
            Serial.write(myFile.read());
        }
        myFile.close();
        Serial.println();
        Serial.println("All contents of test.txt printed!");

        Serial.println("SD mounted, press 'u' to unmount");
    } else {
        Serial.println("SD unmounted, press 'm' to mount");

    }

    Serial.println("Wait 3 seconds...");
    delay(3000);
}
```

devyte, 2018-12-29:
@earlephilhower assigning to you as part of your ongoing FS work.

earlephilhower, 2019-01-14:
Link to #5525.  Not sure what exactly the request it, but we are doing host tests already, might want to add a "change SD card" host test as well.