esp8266/Arduino #5406: issues with IPv6 patch & IPAddress changes

d-a-v, 2018-12-01:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:				any
- Core Version:			latest git
- Development Env:		        any
- Operating System:		any

### Settings in IDE

- Module:  			any
- Flash Mode:			any
- Flash Size: 		        any
- lwip Variant:                   v2
- Reset Method:		any
- Flash Frequency:	        any
- CPU Frequency:		any
- Upload Using:		any
- Upload Speed:   any

### Problem Description

Track issues from IPv6 + IPAddress changes

ref: https://github.com/esp8266/Arduino/commit/5c4db3acf4353870b6a5663ba06d712d67e6c179#diff-d93063c7b382a500a221e8631d6d6902L123
@reaper7 

Identified issues are:
- use of `struct ip_addr` which is espressif and not lwIP. 
  This structure has been renamed to `struct ipv4_addr` within the SDK which is stuck to lwIP-v1.4.
  `struct ip_addr_t`is the real name of that structure, but in arduino `IPAddress` should always be used.
- use of `ipaddr->addr` is stuck to IPv4. It *could* be changed to `ip_2_ip4(ipaddr)->addr`
  with (this is defined in newer IPAddress.h but is needed for older cores):
```
#ifndef ip_2_ip4
// emulate lwIP-v2's ip_2_ip4(x) for lwIP-v1
#define ip_2_ip4(x) (x)
#endif
```
- more generally `IPAddress(ipaddr)` should be used anywhere, thus **generic, simple and portable wrappers** should be proposed for external libraries to keep compatibility with older cores, and IPv6 at the same time (IPv6 is costless when using IPAddress)
- `INADDR_NONE` should be restored as it was (sorry for that, `IPNoAddress` was an intermediary change that I forgot to revert)

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

TBD, using at least EspAsyncTCP library, thingspeak library

reaper7, 2018-12-01:
@d-a-v - tnx for open this issue :)

detailed compile errors for thingspeak->ReadField example:
```
In file included from C:\PROGRAMY\arduino_projekty\libraries\thingspeak-arduino\examples\ESP8266\ReadField\ReadField.ino:28:0:

C:\PROGRAMY\arduino_projekty\libraries\thingspeak-arduino\src/ThingSpeak.h:1729:23: error: 'INADDR_NONE' was not declared in this scope

  IPAddress customIP = INADDR_NONE;

                       ^

C:\PROGRAMY\arduino_projekty\libraries\thingspeak-arduino\src/ThingSpeak.h: In member function 'void ThingSpeakClass::setServer(const char*, unsigned int)':

C:\PROGRAMY\arduino_projekty\libraries\thingspeak-arduino\src/ThingSpeak.h:1700:20: error: 'INADDR_NONE' was not declared in this scope

   this->customIP = INADDR_NONE;

                    ^

C:\PROGRAMY\arduino_projekty\libraries\thingspeak-arduino\src/ThingSpeak.h: In member function 'void ThingSpeakClass::setServer()':

C:\PROGRAMY\arduino_projekty\libraries\thingspeak-arduino\src/ThingSpeak.h:1720:20: error: 'INADDR_NONE' was not declared in this scope

   this->customIP = INADDR_NONE;

                    ^

C:\PROGRAMY\arduino_projekty\libraries\thingspeak-arduino\src/ThingSpeak.h: In member function 'bool ThingSpeakClass::connectThingSpeak()':

C:\PROGRAMY\arduino_projekty\libraries\thingspeak-arduino\src/ThingSpeak.h:1744:24: error: 'INADDR_NONE' was not declared in this scope

   if(this->customIP == INADDR_NONE && NULL == this->customHostName)

                        ^

exit status 1
Błąd kompilacji dla płytki LOLIN(WEMOS) D1 R2 & mini.
```

d-a-v, 2018-12-01:
#5409 will fixe these issues.

> more generally IPAddress(ipaddr) should be used anywhere, thus generic, simple and portable wrappers should be proposed for external libraries to keep compatibility with older cores, and IPv6 at the same time (IPv6 is costless when using IPAddress)

There is no hack in #5409 thus no wrappers are needed (yet, and that's nice).
However, when IPv6 is enabled, updates are needed: Use `IPAddress` everywhere where `struct ip_addr` or `struct ip_addr_t` or `u32_t` or `uint32_t` are used, and this will work too with lwIP-v4 and this core (git/pre-2.5.0).

devyte, 2018-12-02:
Fixed via #5409 . Please open a new issue if there are further compile problems.

d-a-v, 2018-12-02:
It is not fixed yet,
yesterday I fell asleep on my keyboard :-]

d-a-v, 2018-12-03:
Examples fromEspAsyncTCP, ThingSpeak and Syslog now compile fine with #5416.

devyte, 2018-12-03:
Closing via #5416 .

closed 2018-12-03 by devyte