esp8266/Arduino #5408: Missing setFingerprint on WiFiClientSecure

edoardoo, 2018-12-01:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12E]
- Core Version:			[latest git hash or date]
- Development Env:		[Arduino IDE]
- Operating System:		[MacOS]

### Settings in IDE

- Module:  			[NodeMCU 1.0]
- Flash Mode:			[qio|dio|other]
- Flash Size: 		[4MB/1MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz|160MHz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] (serial upload only)

### Problem Description

The example sketch for WiFiClientSecure fails to compile.
`class axTLS::WiFiClientSecure' has no member named 'setFingerprint'`

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp
/*
    HTTP over TLS (HTTPS) example sketch

    This example demonstrates how to use
    WiFiClientSecure class to access HTTPS API.
    We fetch and display the status of
    esp8266/Arduino project continuous integration
    build.

    Limitations:
      only RSA certificates
      no support of Perfect Forward Secrecy (PFS)
      TLSv1.2 is supported since version 2.4.0-rc1

    Created by Ivan Grokhotkov, 2015.
    This example is in public domain.
*/

#include <ESP8266WiFi.h>
#include <WiFiClientSecure.h>

#ifndef STASSID
#define STASSID "your-ssid"
#define STAPSK  "your-password"
#endif

const char* ssid = STASSID;
const char* password = STAPSK;

const char* host = "api.github.com";
const int httpsPort = 443;

// Use web browser to view and copy
// SHA1 fingerprint of the certificate
const char fingerprint[] PROGMEM = "5F F1 60 31 09 04 3E F2 90 D2 B0 8A 50 38 04 E8 37 9F BC 76";

void setup() {
  Serial.begin(115200);
  Serial.println();
  Serial.print("connecting to ");
  Serial.println(ssid);
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());

  // Use WiFiClientSecure class to create TLS connection
  WiFiClientSecure client;
  Serial.print("connecting to ");
  Serial.println(host);

  Serial.printf("Using fingerprint '%s'\n", fingerprint);
  client.setFingerprint(fingerprint);

  if (!client.connect(host, httpsPort)) {
    Serial.println("connection failed");
    return;
  }

  String url = "/repos/esp8266/Arduino/commits/master/status";
  Serial.print("requesting URL: ");
  Serial.println(url);

  client.print(String("GET ") + url + " HTTP/1.1\r\n" +
               "Host: " + host + "\r\n" +
               "User-Agent: BuildFailureDetectorESP8266\r\n" +
               "Connection: close\r\n\r\n");

  Serial.println("request sent");
  while (client.connected()) {
    String line = client.readStringUntil('\n');
    if (line == "\r") {
      Serial.println("headers received");
      break;
    }
  }
  String line = client.readStringUntil('\n');
  if (line.startsWith("{\"state\":\"success\"")) {
    Serial.println("esp8266/Arduino CI successfull!");
  } else {
    Serial.println("esp8266/Arduino CI has failed");
  }
  Serial.println("reply was:");
  Serial.println("==========");
  Serial.println(line);
  Serial.println("==========");
  Serial.println("closing connection");
}

void loop() {
}

```

### Debug Messages

```

/Users/edo/Documents/Arduino/sketch_dec01b/sketch_dec01b.ino: In function 'void setup()':
sketch_dec01b:59:10: error: 'class axTLS::WiFiClientSecure' has no member named 'setFingerprint'
   client.setFingerprint(fingerprint);
          ^
exit status 1
'class axTLS::WiFiClientSecure' has no member named 'setFingerprint'

```

devyte, 2018-12-02:
The provided example is [this one](https://github.com/esp8266/Arduino/blob/92373a98370618dea09718010b30d311a97f3f25/libraries/ESP8266WiFi/examples/HTTPSRequest/HTTPSRequest.ino).
To me, this looks like trying to build the current version of the example with an older version of the core. That won't work, the WiFiClientSecure defaut has changed from AxTLS to BearSSL. 

I can't reproduce this build error. Also, if this didn't build, CI would be screaming.
Closing due to user error.

closed 2018-12-02 by devyte

edoardoo, 2018-12-02:
Sorry for the mistake. My bad.
I've tried to compile this sketch using the library available for Arduino using the boards manager (which is 2.4).  
I wrongly supposed that the github/master and the Arduino repo versions were aligned. 
Thanks for clearing this out.

HBEngineer, 2019-07-01:
> Sorry for the mistake. My bad.
> I've tried to compile this sketch using the library available for Arduino using the boards manager (which is 2.4).
> I wrongly supposed that the github/master and the Arduino repo versions were aligned.
> Thanks for clearing this out.

Sorry..... so, how did you resolve the problems?  What did you install/uninstall?

earlephilhower, 2019-07-01:
Do not install any libraries for WiFi from the board manager.  That's only for the Arduino.cc boards, not the 8266.  The 8266 has all the WiFi libraries already integrated in the core.

aji-web, 2020-03-04:
C:\Users\Aji Pamungkas\Documents\Arduino\libraries\WebSockets\src\WebSocketsClient.cpp: In member function 'void WebSocketsClient::loop()':

C:\Users\Aji Pamungkas\Documents\Arduino\libraries\WebSockets\src\WebSocketsClient.cpp:172:30: error: 'class WiFiClientSecure' has no member named 'setCACert'

                 _client.ssl->setCACert((const uint8_t *)_CA_cert, strlen(_CA_cert) + 1);

                              ^

exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

i got this error.. please how to resolve this problem??

d-a-v, 2020-03-04:
@aji-web This is a closed issue.
You seem to be trying to use AxTLS which is deprecated and replaced by BearSSL.
Please check the [SSL examples](https://github.com/esp8266/Arduino/tree/master/libraries/ESP8266WiFi/examples) out. These examples are ready to use from within the Arduino IDE file menu.