esp8266/Arduino #5413: Can't ever connect to WiFi when USB connected

teo1978, 2018-12-02:
### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [X] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below.

#### Platform

- Hardware:					[SparkFun ESP8266 Thing Dev]
- Core Version:			[latest git]
- Development Env:		[Arduino IDE|]
- Operating System:		[Ubuntu]

### Settings in IDE

- Module:  			[SparkFun ESP8266 Thing Dev]
- Flash Mode:			??
- Flash Size: 		512K no SPIFFS (is what was selected by default in the Arduino IDE, no freaking idea what this is)
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		??
- Flash Frequency:	??
- CPU Frequency:		80 MHz
- Upload Using:		[SERIAL]
- Upload Speed:   [other] (serial upload only)

### Problem Description

Detailed problem description goes here.

I have a sketch (see code below) that **used to work perfectly** fine both when connected via USB and when powered from external power supplies, which is 3 AAA batteries.

Now, when powered from batteries it works fine, but **when powered from USB it systematically cannot connect to Wifi** (to the exact same wifi network, of course).

I haven't touched my code between when it used to work and when it stopped working, and I'm testing it on the exact same board with the exact same settings in the exact same environment (i.e. physical place and the wifi network it tries to connect to). And I don't even think I have updated the version of the esp8266 arduino core, so in theory I **haven't touched anything that should explain why it stopped working**. The only thing I did was install the board definitions for ESP32 boards (without attempting to use them, as this is NOT an ESP32 board), then the sketch wouldn't compile anymore (I know, it doesn't make any sense: I installed a new set of boards but I didn't select any of them: I kept trying to compile the sketch for the same old board, but it wouldn't compile anymore; something is seriously fucked up at https://github.com/espressif/arduino-esp32). So I uninstalled that board and now here I am. So, this should be irrelevant as everything is (supposed to be) back to how it was before I even tried to install the ESP32 boards (without ever using them). 

The sketch now compiles again and generally works, but when connected to USB it exhibits the issue at hand. My guess is that the whole install-Esp32-boards-and-uninstall-them had absolutely nothing to do and this is just a random issue that for some random reason has started to show up now. 

I don't know how to interpret the debug output, or what the messages like "state X -> Y", "scandone", and "reconnect" are supposed to mean, but to me those "reconnect" messages that show up several times look very suspicious. "Reconnect"?? You never even managed to connect, so why reconnect?

BTW are those debug messages documented anywhere? They are very cryptic.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

WiFi.mode(WIFI_STA);
  WiFi.begin(client_ssid.c_str(), client_password.c_str());

  for (int i=0; i<20; i++) {
    Serial.printf("Trying to connect to Wifi '%s' with password '%s'\n", client_ssid.c_str(), client_password.c_str());
    if (WiFi.status()==WL_CONNECTED) {
      Serial.printf("...connected succesfully\n");
  }
  Serial.printf("Cannot connect to WiFi!\n");
```

### Debug Messages

```
mode : sta(60:01:94:3f:eb:a9)
add if0
Trying to connect to Wifi 'Sixty-Seven' with password '*******'
Trying to connect to Wifi 'Sixty-Seven' with password '*******'
Trying to connect to Wifi 'Sixty-Seven' with password '*******'
scandone
state: 0 -> 2 (b0)
Trying to connect to Wifi 'Sixty-Seven' with password '*******'
state: 2 -> 0 (2)
reconnect
Trying to connect to Wifi 'Sixty-Seven' with password '*******'
Trying to connect to Wifi 'Sixty-Seven' with password '*******'
Trying to connect to Wifi 'Sixty-Seven' with password '*******'
scandone
state: 0 -> 2 (b0)
state: 2 -> 0 (2)
reconnect
Trying to connect to Wifi 'Sixty-Seven' with password '*******'
Trying to connect to Wifi 'Sixty-Seven' with password '*******'
Trying to connect to Wifi 'Sixty-Seven' with password '*******'
scandone
state: 0 -> 2 (b0)
Trying to connect to Wifi 'Sixty-Seven' with password '*******'
state: 2 -> 0 (2)
reconnect
Trying to connect to Wifi 'Sixty-Seven' with password '*******'
Trying to connect to Wifi 'Sixty-Seven' with password '*******'
Trying to connect to Wifi 'Sixty-Seven' with password '*******'
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 0 (4)
reconnect
Trying to connect to Wifi 'Sixty-Seven' with password '*******'
Trying to connect to Wifi 'Sixty-Seven' with password '*******'
Trying to connect to Wifi 'Sixty-Seven' with password '*******'
scandone
state: 0 -> 2 (b0)
Trying to connect to Wifi 'Sixty-Seven' with password '*******'
state: 2 -> 0 (2)
reconnect
Trying to connect to Wifi 'Sixty-Seven' with password '*******'
Trying to connect to Wifi 'Sixty-Seven' with password '*******'
Cannot connect to WiFi!
scandone
state: 0 -> 2 (b0)
state: 2 -> 0 (2)
reconnect
```

d-a-v, 2018-12-02:
Please print `Esp.getFullVersion()`

teo1978, 2018-12-02:
Sorry, I left that out accidentally:
```
SDK:2.2.1(cfd48f3)/Core:2.4.2-57-g270788b/lwIP:2.0.3(STABLE-2_0_3_RELEASE/glue:arduino-2.4.1-13-g163bb82)/BearSSL:f55a6ad
```

d-a-v, 2018-12-02:
Can you flash with option `Tools>Erase Flash>All Contents` (only once, put it back to "only sketch" after that) ?

teo1978, 2018-12-02:
> Still observing identical behavior.

Actually, no, it's worse now: now it can't connect even when powered from external supply (of course in that case I can't monitor the serial output and know whether the issue is the same).

I don't know whether this has to do with having erased all flash contents or if it is just random, because even before, it did sometimes fail to connect even when powered by external supply, but only very sporadically (and back when everything worked as expected, it would very sporadically fail to connect whether powered via USB or external supply, regardless).

teo1978, 2018-12-02:
BTW I have updated the code snipped in the report (not my code, that's been the same for a very long time). I had left out the first two lines:
```
WiFi.mode(WIFI_OFF);      // this
WiFi.persistent(false);   // and this
WiFi.mode(WIFI_STA);
WiFi.begin(client_ssid.c_str(), client_password.c_str());
```

Anyway, as I said, this all had always worked fine until today.

(The reason I turn off wifi and persistent at the beginning is that this is actually simplified code from a sketch that manages storing SSID and password to the EEPROM on its own, and can run in AP mode to allow setting up the network from a client)

d-a-v, 2018-12-02:
Do you have another module to try with ?

teo1978, 2018-12-02:
Yes (an identical one) but I'm not sure how much that would help. With this very module, things had been working as expected for months until now (and I had tried the same sketch on two other identical modules where it always worked).
I'm going to re-test with the other module though.

teo1978, 2018-12-02:
BTW I've just seen several consecutive failed attempts to connect even when powered from the AAA batteries. Maybe the whole thing is pretty random, and just "more likely" to fail when on USB power, for some reason. Just speculating...

teo1978, 2018-12-02:
Is there any documentation to decypher what the "state: x -> y (z)" messages mean?

devyte, 2018-12-02:
I don't see a core issue here, so I'm closing, but feel free to continue discussion.
BTW, I suggest a full installation of the IDE from scratch (delete everything, including temp dirs), a full rebuild of your app with the core version you had working before, a full flash wipe, and reflash of the newly built binary.

closed 2018-12-02 by devyte