esp8266/Arduino #5432: gpio 10 doesn't work with new board 2.4.2 on Arduino IDE

claudiofrancesconi, 2018-12-04:
hello i am trying to use gpio 10 as an output on a nodemcu esp8266-12e with Arduino IDE
it used to work like charm when i used the board 2.4.1, after i upgraded to 2.4.2 it's not working anymore.

Any tips?

devyte, 2018-12-04:
Pins 9 and 10 aren't available on 8266 devices, unless you modify the board to cut the pcb lines and, then change the flash mode to not use those lines. Is that what you're doing?
Or do you have some special board?

claudiofrancesconi, 2018-12-04:
as i said with board 2.4.1 worked like charm

torntrousers, 2018-12-04:
It did used to work: https://github.com/esp8266/Arduino/issues/2124#issuecomment-225171544. I haven't tried it recently though.

devyte, 2018-12-04:
And on most 8266 boards using those GPIO with releases prior to 2.4.2 could potentially fry the pins. Or the flash.
You didn't answer my question: what is your specific use case? Did you modify the board by cutting the lines as explained in the above link, or do you have a special board, or something else?

I seem to remember you can still have access to them, but you are now required to change your board settings to one of the generic boards, I don't remember which (generic 8285 maybe?). Alternatively, you have to manually init those pins yourself in the same way the other pins are inited on bootup.

torntrousers, 2018-12-04:
The NodeMCU 1.0 is defined to use DIO mode - https://github.com/esp8266/Arduino/blob/master/boards.txt#L2354 - so you can (at least used to) just be able to use GPIO10 without and mod's anywhere.

claudiofrancesconi, 2018-12-04:
> And on most 8266 boards using those GPIO with releases prior to 2.4.2 could potentially fry the pins. Or the flash.
> You didn't answer my question: what is your specific use case? Did you modify the [...]

as i told you
i didn't change anything
just uploaded the sketch with Arduino IDE and board 2.4.1... i use the pin 10 with TONE command to play a buzzer

Pablo2048, 2018-12-04:
You can try to check SPI flash mode - probably you have DIO in previous version of IDE and QIO in 2.4.2...

devyte, 2018-12-04:
@claudiofrancesconi I understand that you didn't change anything on your end. I'm telling you that some things related to the init of pins 9 and 10 changed in the core with release 2.4.2, because it was deemed dangerous or nonsensical to leave them as they were for the general use case of the specific boards involved. 

Example of nonsensical: init pin as output, write LOW, try to read flash => crash.
Example of dangerous: same as above, but flash pin fries instead of just the ESP crashing.

That's why I asked whether you modified your ESP12e-based Nodemcu when you started using it with 2.4.1 to allow generic use of GPIO 9/10.
See e.g.: [this schematic](https://www.esp8266.com/wiki/lib/exe/fetch.php?media=schematic_esp-12e.png) for an ESP12e that shows GPIO 9 and 10 are physically connected to the flash.

If you didn't modify your board, then maybe you were lucky (some flash chips have short protection) or there is some special board case that needs to be tested and allow the pins for the variants involved.

d-a-v, 2018-12-04:
gpio 9 and 10 have been recently set to usable only in DOUT mode with esp8285 (by @earlephilhower after he rewrote the whole PWM code). 
@claudiofrancesconi, 
- is your board esp8285 based ?
- is it an esp8266 in DOUT mode with gpio9&10 not connected to the flash chip ?
In those case, could you try with latest git version ?
Otherwise, we need to understand your setup.

torntrousers, 2018-12-04:
@devyte do you happen to have the PR to hand that did that change? Doesn't an ESP8266 using DIO mean GPIO 9 and 10 are available and so should be useable with ESP8266/Arduino?

d-a-v, 2018-12-04:
@torntrousers I'm digging looking or it right now

d-a-v, 2018-12-04:
It's #5055. Gpio9&10 are now allowed only with esp8285 (previously always disabled) (that's after 2.4.2).
Changes are needed to allow them in DOUT mode for esp8266 wired like esp8285 with gpio9&10 routed to users.

torntrousers, 2018-12-04:
Interesting. #5055 doesn't have any explanation for why it was changed to disallow using GPIO10 with an ESP8266 in DIO mode. It certainly did used to work ok and didn't 'fry' anything, as per this issue and #2124.

d-a-v, 2018-12-04:
It changed because @earlephilhower rewrote the whole PWM/tone thing, and ommitted gpio9&10 in the first place, because they are flash pins, not commonly available. Then someone asked why gpio9&10 were gone. The rest of the story is #5055.

torntrousers, 2018-12-04:
I think its a regression. There aren't a lot of GPIO's on an ESP8266 so to unnecessarily prevent GPIO10 from being useable seems a shame.

d-a-v, 2018-12-04:
Calling it a shame is a little harsh but I'm not native english speaker so I might misunderstand :)
We are asking for use case, and more details. I personally have not encountered any esp8266 boards with gpio9&10 routed out. 
The only boards that do not have gpio9&10 connected to flash chips for sure are esp8285 and these pins are available for them.
Tell us a way to identify those esp8266 boards that use DOUT only, with those pins floating into the air, so we can (or you can, as being already a contributor you know how things work) propose a PR to make the whole thing shameless.
Things are now being like they are because pinModeing those pins would lead to crashes.

devyte, 2018-12-04:
Guys, you're missing the point. The point is that GPIO 9 & 10 weren't inited one way or the other in any board before. Then it was found that with a hw hack they could be used in some cases. Then some users would go and try to use them in their code without taking into account their board variant, and so in some cases there would be crashes, in other cases there would be damage, and in other cases it would happily work. 
The decision was made to have them disabled in all 8266 cases because in most commercial boards they are hardwired to the flash, and enabled in all 8285 cases where they are free to be used by design.
If there are 8266 board variants where their use is available by design, they can be enabled for that variant. If there are variants where it's unknown, and testing works, it must be analyzed case by case as to why, and if ok they can be enabled for those variants. If it works because the flash chip is protecting against the forced levels, I'd rather not make that "standard use" either (not all flash chips are the same).
The current state is that you can still use them, but you have to init them yourself and thereby be aware that what you're doing is not normal use, and that there could be consequences.

@torntrousers said:
> Doesn't an ESP8266 using DIO mean GPIO 9 and 10 are available

No, it doesn't. Their being usable requires two things:
1. they have to be disconnected from the flash
2. mode has to be set to DIO.

If you only set to DIO and don't cut the lines, the flash pin could contend with the ESP setting of the GPIO level. 
If you're lucky, nothing happens and it "will work". If you're less lucky, the flash chip will fry and not the ESP pin, in which case you can freely use the GPIOs, but you can never again go back to QIO (this happened to me with one board). If you're unlucky, the ESP pin will fry, and then you lost both GPIO and QIO mode.
Or no damage will happen and the ESP will just crash. Or something else (on another of my boards, the whole ESP died).

torntrousers, 2018-12-05:
@d-a-v see examples 6 and 7 [here](https://ludwig.guru/s/seems+a+shame).

I'm really sceptical that using GPIO10 can fry or damage the flash chip or ESP pin - electronically, how could that happen? Both the ESP and flash pins are I/O and work at 3.3v, and in QIO mode they are connected to something going high and low so they are designed to work for that, and in DIO mode its not like this could put an unexpected un-tolerated voltage on the pin, and the flash chip datasheets don't have any warnings about dangers to the pin in DIO mode. And I and others have tried it on lots of ESPs and I've never heard before of it not working. What sort of module were you using when you thought it broke it and how was it wired up and were you doing anything else at the time that could have been the cause?

torntrousers, 2018-12-05:
You say: _If there are 8266 board variants where their use is available by design, they can be enabled for that variant. If there are variants where it's unknown, and testing works, it must be analyzed case by case as to why, and if ok they can be enabled for those variants._ so how about another menu option thats available when using the 'Generic ESP8266 Module' board type that optionally enables the use of GPIO10 if you know what you're doing?

torntrousers, 2018-12-05:
Sorry, I missed this bit _The current state is that you can still use them, but you have to init them yourself_ - if thats the case could you just show a code example of how to do that? And I'd be happy just doing that.

TD-er, 2018-12-05:
You may also want to check wat this pin does at boot, or power-up.
It may very well be the pin is still doing something, even when setting the flash to DIO mode.
So connecting a transistor + relay to it may give accidental triggers on the relay.
Or what if a connected switch is pulling it up and the ESP tries to pull it down.

devyte, 2018-12-12:
Nodemcu 0.9 is set to qio. 
Nodemcu 1.0 is set to dio.
ESP8285 is set to dout.

If you want to use the pins, I think you have to set the correct flash mode and then init the pin on startup in the same way that the ESP8285 does. See the following references:
- [init for esp8285](https://github.com/esp8266/Arduino/blob/master/variants/esp8285/initD9D10Pins.cpp) 
- [isFlashInterfacePin for esp8285](https://github.com/esp8266/Arduino/blob/bc2d4ec18bdaf160b2859d3681dacbc6c9e20516/variants/esp8285/common.h#L34)
- [isFlashInterfacePin for other variants](https://github.com/esp8266/Arduino/blob/92373a98370618dea09718010b30d311a97f3f25/variants/generic/common.h#L33)
- [conditional init in waveform generator](https://github.com/esp8266/Arduino/blob/20f07ecc3195a586791e72b35b1c279cd0693cac/cores/esp8266/core_esp8266_waveform.c#L84).

closed 2018-12-12 by devyte

torntrousers, 2018-12-13:
I've just tried with the 2.5.0-beta1 release and using GPIO-10 seems to be working fine when using DIO, eg with a NodeMCU 1.0. No other changes needed. Try it with the Blink example and just change LED_BUILTIN to 10.