esp8266/Arduino #5437: How to assign a link local address when DHCP client fails:"help wanted"

amontefusco, 2018-12-04:
I need to connect ESP8266 device to ISOLATED networks, so networks where there are no services like DHCP and DNS.

While the mDNS and SD-DNS are working very well, I cannot find how to enable the lower layer of zero network configuration, i.e., how to autoconfigure a link-local address in case the DHCP server is not answering.

Could you point me to info on how to activate this feature?

d-a-v, 2018-12-05:
That is a fair request.
Currently, DHCP client in both lwIP-v1 and v2 are espressif code that do not use 169.254/16 (link-local).
I think upstream lwIP's DHCP client would do that and we should switch to it.
Before we can do that, you can use a timeout in the standard waiting loop (`while (WiFi.status != WL_CONNECTED)...`) and call `WiFi.config(ip,gateway,subnet,dns)` on timeout.

d-a-v, 2018-12-07:
> Currently, DHCP client in both lwIP-v1 and v2 are espressif code that do not use 169.254/16 (link-local).

I am wrong, lack of sleep. It's DHCP client, and I just enabled AUTOIP in lwip2 #5444