esp8266/Arduino #5440: Arduino.h boolean type

mcspr, 2018-12-05:
### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below.

#### Platform

- Core Version:			d5d1126

### Settings in IDE

- Module:  			Generic ESP8266 Module

### Problem Description

I had noticed that the `boolean` is typedefed to `uint8_t`, not `bool` as one would think:
https://github.com/esp8266/Arduino/blob/d5d112683fc92124ca56a7399e1551aa03d5a880/cores/esp8266/Arduino.h#L191
Arduino base repo had an issue about this problem: https://github.com/arduino/Arduino/issues/2147

Gist of the original issue is that boolean & bool comparison through == may yield different than expected results. Just like this wiki article describes benefits of having native `bool` vs. implementing it through int: https://en.wikipedia.org/wiki/Boolean_data_type#C,_C++,_Objective-C,_AWK
> The language guarantees that any two true values will compare equal (which was impossible to achieve before the introduction of the type).

AVR, SAMD etc. Cores have it typedef'd to bool since. After that, https://github.com/arduino/Arduino/issues/4673 was raised about replacing boolean with bool all throughout the Cores.

So my questions are:
- will PR simply replacing uint8_t -> bool in Arduino.h break something here?
- should boolean used here (examples, core etc.) be replaced by bool?

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

From the issue mentioned above, slightly modified for verbosity.
Conditions resolve the same ofc.

```cpp
#include <Arduino.h>

#define CHECK(condition) Serial.print(#condition " is "); Serial.println((condition) ? "true" : "false");

void setup() {
    Serial.begin(115200);
    CHECK(true == (bool)57);
    CHECK(true == (boolean)57);
    CHECK(true == bool( true ));
    CHECK(bool( 1 | 2 | 4 | 8 ) == boolean( 1 | 2 | 4 | 8 ));
    CHECK(true == false + 1);
    CHECK(true + true == true);
    CHECK(bool( true + true ) == true);
    CHECK(boolean( true + true ) == true);
}
void loop() {}
```

### Debug Messages

```
true == (bool)57 is true
true == (boolean)57 is false
true == bool( true ) is true
bool( 1 | 2 | 4 | 8 ) == boolean( 1 | 2 | 4 | 8 ) is false
true == false + 1 is true
true + true == true is false
bool( true + true ) == true is true
boolean( true + true ) == true is false
```

devyte, 2018-12-05:
I wholeheartedly agree with this. However, I've across way too many libs out there that use boolean, and those would break.
I'll target this for release 3.0.0.

TD-er, 2018-12-05:
I think you may also break alignments in structs.
This will be a problem when those structs will be stored and read again.

devyte, 2018-12-05:
That is correct. Also, structs written to/read from EEPROM and rtcmem.

earlephilhower, 2019-01-16:
We discussed this on gitter and also did some checks in GCC code and our own tests.

`sizeof(boolean) == sizeof(bool) == 1`

So alignment issues should not be a problem.  There is still some concern about value interpretation, but I think that's actually the point of this bug, where `bool` will give a saner interpretation.   Still some concern about what happens if you read back a pre-change boolean whose value was, say, "62" and not "1".  Simple tests should verify this either works or not.

TD-er, 2019-01-16:
`sizeof` will always return the same indeed, but as soon as you're dealing with a struct with the packed pragma/attribute, then it really makes a difference whether it is an `uint8_t` or a `bool`.
Also it may be interesting to see what the `sizeof` is of:
- `bool[32]` vs `boolean[32]`
- struct with 32 members of type `bool` vs `boolean`.

earlephilhower, 2019-01-16:
In GCC when `BOOL_TYPE_SIZE` undefined, it's defined to `CHAR_TYPE_SIZE`.  It's undefined in xtensa.h, so to GCC all packing/physical layout will be identical between char, uint8_t, and bool.

Even `#pragma packed` should behave the same because the layout rules will have the same input be it `boolean` or `bool` or `char`.

Some of the confusion may be that the *linker* will place consecutive `uint8_t`s at separate 4-byte alignment on Xtensa and not pack it (like you'd have on an 8-bit CPU like the AVR).  So `char a, b;` takes 8 bytes of heap even though `char a[2]` only takes 4.

earlephilhower, 2019-01-16:
````
typedef struct {
  boolean b;
  uint16_t u;
} __attribute__((__packed__)) s_bp;

typedef struct {
  bool b;
  uint16_t u;
} __attribute__((__packed__))  s_cp;

void setup() {
Serial.begin(115200);
Serial.printf("offsetof s_bp.u=%d\n", offsetof(s_bp, u));
Serial.printf("offsetof s_cp.u=%d\n", offsetof(s_cp, u));
}
void loop() { }
````

Generates:
````
offsetof s_bp.u=1
offsetof s_cp.u=1
````

TD-er, 2019-01-17:
I just did a test with these:
```c++
typedef struct {
  boolean b;
  boolean c;
  boolean d[16];
  uint16_t u;
} __attribute__((__packed__)) s_bp;

typedef struct {
  bool b;
  bool c;
  bool d[16];
  uint16_t u;
} __attribute__((__packed__))  s_cp;
```

With only bool/boolean b & c, both reported an offset of 2.
With the array d added both have an offset of 18.
So it seems both bool and boolean are indeed treated as 1 byte in size.

d-a-v, 2019-01-17:
So there are no mis-alignment issue.
Only issues left would be that some boolean users assume a `boolean` could be different from 0 or 1.
I'd vote to push this change just after release 2.5.0.

PR #5693 (2019-02-07): Use "bool" for "boolean" type
Match current Arduino definition to avoid issues with comparison
operations.

https://github.com/arduino/Arduino/issues/2147
https://github.com/arduino/Arduino/commit/20ac20f6295b5bd923144ab6844564f13ddc8ca8

Fixes #5440
commit e74b1e59 (2019-02-07): Use "bool" for "boolean" type (#5693)

Match current Arduino definition to avoid issues with comparison
operations.

https://github.com/arduino/Arduino/issues/2147
https://github.com/arduino/Arduino/commit/20ac20f6295b5bd923144ab6844564f13ddc8ca8

Fixes #5440
closed 2019-02-07 by devyte