esp8266/Arduino #5446: compile error of smartconfig

i3water, 2018-12-07:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", is related to 3rd party libs, or is related to hardware, please discuss at a community forum like esp8266.com.

INSTRUCTIONS
============
If you do not follow these instructions, your issue may be dismissed.

1. Follow the checklist under Basic Infos and fill in the [ ] spaces with an X.
2. Fill in all the fields under Platform and Settings in IDE marked with [ ] (pick the correct option for you in each case, delete the others).
3. If you haven't already done so, test your issue against current master branch (aka latest git), because it may have been already fixed.
4. Describe your problem.
5. If you have a STACK DUMP decode it:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/stack_dump.html

6. Include a Minimal Complete Reproducible Example sketch that shows your issue. Do not include your entire project, or a huge piece of code.
7. Include debug messages:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/debugging.html 

8. Use markup (buttons above) and the Preview tab to check what the issue will look like.
9. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:					[WiFiduino (ESP-12)]
- Core Version:			[8b3f496](https://github.com/esp8266/Arduino/commit/8b3f4966e966617b8c00b03e82b4672c5b513a7c)
- Development Env:		[Arduino IDE 1.8.8]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[WiFiduino]
- Flash Mode:			[dio]
- Flash Size: 		[4MB-3MBSPIFFS]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] (serial upload only)

### Problem Description

con`t compile with smartconfig. latest three day, i had test with the latest git version, but all con`t compile.the package i download at 27 november can compile smartconfig codes.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp
#include <ESP8266WiFi.h>
#define LED 2
 
void smartConfig()
{
  WiFi.mode(WIFI_STA);
  Serial.println("\r\nWait for Smartconfig");
  WiFi.beginSmartConfig();
  while (1)
[... 52 lines cut ...]

c:/users/12102/documents/arduino/hardware/esp8266/esp8266/tools/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: C:\Users\12102\Documents\Arduino\hardware\esp8266\esp8266/tools/sdk/lib\libsmartconfig.a(sc_airkiss.o):(.text.KISS_Answer+0x1c): undefined reference to `espconn_create'

c:/users/12102/documents/arduino/hardware/esp8266/esp8266/tools/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: C:\Users\12102\Documents\Arduino\hardware\esp8266\esp8266/tools/sdk/lib\libsmartconfig.a(sc_airkiss.o):(.text.KISS_Answer+0x7d): undefined reference to `espconn_port'

c:/users/12102/documents/arduino/hardware/esp8266/esp8266/tools/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: C:\Users\12102\Documents\Arduino\hardware\esp8266\esp8266/tools/sdk/lib\libsmartconfig.a(sc_airkiss.o):(.text.KISS_Answer+0xa8): undefined reference to `espconn_regist_sentcb'

c:/users/12102/documents/arduino/hardware/esp8266/esp8266/tools/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: C:\Users\12102\Documents\Arduino\hardware\esp8266\esp8266/tools/sdk/lib\libsmartconfig.a(sc_airkiss.o):(.text.KISS_Answer+0xb0): undefined reference to `espconn_create'

c:/users/12102/documents/arduino/hardware/esp8266/esp8266/tools/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: C:\Users\12102\Documents\Arduino\hardware\esp8266\esp8266/tools/sdk/lib\libsmartconfig.a(sc_esptouch.o):(.text.TOUCH_Send_platform+0x85): undefined reference to `espconn_sent'

c:/users/12102/documents/arduino/hardware/esp8266/esp8266/tools/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: C:\Users\12102\Documents\Arduino\hardware\esp8266\esp8266/tools/sdk/lib\libsmartconfig.a(sc_esptouch.o):(.text.TOUCH_udp_send_cb+0x6a): undefined reference to `espconn_delete'

c:/users/12102/documents/arduino/hardware/esp8266/esp8266/tools/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: C:\Users\12102\Documents\Arduino\hardware\esp8266\esp8266/tools/sdk/lib\libsmartconfig.a(sc_esptouch.o):(.text.TOUCH_Send_ACK+0x14): undefined reference to `espconn_regist_recvcb'

c:/users/12102/documents/arduino/hardware/esp8266/esp8266/tools/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: C:\Users\12102\Documents\Arduino\hardware\esp8266\esp8266/tools/sdk/lib\libsmartconfig.a(sc_esptouch.o):(.text.TOUCH_Send_ACK+0x52): undefined reference to `espconn_port'

c:/users/12102/documents/arduino/hardware/esp8266/esp8266/tools/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: C:\Users\12102\Documents\Arduino\hardware\esp8266\esp8266/tools/sdk/lib\libsmartconfig.a(sc_esptouch.o):(.text.TOUCH_Send_ACK+0x7d): undefined reference to `espconn_regist_sentcb'

c:/users/12102/documents/arduino/hardware/esp8266/esp8266/tools/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: C:\Users\12102\Documents\Arduino\hardware\esp8266\esp8266/tools/sdk/lib\libsmartconfig.a(sc_esptouch.o):(.text.TOUCH_Send_ACK+0x89): undefined reference to `espconn_regist_recvcb'

c:/users/12102/documents/arduino/hardware/esp8266/esp8266/tools/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: C:\Users\12102\Documents\Arduino\hardware\esp8266\esp8266/tools/sdk/lib\libsmartconfig.a(sc_esptouch.o):(.text.TOUCH_Send_ACK+0x92): undefined reference to `espconn_create'

collect2.exe: error: ld returned 1 exit status
```

earlephilhower, 2018-12-07:
LIBHAL was replaced with an open source version which does not include the espconn* calls.

The old libhal from the repo included a full, hidden, LWIP stack AFAICT with the espconn* calls being wrappers for LWIP calls, I think.

We can restore the old libhal.a and it's stack that's only used in the binary blobs from Espressif, or do a thin wrapper from espconn->tcpconn, like I think Espressif did.

@d-a-v what's your opinion?

earlephilhower, 2018-12-07:
Looks to be defined in liblwip_gcc.a
````
~/Arduino/hardware/esp8266com/esp8266/tools/sdk/lib$ nm liblwip_gcc.a  | grep espconn_create
00000340 T espconn_create
````

d-a-v, 2018-12-07:
#5444