esp8266/Arduino #5451: Compatibility of HTTP Update and WebSocketClient in ESP8266

DeltaVetal26, 2018-12-08:
Hi,

I want to make ESP8266 a public clientthrough WebSocket and can be remotely updated using HTTP Update.

My code:
```c++
#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>
#include <ESP8266HTTPClient.h>
#include <ESP8266httpUpdate.h>
#include <WebSocketsClient.h>
#include <Hash.h>

ESP8266WiFiMulti WiFiMulti;
WebSocketsClient webSocket;

void webSocketEvent(WStype_t type, uint8_t * payload, size_t length) {

	switch(type) {
		case WStype_DISCONNECTED:
			break;
		case WStype_CONNECTED: {
			webSocket.sendTXT("Connected");
		}
			break;
		case WStype_TEXT:
			break;
	}
}

void setup() {
    for(uint8_t t = 4; t > 0; t--) {
      delay(1000);
    }
    WiFi.mode(WIFI_STA);
    WiFiMulti.addAP("SSID", "PASS");

    // WebSocket Connect
    webSocket.begin(11.11.111.11, 1111, "/");
    webSocket.onEvent(webSocketEvent);
    webSocket.setReconnectInterval(5000);
}

void loop() {
    if((WiFiMulti.run() == WL_CONNECTED)) {
      t_httpUpdate_return ret = ESPhttpUpdate.update("http://server/file.bin");
    }
    // WebSocket
    webSocket.loop();
}
  ```

When I compile, I get the following error:

``` WifiHTTPUpdate:23: error: too many decimal points in number

     webSocket.begin(11.11.111.11, 1111, "/"); 

C:\Users\alexm\Desktop\WifiHTTPUpdate\WifiHTTPUpdate.ino: In function 'void setup()':

WifiHTTPUpdate:24: error: 'webSocketEvent' was not declared in this scope

     webSocket.onEvent(webSocketEvent);
exit status 1
too many decimal points in number ```

What could be the problem?
I use:
* - ESP8266 (NodeMCU)
* - ESP8266httpUpdate.h Lib
* - ArduinoWebSockets Lib (https://github.com/Links2004/arduinoWebSockets)
* - Arduino IDE 1.6.12

Jeroen88, 2018-12-08:
This is not an issue for one of the libraries.

Maybe you should read what the compiler is telling you?:

> too many decimal points in number 

Obviously you should use IPAddress::fromString() in:

`webSocket.begin(11.11.111.11, 1111, "/");`

devyte, 2018-12-08:
Also, missing double quotes.
Closing due to not an issue and user error.

closed 2018-12-08 by devyte