esp8266/Arduino #5452: Having a issue with the ESP8266-01 by uploading the sensor data to the localhost.

frustrat01, 2018-12-08:
The below shows the issues when I open the Serial Monitor in the Arduino

` ets Jan  8 2013,rst cause:4, boot mode:(1,0)

wdt reset
`
By the way, I have tried using the WiFiClient example to upload the dummy data to the localhost and it works. But it cannot work with uploading the sensor to the localhost. 
Anyone has solution please help me to get rid of this issue ASAP

_Originally posted by @frustrat01 in https://github.com/_render_node/MDEyOklzc3VlQ29tbWVudDQ0NTQ3MzAwOA==/timeline/issue_comment#issuecomment-445473008_

frustrat01, 2018-12-08:
[ESP8266_upload.txt](https://github.com/esp8266/Arduino/files/2659929/ESP8266_upload.txt)
fgfsdg

mateusz-szafraniec, 2018-12-08:
You completely ignored the issue template. Not a word about hardware, or version of library, sensors etc.

devyte, 2018-12-08:
Previous comment is correct. Also, no explanation of what localhost is, is what uploading sensor data means, etc.
Closing due to lack of info.
I suggest seeking help at a community forum like esp8266.com or stackoverflow. This is an issue tracker, meant for tracking issues in the core hosted in this repo, and not for providing general help.

closed 2018-12-08 by devyte