esp8266/Arduino #5456: Exception(29) raises during sending data to php script

Thoma-HauC, 2018-12-08:
### Basic Infos

- [ ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12]
- Core Version:			[2.4.2]
- Development Env:		[Arduino IDE]
- Operating System:		[Gentoo Linux]

### Settings in IDE

- Module:  			[Nodemcu]
- Flash Mode:			[qio|dio|other]
- Flash Size: 		[4MB]
- lwip Variant:   [v2 Higher Bandwidth]
- Reset Method:		[nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] (serial upload only)

### Problem Description

After several HTTP requests, a exception 29 raises up.

As "ESP Exception Decoder" returns always "Decode failed", I use the following instruction:

> ~/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-addr2line -e /tmp/arduino_build_691542/Test_POST.ino.elf 0x4020cdea

And the exception raises up in the below file:

**/Users/igrokhotkov/e/ESPTools/crosstool-NG/.build/src/gcc-4.8.5/libstdc++-v3/src/c++98/list.cc:130**

### Sketch

```cpp
#include <ESP8266WiFi.h>
#include <WiFiClientSecure.h>

WiFiClientSecure client;
const int port = 443;

String ssid = "MY_SSID";
String pass = "MY_PASSPHRASE";

const char* host = "your.domain.tld";
String url = "/index.php";

const unsigned long length = 128;
String data = "12345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012";

int counter;

void setup()
{
  counter = 0;
  wifiConnection();

  data = data.substring(0, length - 2);
  data += "\r\n";
}

void loop()
{
  String payload = String(counter++) + " (";
  payload += String(millis()) + ") ";

  payload += data.substring(payload.length());
  httpsPost(url, payload);
}

boolean wifiConnection()
{
  Serial.begin(115200);
  WiFi.begin(ssid.c_str(), pass.c_str());
  int count = 0;
  while ( count < 20 )
  {
    if (WiFi.status() == WL_CONNECTED)
    {
      return (true);
    }
    delay(500);
    count++;
  }
  return false;
}

String httpsPost(String url, String data)
{
  unsigned long startTime = millis();
  if (!client.connected())
  {
    client.stop();
    client.connect(host, port);
  }

  startTime = millis();
  client.println("POST " + url + F(" HTTP/1.1"));
  client.println("Host: " + (String)host);
  client.println(F("User-Agent: ESP8266/1.0"));
  client.println(F("Connection: close"));
  client.println(F("Content-Type: application/json"));
  client.print(F("Content-Length: "));
  client.println(data.length());
  client.println();
  client.println(data);
  client.flush();

  unsigned long timeout = millis();
  do
  {
    if (client.available() != 0)
    {
      break;
    }
    delay(50);
  }
  while ((millis() - timeout) < 5000);
  if (client.available() == 0)
  {
    Serial.println(F("Timed out after 5000ms..."));
  }
}
```

### Debug Messages

```
Exception (29):
epc1=0x4020cdea epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: cont 
sp: 3ffffcd0 end: 3fffffd0 offset: 01a0

>>>stack>>>
3ffffe70:  ffffffff 3fffc6fc 00000001 3ffef3c0  
3ffffe80:  3fffdad0 00000000 3fffff8c 3fffbf6c  
3ffffe90:  3ffef284 40204e3c 00000002 3fffff98  
3ffffea0:  3ffef284 40204e0c 00000010 401004e8  
3ffffeb0:  3ffef284 40204274 00000010 40204e0c  
3ffffec0:  00000005 00000001 3fff03c4 40204274  
3ffffed0:  3fff03d8 00000000 3ffe89a8 40204285  
3ffffee0:  3fff03d8 00000000 3fffff10 3fffff98  
3ffffef0:  3ffe851c 0000001e 3ffef284 3fffff98  
3fffff00:  40244cf0 3ffef284 00000003 4020532c  
3fffff10:  3ffe8574 3fffff8c 3ffef284 4020471c  
3fffff20:  3ffe851c 3fffff8c 3ffef284 402054d4  
3fffff30:  3ffe851c 3fffff8c 3ffef284 402027a3  
3fffff40:  00000000 00000000 00000000 00000000  
3fffff50:  00000000 00000000 3fffff8c 40204ad8  
3fffff60:  3fffdad0 00000000 3ffef3b8 3ffef3c0  
3fffff70:  3fffdad0 00000000 3ffef3b8 402028c5  
3fffff80:  00000000 00000000 00000000 3fff0834  
3fffff90:  0000008f 00000080 3fff421c 0000001f  
3fffffa0:  00000019 3fff2bac 0000008f 00000080  
3fffffb0:  3fffdad0 00000000 3ffef3b8 40204fbc  
3fffffc0:  feefeffe feefeffe 3ffe8548 40100739  
<<<stack<<<

last failed alloc call: 40204274(16)

```

Jeroen88, 2018-12-08:
If you use a WiFiClientSecure you should set one of the security options, (insecure, CA cert, fingerprint). See the examples
The payload you send is not a json

devyte, 2018-12-10:
Not a support forum, please seek peer help at a community forum like esp8266.com or stackoverflow.
Also, you should try latest git or wait for the imminent 2.5.0 release, where bearssl is the default for secure comms, and  axtls is left behind as deprecated.
Closing.

closed 2018-12-10 by devyte