esp8266/Arduino #5461: [Problem] ets Jan  8 2013,rst cause:4, boot mode:(1,7)

frustrat01, 2018-12-09:
Having some issues with the ESP8266-01 by uploading the data. 

The problems show: 

chksum 0x42
csum 0x42
~ld
 ets Jan  8 2013,rst cause:4, boot mode:(1,7)
wdt reset

Anyone knows what can I do to fix it?

devyte, 2018-12-09:
Most problems with the esp01 have to do with putting it into and out of bootloader, which is related to the reset circuit that you implement and how you connect it to the ttl-usb cable.
This is the second time you have ignored the issue template. I suggest reading it carefully, as well as reading the issue POLICY doc. Please ask such questions at a community forum like esp8266.com or stackoverflow, and not here.
Closing due to lack of info.

closed 2018-12-09 by devyte

ElectricBeat, 2022-03-29:
@devyte i can't understand, please explain elaborate and  give me some example