esp8266/Arduino #5462: HTTP_UPDATE_FAILD Error (-106)

DeltaVetal26, 2018-12-09:
Hi,

### Basic Infos

Hardware: ESP8266 NodeMCU

### Problem Description

When you try to download a new firmware, an error occurs:
```txt
HTTP_UPDATE_FAILD Error (-106): Verify Bin Header Failed
```

### Default Sketch

```cpp

#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>
#include <ESP8266HTTPClient.h>
#include <ESP8266httpUpdate.h>

#define USE_SERIAL Serial

ESP8266WiFiMulti WiFiMulti;

void setup() {

    USE_SERIAL.begin(115200);
    // USE_SERIAL.setDebugOutput(true);

    USE_SERIAL.println();
    USE_SERIAL.println();
    USE_SERIAL.println();

    for(uint8_t t = 4; t > 0; t--) {
        USE_SERIAL.printf("[SETUP] WAIT %d...\n", t);
        USE_SERIAL.flush();
        delay(1000);
    }

    WiFi.mode(WIFI_STA);
    WiFiMulti.addAP("Hardwell", "123456789");

}

void loop() {
    // wait for WiFi connection
    if((WiFiMulti.run() == WL_CONNECTED)) {

        t_httpUpdate_return ret = ESPhttpUpdate.update("http://mydomain.com/arduino.bin");
        //t_httpUpdate_return  ret = ESPhttpUpdate.update("https://server/file.bin");

        switch(ret) {
            case HTTP_UPDATE_FAILED:
                USE_SERIAL.printf("HTTP_UPDATE_FAILD Error (%d): %s", ESPhttpUpdate.getLastError(), ESPhttpUpdate.getLastErrorString().c_str());
                break;

            case HTTP_UPDATE_NO_UPDATES:
                USE_SERIAL.println("HTTP_UPDATE_NO_UPDATES");
                break;

            case HTTP_UPDATE_OK:
                USE_SERIAL.println("HTTP_UPDATE_OK");
                break;
        }
    }
}

```
### Update Sketch
```cpp

#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>
#include <ESP8266HTTPClient.h>
#include <ESP8266httpUpdate.h>

#define USE_SERIAL Serial

ESP8266WiFiMulti WiFiMulti;

void setup() {
    pinMode(LED_BUILTIN, OUTPUT);
    digitalWrite(LED_BUILTIN, HIGH);
    delay(1500);
    digitalWrite(LED_BUILTIN, LOW);

    USE_SERIAL.begin(115200);
    // USE_SERIAL.setDebugOutput(true);

    USE_SERIAL.println();
    USE_SERIAL.println();
    USE_SERIAL.println();

    for(uint8_t t = 4; t > 0; t--) {
        USE_SERIAL.printf("[SETUP] WAIT %d...\n", t);
        USE_SERIAL.flush();
        delay(1000);
    }

    WiFi.mode(WIFI_STA);
    WiFiMulti.addAP("Hardwell", "123456789");

}

void loop() {
    // wait for WiFi connection
    if((WiFiMulti.run() == WL_CONNECTED)) {

        t_httpUpdate_return ret = ESPhttpUpdate.update("http://mydomain.com/arduino.bin");
        //t_httpUpdate_return  ret = ESPhttpUpdate.update("https://server/file.bin");

        switch(ret) {
            case HTTP_UPDATE_FAILED:
                USE_SERIAL.printf("HTTP_UPDATE_FAILD Error (%d): %s", ESPhttpUpdate.getLastError(), ESPhttpUpdate.getLastErrorString().c_str());
                break;

            case HTTP_UPDATE_NO_UPDATES:
                USE_SERIAL.println("HTTP_UPDATE_NO_UPDATES");
                break;

            case HTTP_UPDATE_OK:
                USE_SERIAL.println("HTTP_UPDATE_OK");
                break;
        }
    }
}

```

### Debug Messages

```txt
[SETUP] WAIT 4...
[SETUP] WAIT 3...
[SETUP] WAIT 2...
[SETUP] WAIT 1...
HTTP_UPDATE_FAILD Error (-106): Verify Bin Header Failed
```

Jeroen88, 2018-12-09:
If you take a look at the source code of ESPHTTPUpdate you can see that this error occurs if either the client can not read 4 bytes from the HTTP stream or the first byte from the stream is not 0xE9. Both errors suggest that you are not having a valid compiled binary on your server.

This is not an issue in the core.

devyte, 2018-12-09:
Thanks @Jeroen88 .
Closing due to user error.

closed 2018-12-09 by devyte

m0o0, 2019-05-20:
@DeltaVetal26 hello 
i am same here error 
 please guaide

m0o0, 2019-05-20:
@Jeroen88 What changes should I make in the code?

Jeroen88, 2019-05-20:
The URL that

`
t_httpUpdate_return ret = ESPhttpUpdate.update("http://mydomain.com/arduino.bin");
`

points at should point to a valid ESP8266 compiled binary (.bin) file.

m0o0, 2019-05-21:
thanks a lot because your reply @Jeroen88 
I use an update.php file for the update in accordance with the instruction:
http://arduino.esp8266.com/Arduino/versions/2.0.0/doc/ota_updates/ota_updates.html

I tested your url and worked correctly, but there is still a problem updating with update.php.
also The first byte of the file starts with e9.

shujie-ali, 2020-05-21:
i have a valid bin file on server (e901 0240 9cf2 1040 00f0 1040 7005 0000) but after downloading via laravel it looks like 0ae9 0102 409c f210 4000 f010 4070 0500, i don't know what is happening while downloading and esp update is failing because of it.