esp8266/Arduino #5471: Installation of NodeMCU in Arduino broken

astir13, 2018-12-10:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", is related to 3rd party libs, or is related to hardware, please discuss at a community forum like esp8266.com.

INSTRUCTIONS
============
If you do not follow these instructions, your issue may be dismissed.

1. Follow the checklist under Basic Infos and fill in the [ ] spaces with an X.
2. Fill in all the fields under Platform and Settings in IDE marked with [ ] (pick the correct option for you in each case, delete the others).
3. If you haven't already done so, test your issue against current master branch (aka latest git), because it may have been already fixed.
4. Describe your problem.
5. If you have a STACK DUMP decode it:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/stack_dump.html

6. Include a Minimal Complete Reproducible Example sketch that shows your issue. Do not include your entire project, or a huge piece of code.
7. Include debug messages:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/debugging.html 

8. Use markup (buttons above) and the Preview tab to check what the issue will look like.
9. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

### Basic Infos

- [ ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform 

N/A
- Hardware:				[ESP-12|ESP-01|ESP-07|ESP8285 device|other]
- Core Version:			[latest git hash or date]
- Development Env:		[Arduino IDE|Platformio|Make|other]
- Operating System:		[Windows|Ubuntu|MacOS]

### Settings in IDE
N/A
- Module:  			[Generic ESP8266 Module|Wemos D1 mini r2|Nodemcu|other]
- Flash Mode:			[qio|dio|other]
- Flash Size: 		[4MB/1MB]
- lwip Variant:   [v1.4|v2 Lower Memory|Higher Bandwidth]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz|160MHz]
- Upload Using:		[OTA|SERIAL]
- Upload Speed:   [115200|other] (serial upload only)

### Problem Description

the link https://github.com/esp8266/Arduino/releases/download/2.5.0-beta1/esp8266-2.5.0-beta1.zip ends in a "Not Found" error from the web server. Arduino IDE is not able to fetch the board descriptions from GitHub. This happens inside the IDE, but also just using a plain browser (Safari on MacOSX Mojave and High Sierra). There, the installation as described in the main page here does not work anymore.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

N/A

### Debug Messages

Error downloading https://github.com/esp8266/Arduino/releases/download/2.5.0-beta1/esp8266-2.5.0-beta1.zip
java.lang.RuntimeException: java.lang.Exception: Error downloading https://github.com/esp8266/Arduino/releases/download/2.5.0-beta1/esp8266-2.5.0-beta1.zip
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$2(ContributionManagerUI.java:175)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.Exception: Error downloading https://github.com/esp8266/Arduino/releases/download/2.5.0-beta1/esp8266-2.5.0-beta1.zip
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:139)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:73)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:55)
	at cc.arduino.contributions.packages.ContributionInstaller.install(ContributionInstaller.java:101)
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$2(ContributionManagerUI.java:172)
	... 1 more
Caused by: java.io.IOException: Received invalid http status code from server: 404
	at cc.arduino.utils.network.FileDownloader.downloadFile(FileDownloader.java:205)
	at cc.arduino.utils.network.FileDownloader.download(FileDownloader.java:132)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:137)
	... 5 more
 Error downloading https://github.com/esp8266/Arduino/releases/download/2.5.0-beta1/esp8266-2.5.0-beta1.zip
java.lang.RuntimeException: java.lang.Exception: Error downloading https://github.com/esp8266/Arduino/releases/download/2.5.0-beta1/esp8266-2.5.0-beta1.zip
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$2(ContributionManagerUI.java:175)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.Exception: Error downloading https://github.com/esp8266/Arduino/releases/download/2.5.0-beta1/esp8266-2.5.0-beta1.zip
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:139)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:73)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:55)
	at cc.arduino.contributions.packages.ContributionInstaller.install(ContributionInstaller.java:101)
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$2(ContributionManagerUI.java:172)
	... 1 more
Caused by: java.io.IOException: Received invalid http status code from server: 404
	at cc.arduino.utils.network.FileDownloader.downloadFile(FileDownloader.java:205)
	at cc.arduino.utils.network.FileDownloader.download(FileDownloader.java:132)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:137)
	... 5 more
 Error downloading https://github.com/esp8266/Arduino/releases/download/2.5.0-beta1/esp8266-2.5.0-beta1.zip
java.lang.RuntimeException: java.lang.Exception: Error downloading https://github.com/esp8266/Arduino/releases/download/2.5.0-beta1/esp8266-2.5.0-beta1.zip
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$2(ContributionManagerUI.java:175)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.Exception: Error downloading https://github.com/esp8266/Arduino/releases/download/2.5.0-beta1/esp8266-2.5.0-beta1.zip
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:139)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:73)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:55)
	at cc.arduino.contributions.packages.ContributionInstaller.install(ContributionInstaller.java:101)
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$2(ContributionManagerUI.java:172)
	... 1 more
Caused by: java.io.IOException: Received invalid http status code from server: 404
	at cc.arduino.utils.network.FileDownloader.downloadFile(FileDownloader.java:205)
	at cc.arduino.utils.network.FileDownloader.download(FileDownloader.java:132)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:137)
	... 5 more

earlephilhower, 2018-12-10:
We are literally in the middle of doing the release, please bear with us.  Try again in an hour so we can get it all squared away.

vascojdb, 2018-12-10:
I managed to force my older release on the settings and it solved the issue for now: http://arduino.esp8266.com/versions/2.4.2/package_esp8266com_index.json

To check your installed version go to: C:\Users\<USER>\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\

devyte, 2018-12-11:
Apologies everyone, we were in the midddle of the release process when TravisCI hit a limit which caused it to partially fail and corrupt the json. The json link should be fixed, but I'll leave this open until the release process is complete.

Swiftnesses, 2018-12-11:
Keeping my eye out for 2.5.0, hopefully it'll be available soon (at least for now the old version can be downloaded...).

vascojdb, 2018-12-11:
@astir13 haha, no need :D Took me about 3 mins to figure out since I have noticed the problem :P

Swiftnesses, 2018-12-11:
Spamming the refresh as I need to deploy some new sensors and would like to try 2.5.0...

What's the ETA guys?

igrr, 2018-12-11:
@Swiftnesses no ETA for now, sorry. If you need to deploy something I would not suggest using a beta version anyway (it is 2.5.0-beta not 2.5.0).

devyte, 2018-12-12:
Package for release 2.5.0-beta1 is done. Closing.

closed 2018-12-12 by devyte

mohamedmostafaa, 2018-12-16:
still can't update ????