esp8266/Arduino #5472: broken board manager url since 2.5.0-beta1 uploaded

buzztiaan, 2018-12-10:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", is related to 3rd party libs, or is related to hardware, please discuss at a community forum like esp8266.com.

INSTRUCTIONS
============
If you do not follow these instructions, your issue may be dismissed.

1. Follow the checklist under Basic Infos and fill in the [ ] spaces with an X.
2. Fill in all the fields under Platform and Settings in IDE marked with [ ] (pick the correct option for you in each case, delete the others).
3. If you haven't already done so, test your issue against current master branch (aka latest git), because it may have been already fixed.
4. Describe your problem.
5. If you have a STACK DUMP decode it:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/stack_dump.html

6. Include a Minimal Complete Reproducible Example sketch that shows your issue. Do not include your entire project, or a huge piece of code.
7. Include debug messages:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/debugging.html 

8. Use markup (buttons above) and the Preview tab to check what the issue will look like.
9. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

### Basic Infos

- [ ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [ ] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12|ESP-01|ESP-07|ESP8285 device|other]
- Core Version:			[latest git hash or date]
- Development Env:		[Arduino IDE|Platformio|Make|other]
- Operating System:		[Windows|Ubuntu|MacOS]

### Settings in IDE

- Module:  			[Generic ESP8266 Module|Wemos D1 mini r2|Nodemcu|other]
- Flash Mode:			[qio|dio|other]
- Flash Size: 		[4MB/1MB]
- lwip Variant:   [v1.4|v2 Lower Memory|Higher Bandwidth]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz|160MHz]
- Upload Using:		[OTA|SERIAL]
- Upload Speed:   [115200|other] (serial upload only)

### Problem Description

Detailed problem description goes here.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}

```

### Debug Messages

```
Debug messages go here
```

buzztiaan, 2018-12-10:
error msgs from arduino IDE 1.8.5 : 

```
 Error downloading https://github.com/esp8266/Arduino/releases/download/2.5.0-beta1/esp8266-2.5.0-beta1.zip
java.lang.RuntimeException: java.lang.Exception: Error downloading https://github.com/esp8266/Arduino/releases/download/2.5.0-beta1/esp8266-2.5.0-beta1.zip
        at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$1(ContributionManagerUI.java:176)
        at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.Exception: Error downloading https://github.com/esp8266/Arduino/releases/download/2.5.0-beta1/esp8266-2.5.0-beta1.zip
        at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:131)
        at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:69)
        at cc.arduino.contributions.packages.ContributionInstaller.install(ContributionInstaller.java:102)
        at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$1(ContributionManagerUI.java:173)
        ... 1 more
Caused by: java.io.IOException: Received invalid http status code from server: 404
        at cc.arduino.utils.network.FileDownloader.downloadFile(FileDownloader.java:194)
        at cc.arduino.utils.network.FileDownloader.download(FileDownloader.java:128)
        at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:129)
        ... 4 more

```

buzztiaan, 2018-12-10:
fyi, this bug means all Arduino IDE users that have internet now cant compile for their boards

diception, 2018-12-11:
lookes like arduino ide is looking for https://github.com/esp8266/Arduino/releases/download/2.5.0-beta1/esp8266-2.5.0-beta1.zip but the file is actually named https://github.com/esp8266/Arduino/releases/download/2.5.0-beta1/Arduino-2.5.0-beta1.zip

buzztiaan, 2018-12-11:
so probably a typo in http://arduino.esp8266.com/stable/package_esp8266com_index.json

buzztiaan, 2018-12-11:
actually, shouldnt that json have more versions anyway, instead of just the latest beta version?

igrr, 2018-12-11:
The json served from http://arduino.esp8266.com/stable/package_esp8266com_index.json has been reverted to contain 2.4.2 and previous release. We are looking into the cause of this issue (new json not containing the previous versions).

devyte, 2018-12-11:
Closing as duplicate.

closed 2018-12-11 by devyte

buzztiaan, 2018-12-11:
what was the duplicate? i searched prior but found none

dav1901, 2018-12-11:
#5471 is the one