esp8266/Arduino #5476: Build with 2.5.0-dev is about 200k larger compared to 2.4.2

TD-er, 2018-12-11:
### Basic Infos

- [ ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [ ] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12|ESP8285 device|other]
- Core Version:			[2.5.0-beta1]
- Development Env:		[Platformio] (test with [own PIO platform](https://github.com/TD-er/ESPEasy/tree/feature/testbuild_core2_5_0_beta1) )
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Generic ESP8266 Module|other]
- Flash Mode:			[dio]
- Flash Size: 		[4MB/1MB]
- lwip Variant:   [v2 Lower Memory]

### Problem Description

I was testing to see if I could build ESPeasy with core 2.5.0-beta1 using PlatformIO.

- [Test branch ESPeasy](https://github.com/TD-er/ESPEasy/tree/feature/testbuild_core2_5_0_beta1)
- [Test branch platform-expressif8266 with core 2.5.0-beta1](https://github.com/TD-er/platform-espressif8266/tree/feature/2.5.0-beta1)
- packageRepositories [manifest.json](https://td-er.nl/ESPeasy/manifest.json)

Almost all versions I tested of our project were either unable to build due to exceeding max_size limit, or have become so large they will be useless for their intended purpose.

As an example the `minimal_ESP8266_1024_OTA` env in our project:
```
ATA:    [=====     ]  54.8% (used 44852 bytes from 81920 bytes)
PROGRAM: [========= ]  92.8% (used 828304 bytes from 892912 bytes)
```

The last Travis build of the same environment using core 2.4.2 was:
```
DATA:    [=====     ]  54.7% (used 44836 bytes from 81920 bytes)
PROGRAM: [=======   ]  69.0% (used 616192 bytes from 892912 bytes)
```

Am I missing something very obvious?
Like including some library twice (maybe moved into the core) ?

Is it something I am doing wrong with creating a PlatformIO environment?
I just made a .tar.gz from the [2.5.0-beta1 tag](https://github.com/esp8266/Arduino/tree/2.5.0-beta1) and uploaded it to my server and mentioned it in the manifest.json.

vascojdb, 2018-12-11:
@TD-er , as for now you can go to your Arduino settings and change the extra board manager to:
http://arduino.esp8266.com/versions/2.4.2/package_esp8266com_index.json

This will force you to use the previous one, while the new one is not stable yet (it is still a development branch)

devyte, 2018-12-11:
@TD-er what happens if you disable exceptions?

devyte, 2018-12-11:
https://github.com/esp8266/Arduino/blob/47d6f83f5f2204c8b6e180e7aeacd4e2473c83a6/tools/boards.txt.py#L887

Notice the -lstdc++-nox as well, as opposed to -lstdc++ in the enabled case.

TD-er, 2018-12-11:
Then I get a lot of these (and still too big, will also test the "buildable" minimal version too)

```
d:\ivan\projects\arduinoesp\toolchain\dl\gcc-xtensa\build-2\xtensa-lx106-elf\libstdc++-v3\libsupc++/cp-demangle.c:4697:(.text+0x3ea0): dangerous relocation: call0: call target out of range: strcmp
c:/users/gijs/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/lib\libsupc++.a(cp-demangle.o): In function `d_print_flush':
d:\ivan\projects\arduinoesp\toolchain\dl\gcc-xtensa\build-2\xtensa-lx106-elf\libstdc++-v3\libsupc++/cp-demangle.c:3702:(.text+0x417d): dangerous relocation: call0: call target out of range: sprintf
d:\ivan\projects\arduinoesp\toolchain\dl\gcc-xtensa\build-2\xtensa-lx106-elf\libstdc++-v3\libsupc++/cp-demangle.c:3703:(.text+0x4182): dangerous relocation: call0: call target out of range: strlen
c:/users/gijs/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/lib\libsupc++.a(cp-demangle.o): In function `d_append_char':
d:\ivan\projects\arduinoesp\toolchain\dl\gcc-xtensa\build-2\xtensa-lx106-elf\libstdc++-v3\libsupc++/cp-demangle.c:3716:(.text+0x4559): dangerous relocation: call0: call target out of range: sprintf
d:\ivan\projects\arduinoesp\toolchain\dl\gcc-xtensa\build-2\xtensa-lx106-elf\libstdc++-v3\libsupc++/cp-demangle.c:3716:(.text+0x455e): dangerous relocation: call0: call target out of range: strlen

c:/users/gijs/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/lib\libsupc++.a(cp-demangle.o): In function `d_append_buffer':
d:\ivan\projects\arduinoesp\toolchain\dl\gcc-xtensa\build-2\xtensa-lx106-elf\libstdc++-v3\libsupc++/cp-demangle.c:3725:(.text+0x47f5): dangerous relocation: call0: call target out of range: sprintf
c:/users/gijs/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/lib\libsupc++.a(cp-demangle.o): In function `d_print_comp':
d:\ivan\projects\arduinoesp\toolchain\dl\gcc-xtensa\build-2\xtensa-lx106-elf\libstdc++-v3\libsupc++/cp-demangle.c:4912:(.text+0x47fa): dangerous relocation: call0: call target out of range: strlen
c:/users/gijs/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/lib\libsupc++.a(cp-demangle.o): In function `d_print_flush':
d:\ivan\projects\arduinoesp\toolchain\dl\gcc-xtensa\build-2\xtensa-lx106-elf\libstdc++-v3\libsupc++/cp-demangle.c:3705:(.text+0x4919): dangerous relocation: call0: call target out of range: sprintf
c:/users/gijs/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/lib\libsupc++.a(cp-demangle.o): In function `d_append_char':
d:\ivan\projects\arduinoesp\toolchain\dl\gcc-xtensa\build-2\xtensa-lx106-elf\libstdc++-v3\libsupc++/cp-demangle.c:3716:(.text+0x491e): dangerous relocation: call0: call target out of range: strlen
c:/users/gijs/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/lib\libsupc++.a(cp-demangle.o): In function `d_print_flush':
d:\ivan\projects\arduinoesp\toolchain\dl\gcc-xtensa\build-2\xtensa-lx106-elf\libstdc++-v3\libsupc++/cp-demangle.c:3703:(.text+0x528e): dangerous relocation: call0: call target out of range: sprintf
d:\ivan\projects\arduinoesp\toolchain\dl\gcc-xtensa\build-2\xtensa-lx106-elf\libstdc++-v3\libsupc++/cp-demangle.c:3705:(.text+0x5293): dangerous relocation: call0: call target out of range: strlen
c:/users/gijs/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/lib\libsupc++.a(cp-demangle.o): In function `d_append_char':
d:\ivan\projects\arduinoesp\toolchain\dl\gcc-xtensa\build-2\xtensa-lx106-elf\libstdc++-v3\libsupc++/cp-demangle.c:3713:(.text+0x5a9b): dangerous relocation: call0: call target out of range: strncmp
d:\ivan\projects\arduinoesp\toolchain\dl\gcc-xtensa\build-2\xtensa-lx106-elf\libstdc++-v3\libsupc++/cp-demangle.c:3713:(.text+0x5aea): dangerous relocation: call0: call target out of range: strlen
c:/users/gijs/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/lib\libsupc++.a(cp-demangle.o): In function `d_mangled_name':
d:\ivan\projects\arduinoesp\toolchain\dl\gcc-xtensa\build-2\xtensa-lx106-elf\libstdc++-v3\libsupc++/cp-demangle.c:1104:(.text+0x5bfd): dangerous relocation: call0: call target out of range: strlen
d:\ivan\projects\arduinoesp\toolchain\dl\gcc-xtensa\build-2\xtensa-lx106-elf\libstdc++-v3\libsupc++/cp-demangle.c:1113:(.text+0x5c41): dangerous relocation: call0: call target out of range: strlen
c:/users/gijs/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/lib\libsupc++.a(cp-demangle.o): In function `d_print_flush':
d:\ivan\projects\arduinoesp\toolchain\dl\gcc-xtensa\build-2\xtensa-lx106-elf\libstdc++-v3\libsupc++/cp-demangle.c:3703:(.text+0x5d2d): dangerous relocation: call0: call target out of range: strlen
c:/users/gijs/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/lib\libsupc++.a(cp-demangle.o): In function `d_demangle_callback':
d:\ivan\projects\arduinoesp\toolchain\dl\gcc-xtensa\build-2\xtensa-lx106-elf\libstdc++-v3\libsupc++/cp-demangle.c:5460:(.text+0x5d3d): dangerous relocation: call0: call target out of range: memcpy
collect2.exe: error: ld returned 1 exit status
*** [.pioenvs\test_ESP8266_4096_VCC\firmware.elf] Error 1
 [ERROR] Took 78.08 seconds
 ```

TD-er, 2018-12-11:
I guess this one is related:  https://github.com/esp8266/Arduino/issues/5423

And according to the description post [here](https://github.com/esp8266/Arduino/commit/6280e98b0360f85fdac2b8f10707fffb4f6e6e31), I should rebuild the toolchain?

d-a-v, 2018-12-11:
You need to update it with `tools/get.py`.

d-a-v, 2018-12-11:
You can disable exceptions, and also select the lwIP 'no features' variant.

d-a-v, 2018-12-11:
Is your local repository even with current master ?

porkyneal, 2018-12-11:
@TD-er 

I personally removed the entire platform folder then used

```
platform = https://github.com/platformio/platform-espressif8266.git#feature/stage 
```

and 

```
build_flags = -DNDEBUG -fno-exceptions -lstdc++-nox -DPIO_FRAMEWORK_ARDUINO_LWIP2_LOW_MEMORY_LOW_FLASH
```

In my platformio.ini file. Everything compiles fine and runs fine.

TD-er, 2018-12-11:
@porkyneal With those settings, you're building with core 2.4.2.
I try to build core 2.5.0 beta1.

@d-a-v  This repository you mean, I assume?
I downloaded from this tag: https://github.com/esp8266/Arduino/tree/2.5.0-beta1

After diner, I will also test to build in Linux to see if it makes a difference.

porkyneal, 2018-12-11:
> @porkyneal With those settings, you're building with core 2.4.2.
> I try to build core 2.5.0 beta1.

@TD-er Did you try?

With those settings I have the very latest GIT being compiled, admittedly not the exact 2.5.0.beta.1.
It is pulling the latest master of this repository. Which as it stands is roughly 2.5.0 beta 1.

See [Using Arduino Framework with Staging version](https://docs.platformio.org/en/latest/platforms/espressif8266.html#using-arduino-framework-with-staging-version)

devyte, 2018-12-11:
Don't rely on the tag as it currently stands. I expect to have to delete and remake it as soon as the currently ongoing work is finished.

TD-er, 2018-12-11:
@porkyneal I just started a build with the feature/stage on a (slow) linux machine, so I can see it after diner.
If that doesn't result in a 2.5.0 dev build, I will try with a new fetch like devyte suggested.

TD-er, 2018-12-11:
I have tried building using the staging version, with the same results.
So at least my initial test (using a self defined platform + download) was done "right" so it seems.
And the stage is indeed a core 2.5.0-dev build.
But the size is still way too big.
After that I tried building with the exception off like suggested before and also with running `get.py`. (update download took forever)
With exception code excluded, it is not possible to build, with the same errors as posted before.

TD-er, 2018-12-11:
Deleting stuff, getting it again, etc. and using the following flags:
```
build_flags               = -D BUILD_GIT='"${sysenv.TRAVIS_TAG}"'
                            -DNDEBUG
                            -DVTABLES_IN_FLASH
                            -fno-exceptions
                            -lstdc++-nox
                            -DPIO_FRAMEWORK_ARDUINO_LWIP2_LOW_MEMORY
                            -DCORE_2_5_0
```
The last one to handle some parts in the code that have to change between core 2.4.2 and 2.5.0

The minimal build I started with, is now:
```
DATA:    [=====     ]  54.5% (used 44664 bytes from 81920 bytes)
PROGRAM: [=======   ]  71.5% (used 638640 bytes from 892912 bytes)
```
It is still about 17k larger compared to core 2.4.2.
And still too big for our intended purpose, but that's another issue :)

But when building any other env. in our project I get this:
```
Linking .pioenvs\normal_ESP8266_4096\firmware.elf
c:/users/gijs/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: .pioenvs\normal_ESP8266_4096\firmware.elf section `.text' will not fit in region `iram1_0_seg'
collect2.exe: error: ld returned 1 exit status
*** [.pioenvs\normal_ESP8266_4096\firmware.elf] Error 1
```

Running `objdump -t firmware.elf` I see almost any function name present.
`objdump -h firmware.elf` from the minimal build (that one finished building):

```
firmware.elf:     file format elf32-little

Sections:
Idx Name          Size      VMA       LMA       File off  Algn
  0 .data         0000074c  3ffe8000  3ffe8000  000000e0  2**4
                  CONTENTS, ALLOC, LOAD, DATA
  1 .noinit       00000004  3ffe874c  3ffe874c  0000082c  2**2
                  CONTENTS, ALLOC, LOAD, DATA
  2 .irom0.text   00092194  40201010  40201010  00009e00  2**4
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
  3 .text         00007d6c  40100000  40100000  00002094  2**2
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
  4 .rodata       00001864  3ffe8750  3ffe8750  00000830  2**4
                  CONTENTS, ALLOC, LOAD, DATA
  5 .bss          00008ec8  3ffe9fb8  3ffe9fb8  00002098  2**4
                  ALLOC
  6 .comment      00001c28  00000000  00000000  0009bf94  2**0
                  CONTENTS, READONLY
  7 .xtensa.info  00000038  00000000  00000000  0009dbbc  2**0
                  CONTENTS, READONLY
....
```

The text part being very close to 32k (0x7D6C = 32108 = 660 bytes left) makes it likely we will cross that one in no-time.

What else can I do to make the iram usage smaller?

d-a-v, 2018-12-11:
try with `-DPIO_FRAMEWORK_ARDUINO_LWIP2_LOW_MEMORY_LOW_FLASH`

TD-er, 2018-12-11:
> -DPIO_FRAMEWORK_ARDUINO_LWIP2_LOW_MEMORY_LOW_FLASH

Same result:
```
Linking .pioenvs\normal_ESP8266_4096\firmware.elf
c:/users/gijs/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: .pioenvs\normal_ESP8266_4096\firmware.elf section `.text' will not fit in region `iram1_0_seg'
collect2.exe: error: ld returned 1 exit status
*** [.pioenvs\normal_ESP8266_4096\firmware.elf] Error 1
 [ERROR] Took 61.85 seconds
Environment normal_ESP8266_4096                 [ERROR]
 [ERROR] Took 61.86 seconds
```

earlephilhower, 2018-12-12:
The flash size increasing by 17K, what is that compared to?  Flash size is definitely not a design point, so it doesn't surprise me that there is some increase to go with the added functionality.  The exceptions stuff you seem to have sorted, so I don't believe it's related to that (but might be things like the floating point support in printf, etc. which weren't in older releases).

As for the IRAM(interrupt) issue, that should be looked at in both your code and the core.  There's not much space to begin with and even the ESP SDK itself continues to increase in size there.

I've given it an honest try, but as I'm not an Atom user, PlatformIO user, or a ESPEasy user, I can't for the life of me figure out how to build what you're building using the tools you're using and the updated bits (obviously in flux) that have to swap out the standard stuff.  Following the Windows guide online, I got as far as getting Atom on my Linux box, getting PIO, and installing the 8266 branch from a .ZIP of the GH branch you referenced, but in addition to a espressif8266@src-eeabd... directory I see it snarfed a couple other 8266 and even ESP32 dirs in ~/.platformio.  And it's been building something like 10 different versions of the ELF in the background as I've been typing.

What you can do is increase the space allocated in the `ld*` file to the IRAM region (to 0x9000 or so from 0x8000). That ELF won't run of course, but you will get a successful link that you will be able to look at.  At that point, you can do an objdump of the .text section to get function names and sizes and can compare to your prior builds to see if there's some standout code section that grew or appeared.  I can think of ~500bytes of low hanging fruit that can be moved to HEAP from IRAM, but more detailed analysis would be needed for more effective results.

TD-er, 2018-12-12:
Thanks, I will do that this morning, to get an idea what is eating up all the iRAM.

As Windows + PlatformIO is not in your normal Workflow, then you're far away from your normal comfort zone, I assume :)
And up for a very steep learning curve if you're planning to get to know all their quirks.
I do both, Windows and Linux, with the same end-results. So at least Windows can be left out of the equation ;)

I will play around with the objdump, which looks to me like a very powerful tool, but I am yet unfamiliar with it.
Do you have some link to a description/example where objdump was used to track down these iRAM squatters? And what is causing code to be placed in iRAM by the compiler/linker?
I know about the `IRAM_ATTR` flag which is meant to place **executable** code into iRAM and there is `ICACHE_FLASH_ATTR` which is meant to do the opposite. (am I right on that?)

As far as I know, we are using ESP8266_NONOS_SDK, so I guess I need to add `ICACHE_FLASH_ATTR` to a lot of functions?
In ESP8266_RTOS_SDK the use of this `ICACHE_FLASH_ATTR` seems to be the default, so isn't needed there. Can we make it default too in the NonOS SDK?

So I guess my first thing to do is to find the biggest users of the iRAM and then attribute them with `ICACHE_FLASH_ATTR` (as long as the code is not being used on time-critical parts like interrupt handlers. (which our code hardly does)

d-a-v, 2018-12-12:
I tried to compile ESPEasy with arduino IDE.
It indeed doesn't fit with 2.5.0-beta1 (no exception, lwip-1.4 or lwip-2 (no features/low flash)).
With 2.4.2 I have
```
Sketch uses 791036 bytes (75%) of program storage space. Maximum is 1044464 bytes.
Global variables use 40804 bytes (49%) of dynamic memory,
leaving 41116 bytes for local variables. Maximum is 81920 bytes.
```

I have both 2.4.2 and 2.5.0-beta1 map files.
We need an (opensource) parser to compare these map files.
I found this freeware/closed source [map file browser](http://www.sikorskiy.net/prj/amap/index.html) which works very well.
Here are the 2.4.2 and 2.5.0 parsed output.

[2.4.2-map-sorted.txt.gz](https://github.com/esp8266/Arduino/files/2671355/2.4.2-map-sorted.txt.gz)
[2.5.0-beta1-map-sorted.txt.gz](https://github.com/esp8266/Arduino/files/2671356/2.5.0-beta1-map-sorted.txt.gz)

TD-er, 2018-12-12:
> I've given it an honest try, but as I'm not an Atom user, PlatformIO user, or a ESPEasy user, I can't for the life of me figure out how to build what you're building using the tools you're using and [...]

About building it, you may want to have a look at the build script I made in the tools directory of ESPeasy.
In comments at the top is a list of dependencies, which have to be met (Ubuntu/Debian).
It will create a Python virtualenv, so it will not mess up your system.
I guess building with that script may be quite easy to do and the .elf files will be in the .pioenv dir in the ESPeasy branch.

TD-er, 2018-12-12:
@d-a-v 
Thanks for the map files.
I assume the 4th column is the size used?
If so, sorting on that column and filtering the second column to ".iram", I found our own Plugin_003 and Plugin_008.  (I already had disabled the use of the Servo library)
Simply disabling them (in `define_plugin_sets.h` ) made it possible again to build.
Another one high up in the list seems to be the Adafruit NeoPixel library.
I will have a look at that one too. Maybe there is a newer version available, or else I may have to look into it to see what is possible to reduce in the iRAM usage.
Other option is to look at the (much requested) "build only what you need".

I am building all now, to see which one will fail and how much the flash size has been increased.

Are the flags   `-fno-exceptions` and `-lstdc++-nox` enough to disable the exception code? Or do I still need to patch the boards.txt.py as suggested here: https://github.com/esp8266/Arduino/issues/5476#issuecomment-446203831   ?

d-a-v, 2018-12-12:
@earlephilhower I played with the sorted files (`grep`ping and using `meld` tool).
It happens that 2.5.0 has lots of entries (in the real map file) like that:
```
 .irom.text./home/gauchard/dev/proj/arduino/ESPEasy/WebServer.ino.4152.1890
                0x00000000402a8be8        0xd /tmp/arduino_build_894040/sketch/ESPEasy.ino.cpp.o
 *fill*         0x00000000402a8bf5        0x3 
 .irom.text./home/gauchard/dev/proj/arduino/ESPEasy/WebServer.ino.4142.1889
                0x00000000402a8bf8        0x7 /tmp/arduino_build_894040/sketch/ESPEasy.ino.cpp.o
 *fill*         0x00000000402a8bff        0x1 
 .irom.text./home/gauchard/dev/proj/arduino/ESPEasy/WebServer.ino.4092.1888
                0x00000000402a8c00       0x1f /tmp/arduino_build_894040/sketch/ESPEasy.ino.cpp.o
 *fill*         0x00000000402a8c1f        0x1 
 .irom.text./home/gauchard/dev/proj/arduino/ESPEasy/WebServer.ino.4062.1887
                0x00000000402a8c20        0x6 /tmp/arduino_build_894040/sketch/ESPEasy.ino.cpp.o
 *fill*         0x00000000402a8c26        0x2 
 .irom.text./home/gauchard/dev/proj/arduino/ESPEasy/WebServer.ino.4061.1886
                0x00000000402a8c28        0xf /tmp/arduino_build_894040/sketch/ESPEasy.ino.cpp.o
 *fill*         0x00000000402a8c37        0x1 
 .irom.text./home/gauchard/dev/proj/arduino/ESPEasy/WebServer.ino.4060.1885
                0x00000000402a8c38        0xe /tmp/arduino_build_894040/sketch/ESPEasy.ino.cpp.o
```
I only looked at `@402... ` symbols. Those lines do not appear in 2.4.2.
The sum of their size is 97504 (grep/calc on sorted-filtered map file). I assume `*fill*` are not in the sorted files (there are ~4000 lines like that).
I can't figure out what `arduino/ESPEasy/WebServer.ino.4062.1887` represents.

@TD-er 
First time I watch so big sketch !
> Are the flags -fno-exceptions and -lstdc++-nox enough to disable the exception code?

I think so

> Or do I still need to patch the boards.txt.py as suggested here: #5476?

This is the boards.txt configurator, generating the menu. There is no patching here, just the arduino IDE to disanle exceptions.

TD-er, 2018-12-12:
The Webserver isn't the prettiest code around, since it will construct the web page on the fly (at impressive speed by the way, am quite proud of the speed).
But maybe these *fill* things are meant to be alignments between the quite small strings used?
I replaced a number of very short `F("bla")` parts by `"bla"` to get the flash size down, as I understood it was taking more resources to have small strings using the F-macro.
Also those normal strings will be de-duplicated by the compiler.

We are planning to just output JSON and let the browser create the HTML. That will make it even more snappy and a lot easier to maintain. Now it is very difficult to change webpage layout.

d-a-v, 2018-12-12:
I had a look at `@402...` too (iram cache is it?)
Biggest difference is block which is present in 2.5.0 only (last number is the size)
```
std::_Rb_tree_decrement(std::_Rb_tree_node_base*) libstdc++-nox.a tree.o _ZSt18_Rb_tree_decrementPSt18_Rb_tree_node_base         15
std::_Rb_tree_increment(std::_Rb_tree_node_base const*) libstdc++-nox.a tree.o _ZSt18_Rb_tree_incrementPKSt18_Rb_tree_node_base        40
std::_Rb_tree_increment(std::_Rb_tree_node_base*) libstdc++-nox.a tree.o _ZSt18_Rb_tree_incrementPSt18_Rb_tree_node_base         39
std::_Rb_tree_insert_and_rebalance(bool, std::_Rb_tree_node_base*, std::_Rb_tree_node_base*, std::_Rb_tree_node_base&) libstdc++-nox.a tree.o _ZSt29_Rb_tree_insert_and_rebalancebPSt18_Rb_tree_node_baseS0_RS_      233
std::_Rb_tree_rebalance_for_erase(std::_Rb_tree_node_base*, std::_Rb_tree_node_base&) libstdc++-nox.a tree.o _ZSt28_Rb_tree_rebalance_for_erasePSt18_Rb_tree_node_baseRS_        486
std::__detail::_List_node_base::_M_hook(std::__detail::_List_node_base*) libstdc++-nox.a list.o _ZNSt8__detail15_List_node_base7_M_hookEPS0_         14
std::__detail::_List_node_base::_M_transfer(std::__detail::_List_node_base*, std::__detail::_List_node_base*) libstdc++-nox.a list.o _ZNSt8__detail15_List_node_base11_M_transferEPS0_S1_        26
std::__detail::_List_node_base::_M_unhook() libstdc++-nox.a list.o _ZNSt8__detail15_List_node_base9_M_unhookEv         10
std::__detail::_List_node_base::swap(std::__detail::_List_node_base&, std::__detail::_List_node_base&) libstdc++-nox.a list.o _ZNSt8__detail15_List_node_base4swapERS0_S1_        75
std::__throw_bad_alloc() libstdc++-nox.a functexcept.o _ZSt17__throw_bad_allocv         12
std::__throw_bad_function_call() libstdc++-nox.a functexcept.o _ZSt25__throw_bad_function_callv         12
std::__throw_length_error(char const*) libstdc++-nox.a functexcept.o _ZSt20__throw_length_errorPKc        12
std::set_terminate(void (*)()) libstdc++-nox.a eh_terminate.o _ZSt13set_terminatePFvvE        11
std::set_terminate(void (*)()) libstdc++-nox.a eh_terminate.o _ZSt13set_terminatePFvvE        4
```

There are still lines like that
```
 .irom.text./home/gauchard/dev/proj/arduino/ESPEasy/_P052_SenseAir.ino.56.4015
                0x00000000402a0604       0x15 /tmp/arduino_build_894040/sketch/ESPEasy.ino.cpp.o
 *fill*         0x00000000402a0619        0x3 
 .irom.text./home/gauchard/dev/proj/arduino/ESPEasy/_P049_MHZ19.ino.478.4014
                0x00000000402a061c       0x19 /tmp/arduino_build_894040/sketch/ESPEasy.ino.cpp.o
 *fill*         0x00000000402a0635        0x3 
```

My (crappy) filter is:
```
cat 2.X.Y.map-sorted.txt | grep 401 | grep -v '\.debug_' | sed 's,\([^/]*\)/.*/\([^/]*\),\1 \2,g' | sort -n -k 5 | awk '{print $5 " " $6 " " $7 " " $8 " " $9 " " $10 " " $11 " " $12 " " $13 " " $14 " " $15 " " $16 " " $4;}'| grep -v ' 0$' | sort -n > 2.X.Y
meld 2.4.2 2.5.0
```
(big sed is to suppress paths, awk print $4=size in the end, sorted by symbol for `meld`)

d-a-v, 2018-12-12:
To compile with Arduino IDE, I simply made an `ESPEasy`/ symlink from the arduino sketchdir to the ESPEasy's `src/` dir (`ESPEasy.ino` in it), and I made symlinks from `libraries/ ` to each library in `lib/` dir.

Instead of patching ESPEasy about IPAddress changes (there are some issues), I fixed the errors in the core and will eventually propose a PR (to ease porting from older cores).

TD-er, 2018-12-12:
>Instead of patching ESPEasy about IPAddress changes (there are some issues), I fixed the errors in the core and will eventually propose a PR (to ease porting from older cores).

What do you mean by that?
The change of signature of some functions in UdpContext?

d-a-v, 2018-12-12:
> The change of signature of some functions in UdpContext?

Yes, those signatures you just committed fixes to, in espeasy (according to the link above)

TD-er, 2018-12-12:
Ah OK, good to hear.
This PR is just a preparation on supporting the core 2.5.0, so we can start testing a.s.a.p.
One of the most anticipated commits is the one I linked in the PR description.
The WiFi issues have taken way too much time to tackle, so I really hope those problems will be much harder to reproduce after this upgrade ;)

earlephilhower, 2018-12-12:
Can someone post a .elf of the 2.4.2 and the 2.5.1-b1 versions of the app so I can run objdump/readelf locally?  @d-a-v, your dump is good but a strange format.

`402...` is flash, that's not the critical bit.  It's `.text` and `401...` where the interrupts are stored.

`fill` sections are just alignment issues.  Code wants to begin on a 32-bit boundary, even though the xtensa has a variable-length ISA.

TD-er, 2018-12-12:
I am building a 2.5.0 set at this moment, so I will have a number of .elf files in a minute.
Any preference of an env-version? As big as possible, or just a small one?

TD-er, 2018-12-12:
The 2.5.0 built files
https://www.dropbox.com/s/gy4r3xcy1vnb1h7/ESPEasy_core_2.5.0.tgz?dl=0

Exceptions disabled on 2.4.2, can they use the same build flags?

earlephilhower, 2018-12-12:
Perfect, thanks.  With both it's not too hard to map a) individual function sizes and b) new functions.

TD-er, 2018-12-12:
I just added the 2.4.2 built files.

N.B. there is also a 'core 2.4.1' version included, since we have that also in the nightly builds due to PWM issues with 2.4.2

earlephilhower, 2018-12-12:
Got 'em, thanks.

earlephilhower, 2018-12-12:
The first big standouts I'm seeing are the 2 functions: `onSdaChange` and `onSclChange`.

Combined they are 1.3KB of code space, and they are only present in 2.5.0-b1.

There was some I2C slave support added in 2.5.0-dev, but I was not following it much.  @d-a-v  or @devyte , any ideas?  These are *big* functions now, where there were none before.

earlephilhower, 2018-12-12:
Also in the same vein as this, there are `twi_*` functions of ~ 768bytes.  Also included with the I2C changes, it seems.

earlephilhower, 2018-12-12:
`<` = 2.4.2 code, `>` = 2.5.0-b1 code
Here's my first pass command line-only compare of the function sizes and names.  Minor differences aren't a big deal, and it doesn't line up perfectly, but it's a good way to get the feel:

````
# objdump -t -j .text old.elf  | sort -k2 > a
# objdump -t -j .text new.elf  | sort -k2 > b
# diff -y a b

firmware.elf:     file format elf32-little                      firmware.elf:     file format elf32-little
00000018 .hidden __ashrdi3                                      00000018 .hidden __ashrdi3
00000000 .hidden __divsf3                                       00000000 .hidden __divsf3
00000000 .hidden __eqdf2                                        00000000 .hidden __eqdf2
00000000 .hidden __eqsf2                                        00000000 .hidden __eqsf2
00000000 .hidden __fixsfsi                                      00000000 .hidden __fixsfsi
00000000 .hidden __floatdisf                                    00000000 .hidden __floatdisf
00000000 .hidden __floatundisf                                  00000000 .hidden __floatundisf
00000000 .hidden __gedf2                                        00000000 .hidden __gedf2
00000000 .hidden __gesf2                                        00000000 .hidden __gesf2
00000000 .hidden __gtdf2                                        00000000 .hidden __gtdf2
00000000 .hidden __gtsf2                                        00000000 .hidden __gtsf2
00000000 .hidden __ledf2                                        00000000 .hidden __ledf2
00000000 .hidden __lesf2                                        00000000 .hidden __lesf2
00000000 .hidden __ltdf2                                        00000000 .hidden __ltdf2
00000000 .hidden __ltsf2                                        00000000 .hidden __ltsf2
00000049 .hidden __modsi3                                       00000049 .hidden __modsi3
00000000 .hidden __mulsf3                                     <
00000000 .hidden __nedf2                                        00000000 .hidden __nedf2
00000000 .hidden __nesf2                                        00000000 .hidden __nesf2
00000000 .hidden __unorddf2                                     00000000 .hidden __unorddf2
00000000 .hidden __unordsf2                                     00000000 .hidden __unordsf2
00000000 .text                                                  00000000 .text
0000003b Cache_Read_Disable_2                                   0000003b Cache_Read_Disable_2
0000002a Cache_Read_Enable_2                                    0000002a Cache_Read_Enable_2
00000032 Cache_Read_Enable_New                                  00000032 Cache_Read_Enable_New
0000000b GetAccess                                              0000000b GetAccess
00000052 NMI_Handler                                            00000052 NMI_Handler
00000020 NmiTimSetFunc                                          00000020 NmiTimSetFunc
0000000b RC_GetAckRate                                          0000000b RC_GetAckRate
00000014 RC_GetAckTime                                          00000014 RC_GetAckTime
0000003a RC_GetBlockAckTime                                     0000003a RC_GetBlockAckTime
00000036 RC_GetCtsTime                                          00000036 RC_GetCtsTime
0000000b RC_GetRtsRate                                          0000000b RC_GetRtsRate
SYMBOL TABLE:                                                   SYMBOL TABLE:
00000042 Tx_Copy2Queue                                          00000042 Tx_Copy2Queue
00000000 _DebugExceptionVector                                  00000000 _DebugExceptionVector
00000000 _DoubleExceptionVector                                 00000000 _DoubleExceptionVector
00000000 _KernelExceptionVector                                 00000000 _KernelExceptionVector
00000000 _NMIExceptionVector                                    00000000 _NMIExceptionVector
00000000 _NMILevelVector                                        00000000 _NMILevelVector
00000000 _UserExceptionVector                                   00000000 _UserExceptionVector
00000000 _UserExceptionVector_1                                 00000000 _UserExceptionVector_1
00000038 _Z21Plugin_008_interrupt1v                           | 00000017 _Z9sws_isr_0v
0000002d _Z21Plugin_008_interrupt2v                           | 00000017 _Z9sws_isr_1v
00000014 _Z27Plugin_003_pulse_interrupt1v                     | 00000017 _Z9sws_isr_2v
00000014 _Z27Plugin_003_pulse_interrupt2v                     <
00000014 _Z27Plugin_003_pulse_interrupt3v                     <
00000014 _Z27Plugin_003_pulse_interrupt4v                     <
00000014 _Z9sws_isr_0v                                        <
00000014 _Z9sws_isr_1v                                        <
00000014 _Z9sws_isr_2v                                        <
000000b6 _ZN21ESPeasySoftwareSerial6rxReadEv                    000000b6 _ZN21ESPeasySoftwareSerial6rxReadEv
00000017 __attachInterrupt                                    | 0000001a _ZNSt8__detail15_List_node_base11_M_transferEPS0_S1_
                                                              > 0000004b _ZNSt8__detail15_List_node_base4swapERS0_S1_
                                                              > 0000000e _ZNSt8__detail15_List_node_base7_M_hookEPS0_
                                                              > 0000000a _ZNSt8__detail15_List_node_base9_M_unhookEv
[... cut ...]
00000344 wDev_ProcessFiq                                        00000344 wDev_ProcessFiq
00000058 wDev_SetFrameAckType                                   00000058 wDev_SetFrameAckType
00000047 wDev_SetWaitingQueue                                   00000047 wDev_SetWaitingQueue
00000063 wdt_feed                                               00000063 wdt_feed
00000009 xPortWantedSizeAlign                                   00000009 xPortWantedSizeAlign
00000000 xthal_set_intclear                                   <
````

TD-er, 2018-12-12:
I forgot to mention, in the platformIO config I used are the following disabled in 2.5.0 builds:
- P003
- P008
- Servo.

earlephilhower, 2018-12-13:
@d-a-v your dump was the key.  Turns out the linker script was placing the C++ std libs into IRAM by default since the name was changed to enable non-exception-mode.  With the 1-line PR IRAM is at 0x7c34 out of 0x8000 in the main ESPEasy build that I'm able to do, which is w/in limits.

adrionics, 2018-12-13:
Do you know 2.5 now uses BearSSL by default? That uses quite a bit more flash.

TD-er, 2018-12-13:
I will include some 2.5.0-beta builds with our next nightly build, so some more people will test it. (let's hope they take the "beta" label and the warnings I add serious so we will not see reports like "I have now 2 dozen units I have to manually flash back"...)

My test units running the 2.5.0 code feel a lot more stable with WiFi.
Only the loop-count is a lot lower. (was 6k - 20k, now 1k - 2k2 runs of `loop()` per second)
Anything added in background processes that may take more time now?

TD-er, 2018-12-13:
@d-a-v 
We have a quite elaborate timing stats page in ESPeasy ;)

For example:

Description | Function | #calls | call/sec | min (ms) | Avg (ms) | max (ms)
-- | -- | -- | -- | -- | -- | --
P_1_Switch input - Switch | ONCE_A_SECOND | 11374 | 2.00 | 0.046 | 0.060 | 0.098
P_1_Switch input - Switch | TEN_PER_SECOND | 113528 | 19.96 | 0.114 | 0.189 | 71.485
P_1_Switch input - Switch | WRITE | 27 | 0.00 | 2.252 | 2.316 | 2.426
P_1_Switch input - Switch | EVENT_OUT | 1718 | 0.30 | 0.046 | 0.062 | 0.094
P_1_Switch input - Switch | FIFTY_PER_SECOND | 562554 | 98.88 | 0.046 | 0.054 | 0.094
P_4_Environment - DHT11/12/22 SONOFF2301/7021 | READ | 956 | 0.17 | 2.414 | 7.361 | 8.460
P_4_Environment - DHT11/12/22 SONOFF2301/7021 | ONCE_A_SECOND | 5687 | 1.00 | 0.050 | 0.051 | 0.064
P_4_Environment - DHT11/12/22 SONOFF2301/7021 | TEN_PER_SECOND | 56764 | 9.98 | 0.041 | 0.047 | 0.068
P_4_Environment - DHT11/12/22 SONOFF2301/7021 | EVENT_OUT | 859 | 0.15 | 0.054 | 0.054 | 0.063
P_4_Environment - DHT11/12/22 SONOFF2301/7021 | FIFTY_PER_SECOND | 281277 | 49.44 | 0.046 | 0.048 | 0.068
Load File |   | 588 | 0.10 | 1.730 | 2.133 | 2.710
Save File |   | 7 | 0.00 | 210.251 | 328.267 | 543.279
Loop |   | 9147325 | 1607.87 | 0.442 | 0.648 | 1717.942
Plugin call 50 p/s |   | 281277 | 49.44 | 2.711 | 2.885 | 3.288
Plugin call 10 p/s |   | 56764 | 9.98 | 2.965 | 3.121 | 74.395
Plugin call 10 p/s U |   | 56764 | 9.98 | 1.025 | 1.166 | 1.323
Plugin call 1 p/s |   | 5687 | 1.00 | 3.515 | 3.968 | 97.650
SensorSendTask() |   | 956 | 0.17 | 5.003 | 50.574 | 68.044
sendData() |   | 859 | 0.15 | 44.564 | 47.023 | 69.655
Compute formula |   | 832 | 0.15 | 0.019 | 0.019 | 0.024
setNewTimerAt() |   | 401824 | 70.63 | 0.136 | 0.168 | 0.279
hostByName() |   | 1 | 0.00 | 34.693 | 34.693 | 34.693
WiFi.isConnected() |   | 27111395 | 4765.51 | 0.025 | 0.028 | 0.134

Or the graphical view with color highlighting the ones > 100 msec:
[image]

There is also JSON output of it ;)

d-a-v, 2018-12-13:
What core version are you comparing with ?
Are you able to identify one particular call which is slower than before ?

TD-er, 2018-12-13:
I have not yet looked at such a detailed level.
The screenshot I took was using core 2.5.0
I just noticed the loop counter was much lower than before. (1607/s in the screenshot)
Another one running an older version (based on core 2.4.2):
[image]

I will flash a 2.4.2 build to the same Sonoff TH16 here on my desk for comparison (using same settings)

TD-er, 2018-12-13:
And a core 2.4.2 build of the same codebase, same settings, same module as the Sonoff TH16 before
[image]

Only running for 160 seconds now, so the loop count could increase a bit more when running longer (lot is done right after start)
It is 
-1607/sec for core 2.5.0 
- 4807/sec for core 2.4.2

earlephilhower, 2018-12-13:
Maybe it's nothing, but the calls/sec # for some of these are double the indicated amount, no?

TD-er, 2018-12-13:
> Maybe it's nothing, but the calls/sec # for some of these are double the indicated amount, no?

That's correct.
If there are 2 plugins of the same plugin-ID active, it will count twice as hard.
So the 50/sec call to a plugin may then report 100/sec after a while.

devyte, 2018-12-15:
@earlephilhower the original issue here was bin size, and then iram. I think both have been handled. Can this be closed?
About timing and loop freq, I suggest a new issue for that.

TD-er, 2018-12-15:
I agree the size issues are indeed handled.

mikekgr, 2019-01-08:
Dear Sirs, although this is a closed issue, today I compiled one of my Arduino sketch, using the master branch of the Arduino Core and I have a remarkable binary file increasing near if not even more of 200 KB. That I think is very much... Please see what can cause it.
Thanks and Best Regards,
Mike Kranidis

adrionics, 2019-01-08:
For me, setting the Arduino menu: "Tools -> Exceptions -> Disabled" solved it.

penfold42, 2019-01-09:
Exceptions off should be the default after https://github.com/esp8266/Arduino/pull/5538

devyte, 2019-01-09:
Correct, there is more work needed before they can be default-enabled.

mikekgr, 2019-01-10:
Dear Sirs, yes, you are absolute right,
The Exceptions off, did the trick !
See the two exactly the same compilations with Exceptions on and Exceptions off :
Exceptions on:
Sketch uses 520116 bytes (49%) of program storage space. Maximum is 1044464 bytes.
Global variables use 37692 bytes (46%) of dynamic memory, leaving 44228 bytes for local variables. Maximum is 81920 bytes.

Exceptions off:
Sketch uses 405372 bytes (38%) of program storage space. Maximum is 1044464 bytes.
Global variables use 37524 bytes (45%) of dynamic memory, leaving 44396 bytes for local variables. Maximum is 81920 bytes.

Thanks