esp8266/Arduino #5480: HTTPS in board manager

VictorTagayun, 2018-12-12:
Why is the board manager [http://arduino.esp8266.com/stable/package_esp8266com_index.json](url) not in https like ESP32 [https://dl.espressif.com/dl/package_esp32_index.json](url) ?
I cannot load ESP8266 in our work office, thus I need to hotspot my mobile to connect and download everytime the json file.

igrr, 2018-12-12:
Switching to HTTPS for the package download is something we will be looking into before the next release, thanks for bringing this up.

VictorTagayun, 2018-12-12:
Short term I am using github [https://github.com/esp8266/Arduino/releases/download/2.4.2/package_esp8266com_index.json](url)
Thanks for considering this.

devyte, 2018-12-12:
I should rephrase: ... in case you want to try it out and provide feedback :)

devyte, 2019-01-18:
@igrr has there been any progress?

devyte, 2019-02-04:
@igrr due date for 2.5.0 was yesterday. I assume this still needs more time and should be pushed back?

igrr, 2019-02-26:
Have set up https://arduino-test.esp8266.com/ which is generated from https://github.com/esp8266/esp8266.github.io. HTTPS seems to work. Next step is to upload existing package .json to that repo, and test that Arduino IDE can use that as a package manager URL.

gvarisco, 2019-04-08:
Hi @igrr! I can confirm that the Arduino IDE correctly works with HTTPS endpoints. Could you please enable HTTPS on **arduino.esp8266.com** as well? Is there anything we can help with, re: testing?

igrr, 2019-04-09:
@gvarisco Sure, can you try installing the package using this boards manager package URL?

https://arduino-test.esp8266.com/stable/package_esp8266com_index.json

(instead of the usual http://arduino.esp8266.com/stable/package_esp8266com_index.json)

If that works, I'll ask the DNS record owner to switch `arduino.esp8266.com` from the current A record to a CNAME record pointing to the github pages site.

igrr, 2019-04-09:
One more thing to test, is that the http->https redirection works, i.e. if the boards manager package URL is set as

`http://arduino-test.esp8266.com/stable/package_esp8266com_index.json`

then Arduino will install the package successfully, after redirection to `https`.

gvarisco, 2019-04-09:
Hi @igrr! I can confirm that installing the package using the board manager's package URL you provided (the https one) works just fine on Arduino IDE. I can also confirm that the http->https redirection (via 301) WORKS as expected, and both the package's index as well the package itself get download without any problems. You should be good to switch the main host over it.

igrr, 2019-04-11:
Okay, done. The package is now available from arduino.esp8266.com both via http and https. I have not enabled automatic redirection from http to https just yet.

Next step is to update release scripts in this repo to automatically upload the new package version for every release.

closed 2019-05-01 by earlephilhower

gvarisco, 2019-05-02:
@igrr thanks for working on #5992. I've noticed there's no 301 redirect on http://arduino.esp8266.com/ (http->https) or even better HSTS. Do you have any plan for it?

igrr, 2019-05-10:
@gvarisco Unfortunately, old versions of Arduino IDE can not verify the certificate of https://arduino.esp8266.com, so we can not enable the HTTP -> HTTPS redirect.

PR #5992 (2019-05-01): Upgrade to https: serving for JSON, links in docs
Fixes #5480
commit 93ef9e70 (2019-05-01): Upgrade to https: serving for JSON, links in docs (#5992)

* Upgrade to https: serving for JSON, links in docs

Fixes #5480

* Update boards.rst documentation

* Update more documentation http: refs to https:

* Remove obsolete staging info

* Drop obsolete versions from JSON programatically

After the final merge is done on the JSON, strip out any named versions
from the final product.

Removing 1.6.5-* and 2.5.0-beta(1,2,3) for now.

* Remove 2.4.0-rc(0/1) from JSON, too