esp8266/Arduino #5481: Change bools to bitfield in struct uart, save a few bytes heap

devyte, 2018-12-12:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:					All
- Core Version:			2.5.0-beta1
- Development Env:		All
- Operating System:		All

### Settings in IDE

- Module:  			All

### Problem Description

There are 4 bool members in a row in struct uart. This is meant to track changing them to a uint8_t:8 bitfield, which should make the struct a few bytes smaller in heap.

devyte, 2018-12-14:
The ArduinoOTA class has a similar case: 3 bools in a row

earlephilhower, 2018-12-17:
`std::vector<bool>` to the rescue!  Need to be careful with IRQ safety then, of course, since updates become RMW.  Same as with uint32->uint8 conversions.

earlephilhower, 2019-01-22:
These bools are already packed using normal GCC struct rules as tightly as possible.  I just did a simple test and `offsetof(struct uart_, rx_enabled) == 8` and `offsetof(struct uart_, tx_enabled) == 9`.

Nothing to do here.

devyte, 2019-01-22:
@earlephilhower you're thinking of byte packing. I said:
> uint8_t:8 bitfield

earlephilhower, 2019-01-22:
So you'll save 3 bytes (max, probably less because the other elements will natural-align to 16 or 32 bits) while increasing the assembly code to do it significantly. That's not worth anyone's time...

devyte, 2019-02-04:
Not critical for 2.5.0, pushing back

earlephilhower, 2019-07-21:
One concern here is the frequency of use of these variables.  Bit-manipulating will take 3x the program memory as simple load/stores, so there is a tradeoff  of more free heap for less free IRAM.

closed 2019-10-15 by devyte