esp8266/Arduino #5483: permission denied on 2.5.0beta

gibeautc, 2018-12-12:
### Basic Infos

- [ x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x ] I have tested that the issue is present in current master branch (aka latest git).
- [x ] I have searched the issue tracker for a similar issue.
- [x ] If there is a stack dump, I have decoded it.
- [ x] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12|]
- Core Version:			[latest git hash or date]
- Development Env:		[Arduino IDE]
- Operating System:		[Ubuntu]

### Settings in IDE

- Module:  			[Generic ESP8266 Module|Wemos D1 mini r2|Nodemcu|other]
- Flash Mode:			[qio|dio|other]
- Flash Size: 		[1MB]
- lwip Variant:   [v1.4]
- Reset Method:		[nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200|other] (serial upload only)

### Problem Description

Detailed problem description goes here.
 
When compiling any code, I get a permission denied error like so 
.arduino15/packages/esp8266/hardware/esp8266/2.5.0-beta1/tools/signing.py: permission denied
Permisions for that folder looked fine, but I even did a chmod 777 on the entire directory to see if that would change anything, it did not. 

This is with the latest beta version of the board info (2.5.0beta) 
When I downgrade it to the previous version, it works fine. 

### Debug Messages

```
Debug messages go here
```
.arduino15/packages/esp8266/hardware/esp8266/2.5.0-beta1/tools/signing.py: permission denied

gibeautc, 2018-12-12:
> Looks like a paranoid system configuration. Can you give the full OS config?
> I'm assuming you're running from a Linux/UNIX-like system, yes?  And there is a "~/" before the ".arduino15" that you've not copied?  If not, then maybe your Arduino copy is having trouble with $(HOME)
> From a bash prompt, please try
>
> python ~/.arduino15/packages/esp8266/hardware/esp8266/2.5.0-beta1/tools/signing.py

Correct on linux, ubuntu 18.04And the path was fully qualified, I just removed the /home/user
I had tried running the command direct from the command line and it does give me the mode not specified message, so there is something about it being called from the ide that's messing with things 

Sent from my Verizon, Samsung Galaxy smartphone

Both should give some form of "ERROR: Mode not specified..." message.
If they don't, then could you have the Arduino tree on a filesystem mounted as NOEXEC (so that nothing can run from it?).  Another possibility is a SELINUX enforcing mode config that doesn't allow executing the file.

[
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/esp8266/Arduino/issues/5483#issuecomment-446466998",
"url": "https://github.com/esp8266/Arduino/issues/5483#issuecomment-446466998",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

theflyingfijian, 2018-12-12:
> Samer problem here on OSX (Mojave), will try suggestions and revert.
> 
> Install was done via board manager.
> 
> UPDATE: Solved by issuing command:
> 
> `chmod u+rwx /Users/xxx/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0-beta1/tools/signing.py`

Worked like a charm :)

Swiftnesses, 2018-12-13:
Welcome guys, glad I could help :)

earlephilhower, 2018-12-13:
Yup, looks like there is a packaging issue in the IDE installed package.  Deleting my GIT copy and using the boards manager to install 2.5.0-b1 I too see:
````
fork/exec /home/earle/.arduino15/packages/esp8266/hardware/esp8266/2.5.0-beta1/tools/signing.py: permission denied
Error compiling for board LOLIN(WEMOS) D1 R2 & mini.

# ls -l /home/earle/.arduino15/packages/esp8266/hardware/esp8266/2.5.0-beta1/tools/signing.py
-rw-rw-r-- 1 earle earle 2832 Dec 11 15:25 /home/earle/.arduino15/packages/esp8266/hardware/esp8266/2.5.0-beta1/tools/signing.py
````

The GIT version itself has the +x bit set on the file, so I'll have to peek at our packing script...

earlephilhower, 2018-12-13:
[image]

closed 2018-12-13 by devyte

a-x-, 2018-12-14:
Still actual

hack to fix (more general solution than @Swiftnesses's):
```
chmod u+rwx $HOME/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0-beta1/tools/signing.py
```

earlephilhower, 2018-12-14:
Until -beta2 is released this is not going to show up ini tour board-manager installs.

However, the fix doesn't actually amke signing work.  The platform.txt file needs to be corrected, which is that the PR does and you'll get with the next beta.

a-x-, 2018-12-14:
Why beta versions available in stable update channel?

devyte, 2018-12-14:
@a-x- I suggest reading our [release model](https://github.com/esp8266/Arduino/tree/master/package#release-model).
1. It's a release channel, not a "stable" release channel.
2. The term "stable release" usually refers to the head of a version branch, where the branch base is the base release, and commits after that are only bug fixes. Hence, releases from that branch are usualy progressivley stable. We don't follow that release model, as explained in the link above. That means we don't do "stable" releases, we just do releases.
3. Beta releases require wide testing (that's actually true of all releases, but betas more so). Wide testing requires exposure to users. Exposure to users means the release must be made available through all channels.
4. The fact that the release version explicitly says "beta" should be enough for users to expect some issues. It's up to each user to decide installing, testing, reporting issues, and helping out with fixing them.

aahco, 2019-03-01:
I have spend a few day with different versions old and new, any kind of config none of working.

Why this that happen with esp-01 since i have remember off and on bump with uploading code we got error: espcomm_open failed.

PR #5491 (2018-12-13): Keep signing commands in platform.txt on release
The boards packager was deleting what was at the time it was written
unused lines in the platform.txt file before deploying to Arduino.
One of these lines is now needed for signing to work, so don't delete
it.

Also, explicitly call "python signing.py" because it looks like Arduino
is sanitizing/removing executable bits on files when extracting from
boards manager installations.

Fixes #5483
commit 68f5a3d0 (2018-12-13): Keep signing commands in platform.txt on release (#5491)

* Keep signing commands in platform.txt on release

The boards packager was deleting what was at the time it was written
unused lines in fht platform.txt file before deploying to Arduino.
One of these lines is now needed for signing to work, so don't delete
it.

Also, explicitly call "python signing.py" because it looks like Arduino
is sanitizing/removing executable bits on files when extracting from
boards manager installations.

Fixes #5483

* Create the build subdir, if needed, for autosign

If the temporary build/core directory isn' [... cut]