esp8266/Arduino #5486: Wifi link connection lower with 2.5.0 (beta 1) than with 2.4.2

Swiftnesses, 2018-12-12:
### Basic Infos

- [ x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x ] I have tested that the issue is present in current master branch (aka latest git).
- [x ] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [x ] I have filled out all fields below.

#### Platform

- Hardware:				[NodeMCU V1.0]
- Core Version:			[Latest Arduino IDE build]
- Development Env:		[Arduino IDE]
- Operating System:		[MacOS]

### Settings in IDE

- Module:  			[NodeMCU V1.0]
- Flash Mode:			[qio|dio|other]
- Flash Size: 		        [4MB (No SPIFFS)]
- lwip Variant:                  [v2 Higher Bandwidth]
- CPU Frequency:		[80Mhz|160MHz]
- Upload Using:		[OTA]

### Problem Description

With 2.4.2, I consistently see connections circa 20Mbps uplink / downlink and pings below 10m. After updating to 2.5.0 (Beta 1), link speed is significantly reduced and ping times often 20-100ms.

[image]

Sketch contains the following parameters:

```
  // Prevent wifi sleeping
  WiFi.setSleepMode(WIFI_NONE_SLEEP);
  // 802.11n
  WiFi.setPhyMode(WIFI_PHY_MODE_11N);
  // Station mode
  WiFi.mode(WIFI_STA);
```

Swiftnesses, 2018-12-13:
Looking into this now.

Certainly something has changed which effects sensors at the outer edge of the wifi signal (40% and below). Before they connect at 20+ Mbps TX / RX, now they connect at 1 or 2 Mbps.

The most inconvenient part is the sensors no longer present themselves for OTA updates (even with the new dns update in the loop), I assume they're dropping packets as the ping is also slower (sometimes they drop connection entirely!).

Swiftnesses, 2018-12-13:
I will try to revert this:

`void user_rf_pre_init()
{
    // *((volatile uint32_t*) 0x60000710) = 0;
    spoof_init_data = false;
    volatile uint32_t* rtc_reg = (volatile uint32_t*) 0x60001000;
    if((rtc_reg[24] >> 16) > 4) {
        rtc_reg[24] &= 0xFFFF;
        rtc_reg[30] = 0;
    }

    system_set_os_print(0);
    int rf_mode = __get_rf_mode();
    if (rf_mode >= 0) {
        system_phy_set_rfoption(rf_mode);
    }
    __run_user_rf_pre_init();
}`

WIll update if it helps.

d-a-v, 2018-12-14:
About OTA, check #5487 or run with latest git.

About connection speed, I cannot say right now.
Are you able to run in arduino IDE, enable all debug options, enable timestamp in serial logger, and copy paste the boot/dhcpclient process with 2.4.2 and 2.5.0-beta1 so we can see firmware debug log with relative durations ?

devyte, 2019-10-31:
No feedback in almost a year. Closing.

closed 2019-10-31 by devyte