esp8266/Arduino #5488: DNS-SD no more working after last update

fabiuz7, 2018-12-12:
### Basic Infos

- [x ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x ] I have tested that the issue is present in current master branch (aka latest git).
- [ x] I have searched the issue tracker for a similar issue.
- [x ] If there is a stack dump, I have decoded it.
- [x ] I have filled out all fields below.

#### Platform

- Hardware:					ESP-01
- Core Version:			20f07ecc3195a586791e72b35b1c279cd0693cac
- Development Env:		Arduino IDE
- Operating System:		Windows

### Settings in IDE

- Module:  			LolinD1 mini  v3.1.0
- Flash Mode:			dio
- Flash Size: 		4MB
- lwip Variant:   v2 Lower Memory
- Reset Method:		ck
- Flash Frequency:	40Mhz
- CPU Frequency:		80Mhz
- Upload Using:		SERIAL
- Upload Speed:             912600

### Problem Description
Hi all, 
I saw the last major update 4 days ago about mDNS, so I was curious to test it. It just run the basic example mDNS-SD_Extended.ino added the following loop code to make a continuos network scan and flashed on 2 different D1 mini.
The scan always returns no results but if I switch back to arduino version 2.4.2 it works seamlessy (both can discover each other).
Is anyone experiencing this problem?  

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```
void loop() {
 Serial.println("Sending mDNS query");
  int n = MDNS.queryService("esp", "tcp"); // Send out query for esp tcp services
  Serial.println("mDNS query done");
  if (n == 0) {
    Serial.println("no services found");
  } else {
    Serial.print(n);
    Serial.println(" service(s) found");
    for (int i = 0; i < n; ++i) {
      // Print details for each service found
      Serial.print(i + 1);
      Serial.print(": ");
      Serial.print(MDNS.hostname(i));
      Serial.print(" (");
      Serial.print(MDNS.IP(i));
      Serial.print(":");
      Serial.print(MDNS.port(i));
      Serial.println(")");
    }
  }
  Serial.println();

  Serial.println("loop() next");

}
```

devyte, 2018-12-12:
You have to call MDNS.update() in the loop(). 
This was required in the previous implementation as well, but was commonly omitted. In that case it still worked, but basically due to luck, and depending on the phase of the moon and planet alignment some timings were not respected, and you would get strange behavior and brokenness.
With the new implementation it is required for normal operation.
Please let me know if adding the call in the loop works in your case.

fabiuz7, 2018-12-12:
Thanks, it was the problem, now it works!

However I'm still wondering how many time per second should I call update()..

devyte, 2018-12-12:
@fabiuz7 you don't need to worry about that.
If you're curious, the MDNSResponder implements a polledTimeout object variation (see PolledTimeout.h and MDNSTimeFlag.h) and only does work after a specific time has elapsed. It formalizes the typical 
```cpp
unsigned startTime = millis();
....
if(millis() - startTime > timeout)
    doWork()
```
code pattern, so the real work happens only every timeout ms, and not every time that update() is called.

Closing.

closed 2018-12-12 by devyte