esp8266/Arduino #5493: Rewrite PUYA patch to be more universal and mem friendly.

TD-er, 2018-12-14:
The (current) patch:
1. uses a static 4KB buffer, which means everyone using the patch has 4KB less heap, including those users who do _not_ have a puya chip. Instead, a dynamically allocated buffer should be used.
2. evaluates the first if condition every single call of the function, although only the first one after boot actually does anything. Instead of a private static, a global could be used that is set elsewhere only once at bootup, and then if condition is no longer needed. Alternatively, The ESPClass could get a new data member (i.e.: instead of the global flag) which is set on construction, and a method to return its value.
3. calculates the flash_chip_id condition in the second if every single call of the function, which makes no sense, because it can't ever change at runtime. Instead, the condition should be fully calculated once and the result stored in a flag, which should then be checked.
4. detects the condition based on flash_chip_id, which isn't very robust, as has been shown with the flash chip size. Instead, a general purpose detection approach should be implemented that transparently covers other potential brands and manufacturers that have the same quirk, should they appear.
Example: on first bootup after flashing, choose a sector, e.g.: within the update area of the flash layout., preferably one that is unlikely to be used. Do a write test and read back sequence known to behave differently with puya chips. If puya is detected, set a quirk flag somewhere. On all successive bootups, detect that the quirk flag is set, so don't do the test again. Use the quirk flag in the write function to decide whether the additional logic is done or not.

Given the above, the patch won't be accepted as-is into our core. There have been several internal discussions on exactly what to do, but given that the patch is available, there have been, and still are, higher priorities, e.g.: figuring out how to ease IRAM usage.

There have been a lot of comments from many users in this discussion, and I see the patch is even being applied elsewhere, but I still don't see a PR here. Instead of wondering what will happen, I invite those involved to make a PR with the proposed patch, and implement at least points 1-3 above on top of it. If you come up with a good solution for point 4, even better.

_Originally posted by @devyte in https://github.com/esp8266/Arduino/issues/4061#issuecomment-447076931_

devyte, 2018-12-14:
Note: From #4061 :
Patch v3 in use by ESPEasy, which needs rework per points quoted above:
https://github.com/letscontrolit/ESPEasy/blob/mega/patches/puya_v3.patch#L8

mrWheel, 2018-12-14:
@devyte 
Good to know, but I do a lot of development for various projects that use the ESP-01 and so .. I'm stuck with the PUYA chip problems.....

TD-er, 2018-12-14:
Where is a good place to check for PUYA chip at boot? (and then allocate read buffer for the write function)
Inside the `getFlashChipId()` function?
Inside the `flashRead()` function is also possible, but then we're checking it for every read, which is silly.

By the way, this is a simple test with the dynamic allocation active in a simulated setup (faked PUYA detection) :

core 2.5.0 PUYA patch, no puya:

Description | #calls | call/sec | min (ms) | Avg (ms) | max (ms)
-- | -- | -- | -- | -- | --
Save File | 4 | 0.25 | 34.755 | 45.264 | 67.620

- Free Mem:	16168

core 2.5.0 PUYA patch, Faked Puya detect:

Description |  #calls | call/sec | min (ms) | Avg (ms) | max (ms)
-- | -- | -- | -- | -- | --
Save File | 2 | 0.04 | 41.332 | 57.544 | 73.756

- Free Mem:	11560

See the code (Work-in-progress): https://github.com/TD-er/Arduino/tree/bugfix/5493_puya_patch

devyte, 2018-12-14:
Do you mean the simple check similar to what's in the patch, or the general check I described in point 4?

TD-er, 2018-12-14:
@devyte The moment to allocate the 4k buffer.
When allocated as soon as possible, it will give the least amount of fragmentation. (and the most chance of success to allocate the block)

Point 1 - 3 should be done by the patch as far as I have it.

My idea to detect PUYA write behavior is to check for the first 4 bytes after the sketch to see if there is the word "PUYA" written.
In order to write it, I would like to write the NOT representation of "PUYA" in ASCII and then again the text "PUYA".
If it is a PUYA chip, it will contain the text "PUYA". If it is a normal flash chip, it will contain only zeroes.
If not tested, it will contain 0xFF.

Or am I missing something in my thought experiment?

devyte, 2018-12-14:
It sounds reasonable. However, please investigate the flash mem layout and start/stop addresses thoroughly (SPIFFS/no SPIFFS and update area, btw from your description I understand you're using the very start of the update area), and keep in mind that you have to handle an entire 4KB flash sector when writing ~(PUYA)PUYA.
I suggest implementing a struct with 2 const data members that are const char uint8_t[4]. Then you write the struct as a whole to the flash address. If there are additional 1-time values to add in the future, the struct and logic could then be extended.

TD-er, 2018-12-14:
Why must I write an entire flash sector?
The check does not work when it will do an erase like the normal write procedure.
The whole issue with SPIFFS not working was related to low-level writing without erasing.
Or can write calls at low level only write an entire sector at a time?

TD-er, 2018-12-14:
I will indeed look into the flash structure and right now it looks more complicated than expected.

And indeed 4 kB used even when no writes are needed is a waste.
I will think of a way to use less memory.
Would it be OK to preset some kind of 'flash profile' which determines the allocation strategy?

Only thing is, I had trouble finding my "PUYA" nodes to test.
I should have a bag full of them somewhere...

TD-er, 2018-12-14:
One other thing I notice.
There isn't a while loop in there and also no check for the size.

What guarantee do we have the write cycle is always 4 kB max?

TD-er, 2018-12-14:
In this last commit, I am using a 512 byte buffer and do checks on the amount of data to write.
https://github.com/TD-er/Arduino/commit/ed5a14e412ae14049e56a65f2c29ca6395b1937f

I have tested writing files (to SPIFFS) of 2 kB and partial updates in files, but not sure if SPIFFS also divided the amount of data to be sent in parts of 512 bytes.

N.B. would be nice if it is also tested on real PUYA hardware.

igrr, 2018-12-15:
PUYA chips have software erase page size of 256 bytes, so 256 byte buffer should also be okay.

TD-er, 2018-12-15:
It has been tested on a module with PUYA, with the suggested 256 Byte buffer and appears to be working.

devyte, 2018-12-19:
@TD-er Thank you for working to get this done!
Closing via #5504 .

closed 2018-12-19 by devyte

uzi18, 2018-12-19:
Nice job @TD-er

johncoles, 2018-12-20:
Hello I have tried pulling this in to my machine from Git but I still seem to have issues. Any hints on how to check this is working?

I've run the test Sketches from the [ initial thread](https://github.com/esp8266/Arduino/issues/4061) but running the test write code is still yielding empty lines?

Sketches:
[https://gist.github.com/johncoles/d4a2a6975fcb24ca9aa1515ece66578d](WriteReadTest.ino) [https://gist.github.com/johncoles/44e69dbc0acb71af4d610267c358f97b](FlashInfo.ino)

```
11:54:46.909 -> ====== Writing to SPIFFS file =========
11:54:46.943 -> 62296
11:54:46.943 -> 62296
11:54:46.943 -> 62296
11:54:46.943 -> 62296
11:54:46.977 -> 62296
11:54:46.977 -> 62296
11:54:46.977 -> 62296
11:54:46.977 -> 62296
11:54:47.012 -> 62296
11:54:47.012 -> 62296
11:54:47.012 -> ====== Reading from SPIFFS file =======
11:54:47.948 -> 1:
11:54:48.925 -> 2:
11:54:49.946 -> 3:
11:54:50.931 -> 4:
11:54:51.931 -> 5:
11:54:52.932 -> 6:
11:54:53.936 -> 7:
11:54:54.946 -> 8:
11:54:55.943 -> 9:
11:54:56.934 -> 10:
```

TD-er, 2018-12-20:
Did you set the build flag to activate this patch?
`PUYA_SUPPORT` must be set to 1
Thus: `-DPUYA_SUPPORT=0`

johncoles, 2018-12-20:
AH! That makes sense, sorry 6 hours of debugging my own code then I found this :D 
How do I set that??

TD-er, 2018-12-20:
You must define it as a global define when building it and probably make a clean build.
Or else it will not re-build the core libs.
You can also edit the cores/esp8266/Esp.h file to set the define to 1.

```c++
 #ifndef PUYA_SUPPORT
   #define PUYA_SUPPORT 1
 #endif
```

carlosmtolosa, 2019-01-03:
Hello @all you Gurus...

Just a favor for us amateurs and ignorants in the matter: can you please give a link with the patch and a small explanation in mortal language of how to install (use) it?

Many THANKS for your incredible work.

TD-er, 2019-01-03:
@carlosmtolosa As an example you can see how we use it in the PlatformIO.ini file for ESPeasy:
https://github.com/letscontrolit/ESPEasy/blob/5ee18ec556c9c58802af29f5fd78593905ef35c1/platformio.ini#L80-L88

daniftodi, 2019-01-04:
@TD-er in which version of platform IO was this patch merged? (If it's not merged yet in a published version, what are the steps to use it?

I was waiting for months for this patch. Thank you for fixing this.

uzi18, 2019-01-04:
same file, but look at lines 106-108

TD-er, 2019-01-04:
@daniftodi It is included in the staging version of core 2.5.0
The important lines are:
- Define PUYA_SUPPORT=1  (`-DPUYA_SUPPORT=1`)
- Use `platform                  = https://github.com/platformio/platform-espressif8266.git#feature/stage`  (like @uzi18 pointed out)

I will try to make a git branch on my own repository, so it is included in the older core versions too.
This makes it easier to build patched versions on core 2.4.x (and 2.3.x)

daniftodi, 2019-01-09:
Thank @TD-er. I'm using espressif8266 and have written some code based on this library. Can you tell me how to apply your patch on this? Because I have tried with development version (indicated inside platformio.ini the github link to project) and with PUYA_SUPPORT=1 flag and it didn't worked for me.

May be that I have to format somehow the SPIFFS before will be able to use PUYA_SUPPORT?

TD-er, 2019-01-09:
> @TD-er what should be done on our site?

I think it would be nice to have the patch included in a platformio-archive of older core versions, like I did.
The core versions already released will not be backported with this Puya patch, but apparently there is some demand for it.
So if you can add the patch to a new build of core libs (2.3.0, 2.4.0, 2.4.1 and 2.4.2) so people can use them, I think that would be nice.

Like @daniftodi already showed, it is apparently not that obvious how to change the platformio.ini to include the right platform (e.g. `https://github.com/TD-er/platform-espressif8266.git#patch/v1.7.3_Puya`) and activate the Puya support. (add `-DPUYA_SUPPORT=1` to the build flags)
So if it is included in the 'automatic update' way you normally update a mayor release with `.1` etc. then it would make it a lot easier to use I guess.
Also since I had to add something like "-Puya" to it, I had to set the matching requirements to be the exact version or else things would not include the patch anymore if you release a newer version, but that also takes away the beauty of the semi-automatic updates you can normally apply without users having to worry.

ivankravets, 2019-01-09:
We stopped maintaining custom packages and use only official releases. You are able to create own package and use it. I hope this feature https://github.com/platformio/platformio-core/issues/1367 will help to use a custom package per build environment.

TD-er, 2019-01-09:
@ivankravets As you may have seen, I already created a custom version including this patch and we're using it for ESPeasy to build the nightly builds with it.
But like I said, others may also be interested so that's why I thought it would be nice to have included in the platformIO packages.
People can now use the version I supplied, but I think lots of people will loose some valueable time searching for the solution when writing to SPIFFS is not working.

But on the other hand, I get it when you don't want to add various patches and thus move away from the official core releases.

Is there a way for me to get notified when you create a new version of those older core packages?
Then I can keep up the merging/patching.

rupin, 2019-11-09:
I came to this issue from the original issue  #4061, and took sometime to save my settings into EEPROM. Now its not the best solution (EEPROMs have limited write cycles), but it is possible and it works. You will be able to see sample code here. https://github.com/rupin/IoT-Projects/blob/master/TriggerAlarm/TriggerAlarm.ino  I hope someone finds this useful.