esp8266/Arduino #5503: code freezes if wifi off during a client.print()

mars000, 2018-12-15:
----------------------------- Delete below -----------------------------
Using V2.4.2 Core
Hardware: Wemos Mini

Problem:
I have two ESP8266 Wemos devices. One is a client and the other the server.
The client communicate fine with the server; no problems. However if I turn off the server Wemos the client freezes (most time) and based on the Serial output (see below) it seems to be stuck at the client.println() statement. No watchdog kicks in it just stops there and the only way to break the freeze is to do a hard reset on the Wemos.

Client Code snipped:
```
String json_message;
  root.printTo(json_message);

  Serial.println("============ NEW MESSAGE ====>");
  client.setTimeout(5000);
  if (!client.connected())
  {
    if (!client.connect(gateway_apIP, 80)) // Connection to the server
    {
      Serial.println("FAILED to connected to gateway_apIP");
      connected_flag = false;
    }
    else
    {
      connected_flag = true;
      client.keepAlive(7200, 75, 100);
    }
  }

  if (connected_flag)
  {
    Serial.print("getKeepAliveCount() = ");
    Serial.print(client.getKeepAliveCount());
    if (client.isKeepAliveEnabled())
    {
      Serial.print(" CLIENT: KEEP-ALIVE - active  ");
    }
    else
    {
      Serial.print(" CLIENT: KEEP-ALIVE - NOT active  ");
    }

   Serial.println ("***581***");
   Serial.print ("status = "); Serial.println (client.status());
    if (client.connected())
    {
      Serial.println ("***584***");

     client.println(json_message); // send data to server via TCP only if connected to server  [FREEZES HERE]

     Serial.println ("***590***");
   
    client.print('\n');
    }
     Serial.println ("***599***");
    Serial.println(json_message); // print to serial port.....
  }
}
```

Serial Output:
```
============ NEW MESSAGE ====>
getKeepAliveCount() = 100 CLIENT: KEEP-ALIVE - active  ***581***
status = 4
***584***               <==== freezes right here (only occurs when I turn off Server Wemos)
```

d-a-v, 2018-12-17:
Please
- try with `client.keepAlive(5, 5, 2);`
- retry with 2.5.0-beta2,
- provide a full demonstrating MCVE

earlephilhower, 2019-06-28:
Closing as obsolete, with 6 months and no feedback and 3 releases since the version tested with IIRC a fix where potential infinite loops in WiFiClient were cleaned up.

closed 2019-06-28 by earlephilhower