esp8266/Arduino #5509: HandleClient OTA over webbrowser triggers Watchdog

RobertHolscher, 2018-12-16:
Hi , my ESP8266 (Nodemcu v3) is rebooting most of the time when trying to do an OTA over the webbrowser. Yes I do a reset after the serial update but the OTA is triggering the wtd function of the ESP.

I can not find any documentation how the HandleClient() is working and why it triggers the watchdog.
I have the handleclient in my loop. The loop() is only having 3 calls , including the handleclient. The software is really up to its limits and I have the feeling that finding enough free space to upload the OTA is causing the watchdog. Is there a good way to put a yield() somewhere in the library of the Httpserver?

Thanks in advance for your help. Obviously the answer would be to reduce the memory footprint, but trust me, all bits have been accounted for.

d-a-v, 2018-12-17:
Are you triggering OTA from a webserver callback ?

RobertHolscher, 2018-12-17:
I am using the webbrowser upload function where my Nodemcu is acting as the webserver. There is a file select button in the library and an upload. By pressing the upload button the username and password are asked. After putting in the credentials and pressing ok, the board reboots directly which is not good.

arihantdaga, 2018-12-19:
@RobertHolscher In your case -  Are continuous wdt resets happening after doing the OTA ? or during the OTA, In your case - OTA Is not able to complete and device hits a wdt reset but start working fine after it restarts ? Or Does your issue looks something related to this - https://github.com/xoseperez/espurna/issues/1436 ?

RobertHolscher, 2018-12-20:
Thank you all for your remarks. The firewall is not the problem, because sometimes it works and I guess the firewall would always block the signal and then it would not work in 1 out of the 6 tries.
I have two type of problems with the OTA. The OTA over the webbrowser will recognize the username and password clarifications and then will go to an exception 29 or to a hardware WDT cause 4.
The third option is that it works and the update is performed. The reason why it think it is memory related is the unpredictable behavior. The hardware watch dog I can not analyse since there is no stack. Any ideas would be appreciated.
When my program is compiled the following is reported about memory usage during compile time.: De schets gebruikt 422784 bytes (40%)  programma-opslagruimte. Maximum is 1044464 bytes.
Globale variabelen gebruiken 54436 bytes (66%) van het dynamisch geheugen. Resteren 27484 bytes voor lokale variabelen. Maximum is 81920 bytes. (translated Global variables use 54436 bytes (66%) of dynamic memory. 27484 for local variables (max is 81920 bytes. The ESP8266 is a 4 M memory. 1M is spiffs (prefered is 2 M spiff but I am experimenting with this.

arihantdaga, 2018-12-21:
@RobertHolscher  Which core version are you using ?
Can you try with the latest Core. Take a pull from master and (You might have to redo the `python get.py` in tools folder. I am not sure, you can try reinstalling the complete core again). In my case with the latest core OTA  is working fine.