esp8266/Arduino #5512: Unable to use all 4Mbyte of flash for program with custom scatter file

Menion2k, 2018-12-17:
Hello
I am trying to use the entire SPI 4Mbyte of flash of my NodeMCU ESP 12-E for program, rather the available Arduino partitioning that leave to program about 1Mbyte and the rest for SPIFFS
In order to do so I have modified the board.txt of ESP8266 2.4.2 core with this:

```
nodemcu.menu.FlashSize.4M0=4M (no SPIFFS)
nodemcu.menu.FlashSize.4M0.build.flash_size=4M
nodemcu.menu.FlashSize.4M0.build.flash_size_bytes=0x400000
nodemcu.menu.FlashSize.4M0.build.flash_ld=eagle.flash.4m0.ld
nodemcu.menu.FlashSize.4M0.build.spiffs_pagesize=256
nodemcu.menu.FlashSize.4M0.upload.maximum_size=4168704
nodemcu.menu.FlashSize.4M0.build.rfcal_addr=0x3FC000
```
and writing a new scatter file eagle.flash.4m0.ld

```
/* Flash Split for 4M chips */
/* sketch 4071KB */
/* eeprom 20KB */

MEMORY
{
  dport0_0_seg :                        org = 0x3FF00000, len = 0x10
  dram0_0_seg :                         org = 0x3FFE8000, len = 0x14000
  iram1_0_seg :                         org = 0x40100000, len = 0x8000
  irom0_0_seg :                         org = 0x40201010, len = 0x3f9ff0
}

PROVIDE ( _SPIFFS_start = 0x405FB000 );
PROVIDE ( _SPIFFS_end = 0x405FB000 );
PROVIDE ( _SPIFFS_page = 0x0 );
PROVIDE ( _SPIFFS_block = 0x0 );

INCLUDE "../ld/eagle.app.v6.common.ld"
```
This works as soon as my binary is within 1Mbyte, as soon as I link in the data that makes the binary 1.6Mbyte about, just linking it, the board does not start, it outputs crap on serial-
Honestly I had to write the scatter file trying to follow what done in the other, some info are not clear to me, I suspect that there is some wifi calibration data reserved in flash, I tried to do the same in my linker also.

igrr, 2018-12-17:
ESP8266 hardware does not support mapping more than 1 megabyte of external Flash into the CPU address space at a given time. You may experiment with bank switching (reconfiguring mapping on the fly when an IllegalInstruction exception happens), but such things are definitely out of scope for this project, at least for now.

Menion2k, 2018-12-17:
Ah. Did not know it. Also ESP32 shares the same limitation? Anyhow I will go for SPIFFS, hoping that it won't use too much RAM

igrr, 2018-12-17:
ESP32 can map up to 4MB into code and 4MB into data address space each. SPIFFS is probably a reasonable approach if you need to store data.

closed 2018-12-17 by igrr